openrazer/openrazer #107: Razer Blade 2016 Supper

aushani, 2016-12-07:
Device (from lsusb):

Bus 001 Device 005: ID 1532:020f Razer USA, Ltd

terrycain, 2016-12-07:
Did you look at https://github.com/terrycain/razer-drivers/issues/68

aushani, 2016-12-07:
#68 is the stealth, would that be different?

terrycain, 2016-12-07:
Technically no as the keyboard is the same `1532:020f'. That branch still needs a bit of testing though.

aushani, 2016-12-07:
I tried the forked branch and that doesn't seem to work if that's what you wanted me to try.

terrycain, 2016-12-07:
What did it say?

aushani, 2016-12-07:
With the fork, here's what I get:

```
$ tree /sys/bus/hid/drivers
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:046D:C52F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:046D:C52F.0001
│   ├── 0003:046D:C52F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:046D:C52F.0002
│   ├── 0003:1532:020F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0003
│   ├── 0003:1532:020F.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-razer
│   ├── 0003:1532:020F.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0005
│   ├── bind
│   ├── module -> ../../../../module/hid_razer
│   ├── new_id
│   ├── uevent
│   └── unbind
└── hid-rmi
    ├── 0018:06CB:5F41.0006 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-1532020F:00/0018:06CB:5F41.0006
    ├── bind
    ├── module -> ../../../../module/hid_rmi
    ├── new_id
    ├── uevent

$ polychromatic-tray-applet 

** (polychromatic-tray-applet:7948): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-9rzsgBMngH: Connection refused
Using "/usr/share/locale/" or falling back to "en_US" for translations.

Tray applet couldn't be initalised!
Reason: Unable to connect to daemon.
Reloading in 10 seconds.
```
 With the 020f branch, the keyboard just simply doesn't work. The mouse still works (although the mouse clicks don't work)

terrycain, 2016-12-08:
Oh the hid-razer module isnt helping as its binding to the device we want. Can you successfully run `modprobe razerkbd` ?

aushani, 2016-12-12:
```
$ modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.8.7-040807-generic
```

(Sorry for the delayed response)

aushani, 2016-12-12:
OK, I must have been in a weird state from a few days ago, cause I can get `modprobe razerkbd` to run successfully now.

`$ tree /sys/bus/hid/drivers
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:046D:C52F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:046D:C52F.0001
│   ├── 0003:046D:C52F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:046D:C52F.0002
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-razer
│   ├── bind
│   ├── module -> ../../../../module/hid_razer
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-rmi
│   ├── 0018:06CB:5F41.0006 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-1532020F:00/0018:06CB:5F41.0006
│   ├── bind
│   ├── module -> ../../../../module/hid_rmi
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── 0003:1532:020F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0003
    ├── 0003:1532:020F.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0004
    ├── 0003:1532:020F.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0005
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind
`

However, I still get the same error with `polychromatic-tray-applet`

terrycain, 2016-12-12:
Thats because theres no daemon support yet.. Gimme a sec and i'll add it

terrycain, 2016-12-12:
Also whats the output of `ls -lah /dev/input/by-id`

Zerophase, 2016-12-13:
Anyone know if the pro is the same keyboard / touchpad model as well?

terrycain, 2016-12-13:
run `lsusb`

Zerophase, 2016-12-13:
I would, but I don't own that laptop yet. Just checking if I order it after the new year, if  I'll need to add it.

terrycain, 2016-12-13:
it'll be 50 50 ha.

laxdragon, 2016-12-13:
I just got this laptop myself, doing a lsusb I get:

`Bus 001 Device 005: ID 1532:020f Razer USA, Ltd `

This might help too:

```
ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 160 Dec 13 16:31 .
drwxr-xr-x 4 root root 520 Dec 13 16:31 ..
lrwxrwxrwx 1 root root  10 Dec 13 16:31 usb-11121119-000J7421W_USB_Camera_200901010001-event-if00 -> ../event11
lrwxrwxrwx 1 root root  10 Dec 13 16:31 usb-ELAN_Touchscreen-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 Dec 13 16:31 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dec 13 16:31 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Dec 13 16:31 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Dec 13 16:31 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0`
```

terrycain, 2016-12-14:
Thats all I need. Will add 0x020F support tonight.

laxdragon, 2016-12-14:
Awesome. I look forward to seeing it pushed up to the apt repos.

cyanogen, 2016-12-14:
Have a look at my fork for Blade Pro stuff

jd0yle, 2016-12-19:
I just tried the `feature_blade0x0F` branch on my 2016 Blade and can confirm that the keyboard is configurable with polychromatic.

terrycain, 2016-12-19:
YES

terrycain, 2016-12-19:
Am just trying to fix something else, then i'll merge and release

terrycain, 2016-12-19:
@jd0yle Mind trying out the branch again quickly? just merged in a fair few changes from master. If all is good then I'll release a new package tonight :)

jd0yle, 2016-12-19:
Only some minor issues:

I'm not sure if polychromatic is supposed to allow you to set colors/effects on a per-key basis. If it does have that functionality, its not working on my Blade. If it doesn't, then working as intended!

And there's an issue with the "Ripple" effect. When I choose it in Polychromatic, the keyboard LEDs are all set to the primary color (except the 'D' key, which is the secondary color) and they stay that color regardless of keypresses.

I rolled back to c6b3a8 just to check, and the issue exists there, too. I'm not sure if the cause is in razer-drivers or polychromatic.

Other than that, everything seems all right to me!

terrycain, 2016-12-20:
Hmm, in theory it should work but we can deal with that once I've released it.

terrycain, 2016-12-20:
@jd0yle Am releasing package soon (i.e hopefully today/thursday). After its out you able to join the telegram chat to look into per key led lighting? (its slightly easier to do in realtime rather than waiting on issue comments).

jd0yle, 2016-12-21:
@terrycain Realtime is fine with me. Telegram won't send the SMS with the code I need to create an account, so I'm waiting on a response from their support to get that sorted. If that isn't resolved soon enough, we can use something else (Google Hangouts, Slack, 2 cans and 4000 miles of string, etc.).

I'm in UTC-6 time; Any time between 17:00 UTC and 06:00 UTC should work.

terrycain, 2016-12-21:
Yeah get home from work around 1830 UTC so should be doo-able. Will probably attempt to troubleshoot this January time if thats ok with you, as have some other stuff on the go atm.

aushani, 2016-12-25:
Sorry I've been MIA, end of semester and holidays and such. Anyways, don't have much to report, just to say that I can get the drivers install and working, but can't get the polychromatic applet to connect them (and in fact when I do for some reason it seems like it boinks my keyboard and I have to plug in an external keyboard to uninstall everything). I'm on the 4.8 kernel if that changes anything.

terrycain, 2016-12-25:
Define boinks, what actually happens

aushani, 2016-12-25:
I can type on my keyboard all I want, but the key presses don't register.

terrycain, 2016-12-25:
At all? Ok that sounds like the driver got unbound from `hid-generic` and then failed to bind to `razerkbd`

Output of `cat /var/log/syslog | grep -i binding -A5 -B5` and contents of `/lib/udev/razer_mount` and `/lib/udev/rules.d/99-razer.rules`

The last 2 would probably be better on pastebin

aushani, 2016-12-25:
Yeah, at all. I'll have to try this later.. I'll have to make sure I have a spare keyboard on hand when I do it.

terrycain, 2016-12-25:
:) Also 
```
ls /sys/bus/hid/drivers
ls /sys/bus/hid/drivers/hid-generic
ls /sys/bus/hid/drivers/razerkbd
ls /sys/bus/hid/devices
```

vifino, 2017-01-29:
Hello.

My RB 14" 2016 works fine with current git (`2dacb48`) on Gentoo on kernel 4.9.

Polychromatic controls it fine too, everything seems to work except Ripple mode.
@CalcProgrammer1's KeyboardVisualizer works as well.

By the way, I have rudimentary ebuilds for Gentoo for razer-drivers and polychromatic.
They are in [my overlay](https://github.com/vifino/vifino-overlay).

Thank you for your work, @terrycain.

terrycain, 2017-01-29:
Awesome, thanks @vifino. If you can join telegram (I'm @terrycain) and we can discuss gentoo packaging if you'd like.

And yeah ripple is  a wip

closed 2017-01-29 by terrycain