openrazer/openrazer #112: USB overflows, device busy errors

cyanogen, 2016-12-14:
I'm not well versed in the actual USB protocol used by these devices yet, but I'm getting massive logspam especially when using the ripple effect.

[31877.083792] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff0000180068f428f0000068f400e0f0 .
[31877.146946] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff04001828f00028f0000068f40068f4 .
[31877.174711] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff04001828f00000e0f00068f400f088 .
[31877.176800] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff0500180068f428f0000068f400f088 .
[31877.203220] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff05001828f00000e0f00068f400f088 .
[31877.275775] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff00001800e0f000e0f00068f400f088 .
[31877.366006] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff0200180068f400f0880068f44800f0 .
[31877.394266] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff0200180068f400f0884800f00068f4 .
[31877.396494] razerkbd: Device is busy. Start Marker: 01 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff0300180068f400f0884800f00068f4 .

Usually it still works fine, but after unsuspending today it won't accept any commands at all. Going to try and debug it, but any hints are appreciated.

terrycain, 2016-12-15:
Ripple effect is just running too quickly. The devices can only handle so many requests per second.

terrycain, 2016-12-15:
Ideally We should add a wait for ?ms whenever we send a rgb frame.

CalcProgrammer1, 2016-12-20:
Is there any way you can dial the messages down?  As long as the driver doesn't crash or do anything horribly wrong when the device is busy, it's not worth spamming up dmesg over.  The official SDK lets you send as many updates as you want and it will handle getting the best possible refresh rate out of the devices for you.

I'm getting these errors a lot on my Keyboard Visualizer port.  I by default wait 50ms between updates (same as in the Windows version), and I update all devices at once.  Problem is, it seems the keyboards especially respond slowly compared to the Tartarus, Mamba TE, and Firefly.  I lower the delay to 15ms which makes the keyboard a lot more responsive, but now dmesg is full of delay warnings.  I'd say just don't print the warnings and ignore any data that comes in when the device is busy.  If it's being updated that quickly it's almost certainly because the application is trying to do smooth animation, and who cares if a frame is lost when they're going by so fast.

terrycain, 2016-12-20:
https://github.com/terrycain/razer-drivers/commit/9e72bcb1b21674e6c30f59922c8fd6acb340ed40 - Did it earlier :wink:, did some testing, seems that even though it claims its busy it still processes commands so I have deemed it as an informational message not a warning/error. Stuff should be quieter now.

closed 2016-12-20 by terrycain