openrazer/openrazer #113: Razer BlackWidow X Ultimate Support

dfrankland, 2016-12-15:
There's a pull-request for support (#110), but let's do some driver testing before we merge 👍

terrycain, 2016-12-15:
Essentially exactly the same as a Ultimate 2016 with a different form factor

terrycain, 2016-12-15:
I'll think of what we need to do on weekend

Zerophase, 2016-12-15:
After I get my Anansi driver up I want to rework the code a bit, so instead of storing the codes in an array we can just bit shift (believe we'll need bit shifting) them into a `char *`, and that way we wouldn't need as many different methods for supporting different keyboards. (There's a bunch of other clean ups I want to do as well)

terrycain, 2016-12-31:
Pasting this here for reference

*   Brightness
*   Static On / Off
*   Starlight
*   Ripple
*   Breathe
*   Reactive
*   Wave

Does it have macro or game LEDs?

dfrankland, 2016-12-31:
Yes, it has macro, game, caps lock, and num lock LEDs.

terrycain, 2017-01-04:
Have seen this, I plan on doing driver stuff round weekend.

terrycain, 2017-01-07:
What distro you on?

dfrankland, 2017-01-07:
Ubuntu Yakkety Yak

z3ntu, 2017-01-07:
And what model / product number is the keyboard (underside or on the packaging)? :)

dfrankland, 2017-01-07:
I am not in front of the keyboard right now, but I pulled the model number from Amazon: `RZ03-01760800-R3U1`

z3ntu, 2017-01-07:
That seems to be the product number of the **Razer BlackWidow X Chroma**, that's why I asked you ;)

terrycain, 2017-01-07:
When you are infront of the keyboard, Grab the debs from [here](https://drive.google.com/drive/folders/0BydYgD2LzDUCVkVyM2NPR0VrZk0?usp=sharing) install with
```
sudo dpkg -i *.deb

# It'll say add your user account to plugdev group, so do that, then do the command below

sudo apt install -f
```
Then reboot.
Then post output of 
```
ls -lah /sys/bus/hid/drivers
ls -lah /sys/bus/hid/drivers/razerkbd
ls -lah /sys/bus/hid/drivers/razerkbd/0*/
```

dfrankland, 2017-01-07:
@z3ntu, @terrycain: will do later today, and update here asap.

dfrankland, 2017-01-14:
@z3ntu, @terrycain: I had a pretty major power outage last weekend, so I am just getting to this now.

After checking the bottom of the keyboard the two numbers given are:
*   Model No. : RZ03-0176
*   Product No. : RZ03-01760800-R3U1

I am installing the drivers, daemon, etc. as I type... Will post back with output in a few minutes.

dfrankland, 2017-01-14:
@terrycain, here's the output:

```text
dylan@dylan-XPS-13-9360:~$ ls -lah /sys/bus/hid/drivers
total 0
drwxr-xr-x 8 root root 0 Jan 13 19:46 .
drwxr-xr-x 4 root root 0 Jan 13 19:45 ..
drwxr-xr-x 2 root root 0 Jan 13 19:45 hid-generic
drwxr-xr-x 2 root root 0 Jan 13 19:45 hid-multitouch
drwxr-xr-x 2 root root 0 Jan 13 19:45 logitech-djreceiver
drwxr-xr-x 2 root root 0 Jan 13 19:45 logitech-hidpp-device
drwxr-xr-x 2 root root 0 Jan 13 19:45 razerkbd
drwxr-xr-x 2 root root 0 Jan 13 19:45 wacom
dylan@dylan-XPS-13-9360:~$ ls -lah /sys/bus/hid/drivers/razerkbd
total 0
drwxr-xr-x 2 root root    0 Jan 13 19:46 .
drwxr-xr-x 8 root root    0 Jan 13 19:46 ..
lrwxrwxrwx 1 root root    0 Jan 13 19:45 0003:1532:0217.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.0/0003:1532:0217.0005
lrwxrwxrwx 1 root root    0 Jan 13 19:45 0003:1532:0217.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.1/0003:1532:0217.0006
lrwxrwxrwx 1 root root    0 Jan 13 19:45 0003:1532:0217.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.2/0003:1532:0217.0007
--w------- 1 root root 4.0K Jan 13 19:45 bind
lrwxrwxrwx 1 root root    0 Jan 13 19:46 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Jan 13 19:46 new_id
--w------- 1 root root 4.0K Jan 13 19:45 uevent
--w------- 1 root root 4.0K Jan 13 19:46 unbind
dylan@dylan-XPS-13-9360:~$ ls -lah /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:0217.0005/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 13 19:45 .
drwxr-xr-x 5 root root       0 Jan 13 19:45 ..
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 country
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 input
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 modalias
drwxr-xr-x 2 root plugdev    0 Jan 13 19:45 power
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4.0K Jan 13 19:45 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0217.0006/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 13 19:45 .
drwxr-xr-x 5 root root       0 Jan 13 19:45 ..
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 country
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 input
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 modalias
drwxr-xr-x 2 root plugdev    0 Jan 13 19:45 power
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4.0K Jan 13 19:45 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0217.0007/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 13 19:45 .
drwxr-xr-x 5 root root       0 Jan 13 19:45 ..
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 country
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:46 device_mode
-r--r----- 1 root plugdev 4.0K Jan 13 19:45 device_serial
-r--r----- 1 root plugdev 4.0K Jan 13 19:45 device_type
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4.0K Jan 13 19:45 firmware_version
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:45 game_led_state
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 13 19:45 input
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:45 macro_led_effect
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:45 macro_led_state
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:45 matrix_brightness
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_custom_frame
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_breath
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_custom
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_none
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_reactive
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_starlight
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_static
--w--w---- 1 root plugdev 4.0K Jan 13 19:45 matrix_effect_wave
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 modalias
drwxr-xr-x 2 root plugdev    0 Jan 13 19:45 power
-r--r--r-- 1 root plugdev 4.0K Jan 13 19:45 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 13 19:45 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4.0K Jan 13 19:45 test
-rw-r--r-- 1 root plugdev 4.0K Jan 13 19:45 uevent
-r--r----- 1 root plugdev 4.0K Jan 13 19:45 version
```

terrycain, 2017-01-14:
anything in `~/.razer-service/logs`

dfrankland, 2017-01-16:
Yep, just a log file:
```text
dylan@dylan-XPS-13-9360:~$ ls -la ~/.razer-service/logs
total 152
drwxr-xr-x 2 dylan dylan   4096 Dec 17 16:43 .
drwxr-xr-x 4 dylan dylan   4096 Dec 17 16:43 ..
-rw-r--r-- 1 dylan dylan 135075 Jan 15 19:39 razer.log
```

terrycain, 2017-01-16:
put that logfile on pastebin,
Also does polychromatic work? that sould work if the daemon is done right

dfrankland, 2017-01-16:
Polychromatic works very nicely, it shows up with proper name and effects (besides allowing to set the color which has no effect).

I'll be uploading the log file soon.

terrycain, 2017-01-16:
Nice, yeah thats the same with the normal ultimate2016, technically the keyboard allows rgb to be sent to it though it only acts on the green component, you can select different shades of green seperate to the brightness :wink: Wicked, ill merge this soon then :smile:

terrycain, 2017-01-16:
PPA updated.

closed 2017-01-16 by terrycain