openrazer/openrazer #117: razer driver crashes udev

DasHammett, 2016-12-18:
Hi, 

After installing the driver and rebooting, I lose access to the keyboard (is unresponsive). Even changing to another USB keyboard does not help. Booting another kernel or booting a "recovery" kernel (just loads the very basic, including udev) does not solve the issue.

I managed to uninstall the driver and I re-gain control of the keyboard, so the driver installation must be doing something my udev does not like one bit.

System: Gentoo Linux, kernel 4.9.0
udev: eudev-3.1.5 (https://wiki.gentoo.org/wiki/Eudev)
Keyboard: Razer BlackWidow Ultimate 2016

Please let me know if you require more information

terrycain, 2016-12-18:
logs from syslog/dmesg

DasHammett, 2016-12-18:
That is the issue!

I cannot see any obvious errors in dmesg or syslog. The system will boot just fine with razer-driver udev rule, but all keyboards will just not work.

terrycain, 2016-12-18:
ls /sys/bus/hid/devices
ls /sys/bus/hid/drivers

DasHammett, 2016-12-18:
ls /sys/bus/hid/devices/
`0003:046D:C07E.0001  0003:046D:C07E.0002  0003:1532:0214.0003  0003:1532:0214.0004  0003:1532:0214.0005`

ls /sys/bus/hid/drivers/hid_generic
```
0003:046D:C07E.0001  0003:1532:0214.0003  bind    uevent
0003:046D:C07E.0002  0003:1532:0214.0004  new_id  unbind
```

DasHammett, 2016-12-19:
Hi, 

I gave it another go today and somehow now it is working correctly. Please go ahead and close issue

terrycain, 2016-12-19:
Yeah if your getting the code from the master branch its not the most stable of places. If your not already, the stable branch would be slightly more stable :wink:

closed 2016-12-19 by terrycain