openrazer/openrazer #118: Razer Blade Late 2016 (GTX 1060) Support

yonilerner, 2016-12-18:
[image]

z3ntu, 2016-12-18:
`lsusb` and `ls -lah /dev/input/by-id` please

yonilerner, 2016-12-18:
`lsub`:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:020f Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`ls -lah /dev/input/by-id`:
```
total 0
drwxr-xr-x 2 root root 140 Dec 18 05:51 .
drwxr-xr-x 4 root root 480 Dec 18 05:51 ..
lrwxrwxrwx 1 root root  10 Dec 18 05:51 usb-11121119-000J8563J_USB_Camera_200901010001-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 Dec 18 05:51 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dec 18 05:51 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Dec 18 05:51 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Dec 18 05:51 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

z3ntu, 2016-12-18:
See #68

terrycain, 2016-12-18:
Is open elsewhere as @z3ntu said.  Isnt released yet.

closed 2016-12-18 by terrycain

krutoileshii, 2016-12-30:
Try running both daemon and polychromatic under root or sudo. If works then find the udev rule

yonilerner, 2016-12-30:
@krutoileshii At the time I had tried that, though Ive since returned the laptop for other reasons. Thanks though.