openrazer/openrazer #120: Kraken Chroma v1/v2 Support?

CalcProgrammer1, 2016-12-20:
Has anyone looked at these yet?  If not I can capture some data on them.  From my work on the Windows side it appears the Kraken Chroma v1 does not support custom effects (at least not via the official SDK).  The Kraken Chroma v2 does.  If you send multiple STATIC effects to the v1 it slow-fades between them.

They only have one lighting zone, which is the same color on both earcups.  Other functionality (7.1 surround, equalizer, bass boost, etc) I'm pretty sure is all done in software on the Synapse side.  The only other hardware function seems to be the ability to set the mic mute function (red LED in the mic boom) from software.  I can try to packet capture that as well.

z3ntu, 2016-12-20:
Kraken v2 is in the works and for v1 see #57. Basically when Kraken v2 is finished, Kraken v1 shouldn't be too much of a problem according to @terrycain .

terrycain, 2016-12-20:
All packet captures are welcome, especially if they're heavily documented and seperate functions per capture ;)

terrycain, 2016-12-31:
Done

closed 2016-12-31 by terrycain