openrazer/openrazer #122: Error binding OpenSuse Leap

navtek, 2016-12-21:
Hi, I'm having issues trying to use the drivers in OpenSuse, I've been following the guide but I can't bind the keyboard.

I have a BlackWidow X Chroma Keyboard and whenever I try to bind I get the following error:

`bash: bind: Permission denied`

Already installed twice, it appears to be making and installing correctly (no error messages), so I don't know where the problem might be.

Thanks in advance

z3ntu, 2016-12-21:
So you tried the rpms that are available [here](https://github.com/terrycain/razer-drivers/releases/tag/v1.1.3)? If yes, these rpms were only tested on Fedora 25. But the error sounds like a udev problem or something like that

navtek, 2016-12-22:
I installed the rpms, but still no luck, I also tried to bind using a root shell and it now says the following:

`bash: echo: write error: No such device`

When I run `ls -l /sys/bus/hid/devices/` I get the following:

`lrwxrwxrwx 1 root root 0 Dec 21 08:10 0003:03F0:134A.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/0003:03F0:134A.0001

lrwxrwxrwx 1 root root 0 Dec 21 08:10 0003:1532:0216.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.0/0003:1532:0216.0002

lrwxrwxrwx 1 root root 0 Dec 21 08:10 0003:1532:0216.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.1/0003:1532:0216.0003

lrwxrwxrwx 1 root root 0 Dec 21 08:10 0003:1532:0216.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-12/3-12:1.2/0003:1532:0216.0004`

0003:1532:0216.0004 is the interface I should be binding to.

terrycain, 2016-12-22:
What commands did you run to bind

navtek, 2016-12-22:
I ran `echo -n "0003:1532:0216.0004" > bind` inside `/sys/bus/hid/drivers/razerkbd`

z3ntu, 2016-12-22:
Did you unbind the device first?

navtek, 2016-12-22:
I couldn't unbind, the device doesn't show up when I run the following:

`cd /sys/bus/hid/drivers/hid-generic`
`ls -l`

This is the result:

`lrwxrwxrwx 1 root root    0 Dec 21 08:14 0003:03F0:134A.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11:1.0/0003:03F0:134A.0001

--w------- 1 root root 4096 Dec 21 08:14 bind

lrwxrwxrwx 1 root root    0 Dec 21 08:14 module -> ../../../../module/hid_generic

--w------- 1 root root 4096 Dec 21 08:14 new_id

--w------- 1 root root 4096 Dec 21 08:14 uevent

--w------- 1 root root 4096 Dec 21 16:20 unbind`

terrycain, 2016-12-22:
anythign in dmesg?

navtek, 2016-12-22:
I get the following

[dmesg.txt](https://github.com/terrycain/razer-drivers/files/669651/dmesg.txt)

z3ntu, 2016-12-22:
So you are using openSUSE Leap 42.2, right?

navtek, 2016-12-22:
Yes, that's right

terrycain, 2016-12-22:
Least theres no errors in dmesg, thats a start ;)

navtek, 2016-12-23:
How can I make sure it's using the right driver version?

terrycain, 2016-12-23:
latest checkout of this repo right?

navtek, 2016-12-23:
Yes, I checked out 2 days ago, we're there any essential changes since?

z3ntu, 2016-12-23:
> I installed the rpms

Yesterday I installed OpenSUSE Leap in a virtual machine just for testing and I managed to install the rpms which are available in the release section of this repo more or less fine (`dkms` package was not resolvable -> install from some 3rd party source & `python3-dbus` was not available -> `sudo pip install pydbus`) and then the driver just worked(tm).

navtek, 2016-12-23:
So I realized I only installed the kernel modules (my bad), I tried to install all other dependencies for the other rpms and I got stuck with `python3-dbus`, used `sudo pip install pydbus` but still not taking it.

z3ntu, 2016-12-23:
Binding should still work without the other packages though. You can ignore the `python3-dbus` dependency if you use `sudo pip install pydbus` to install it. Try recompiling the kernel module with `sudo dkms remove razer_chroma_driver/1.0.0 && sudo dkms install razer_chroma_driver/1.0.0` and reboot afterwards.

z3ntu, 2016-12-28:
@navtek Could you please try the packages from the obs repo [here](https://software.opensuse.org/download.html?project=home%3Az3ntu&package=razer-drivers)?

navtek, 2016-12-28:
@z3ntu Will try right now

navtek, 2016-12-28:
@z3ntu Got the driver working, when I go to `sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.0005` and trigger `cat device_type` I get `Razer BlackWidow X Chroma` but when I try to run any command I get an error, i.e. if I try to run `echo -n "1" > mode_pulsate` I get `bash: mode_pulsate: Permission denied`, already tried with sudo and root shell

navtek, 2016-12-28:
Got the driver working and now I can interact with the matrix functions, got some problems with the daemon, will check out in a bit...

z3ntu, 2016-12-28:
Please do `sudo zypper refresh && sudo zypper update` because there was a dependency missing (but it's fixed now). Hopefully everything works then :) If you have any other problems it's easier if you contact me (@z3ntu) or @terrycain on Telegram because it's much easier to communicate with people over a realtime chat.

terrycain, 2016-12-31:
Pretty sure theres packages for this now

terrycain, 2016-12-31:
Technically binding is working now, Try rpms listed under the opensuse section here - https://terrycain.github.io/razer-drivers/

closed 2016-12-31 by terrycain