openrazer/openrazer #124: Udev Rules for automatically selecting the razer-drivers driver over generic ones

simcop2387, 2016-12-22:
I've created a udev rule for myself that'll select the razerkbd driver over hid-generic.

```
# Razer BlackWindow Classic (RZ03)
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="010e", RUN{built-in}="kmod load razerkbd", RUN{program}="/bin/sh -c 'echo -n $kernel > /sys/bus/hid/drivers/hid-generic/unbind; echo -n $kernel > /sys/bus/hid/drivers/razerkbd/bind'"
```

It works by using the built-in 'kmod' to make sure the razerkbd driver is loaded, and then runs a small shell script to automatically unbind from hid-generic and then bind to razerkbd.  This should work for every device that razer-drivers supports and allow for a much better user experience.

terrycain, 2016-12-22:
you mean something like [this](https://github.com/terrycain/razer-drivers/blob/master/install_files/udev/99-razer.rules)?

simcop2387, 2016-12-22:
Oh I mised that entirely when looking through the repo.  Figures.  I'll close this then.

closed 2016-12-22 by simcop2387

terrycain, 2016-12-22:
np