openrazer/openrazer #127: Razer DeathStalker Ultimate

Paladinleeds, 2016-12-27:
Any chance this particular keyboard could be supported. Installed the drivers but keystrokes are still going missing. I plan to use this machine as a Java developer box so I need support for this.

terrycain, 2016-12-27:
Run `lsusb`. I assume no keys work at all? Most likely its udev unbinding it but failing to bind to keyboard driver.

Most likely you'll have to do [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol), if you get [telegram](https://telegram.org/) and add `@terrycain` I'll add you to the razer linux group (most of them can help out with various things like the reverse engineer link above.

Paladinleeds, 2016-12-31:
No, it isn't a case of no keys work at all. But it's random keystrokes going missing. So I'm typing away, and let's say with this message, with the normal speed I type at, on Windows it'll come out fine, responds to every keystroke as I do it, but when I try and type on Linux at the same speed, only about 80% of the keystrokes are registered.

cb22, 2017-03-01:
@Paladinleeds I had the same issue, here's a comment I posted on a blog a while back:

I was in the same situation - brand new keyboard, drops keypresses left right and center. Extremely annoying. Did some testing and made some observations:

1. Powered hubs didn't help. Tried several, USB 3, USB 2, ranging from high quality to cheap crap. Even tried with the power USB cord plugged into a 2A iPad charger, still no joy.

2. Linux / Windows didn't seem to make a difference. I didn't test this fully, but running a VirtualBox VM and passing through the USB keyboard (and using the Synapse software (rubbish that requires you to log in, etc)) the same symptoms seemed to occur.

3. Tried on different hardware - a laptop - running Linux, same dice.

4. Tried on older hardware running Windows, seemed to work (?!)

Eventually, I reached the conclusion that this keyboard can't run properly at USB 2 speeds. I stand under correction here, but it seems like on all the PCs I tested and it didn't work, they all had USB 3 (xHCI) controllers only - which provided USB 2/1.1 support.

On the computer it did work, the ports were USB 2 (eHCI / uHCI). Interested, I plugged it in to a Raspberry Pi. Sure enough, keys were missed, but as soon as I downgraded the speed to USB 1.1 (using the kernel parameter), it worked flawlessly.

As I type this, the keyboard is connected to a Raspberry Pi, and being exported via USBIP to my desktop machine, and it's working great. Not really an ideal situation, but I guess that will teach me not to buy Razor crap in the future. Hopefully I can get my hands on a cheap USB 1.1 hub, or a USB 2 PCI card and do some further testing.

z3ntu, 2017-03-01:
Serioulsy?! dafuq

z3ntu, 2017-07-18:
Any update here on how things are going?

HadesDX, 2017-08-21:
hi, does this keybord have support now? D: !?

z3ntu, 2017-08-21:
Nope. We need info and packet captures to support it.

HadesDX, 2017-08-21:
I just landed here, do you have documentation that I can check to help for this keyboard?

terrycain, 2017-08-21:
Please post the output of
`lsusb -d 1532:003e -v`
and
`ls -lah /dev/input/by-id/`
Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
Screenshots of all available options on Razer Synapse are needed.
And lastly, the product & model number of the device for documentation reasons please.

HadesDX, 2017-08-21:
ok, I will check it :D

minecrawler, 2017-09-06:
Wow. Just read [a Phoronix article](https://www.phoronix.com/vr.php?view=25140) about how awesome OpenRazer has become and immediately searched for my lovely Deathstalker Ultimate; one of the few overpriced products which is total bullcrap. And again: no support :(

However I found this issue with the exact problem I am experiencing (20% of key-events are lost), so I will give my best to supply you with what I have. Luckily, I have Windows and (Gentoo) Linux on my desktop, so I can provide you with Synapse screenshots and run the mentioned commands.

For now, I don't expect every function to work, however just getting the keyboard functionality back working would be very awesome! Ideally, the touchscreen should get some love, too, however I guess that's a bit harder, right?

Well, for now I will send you the Synapse stuff as soon as I get home tonight. I will see what I can do about the packages, but can't promise too much, since I never reversed USB before. Unfortunately, I don't have Python experience, so I won't be able to help with the implementation :/

minecrawler, 2017-09-06:
| Key | Value |
| --- | --- |
| Product | [Razer DeathStalker Ultimate](https://www2.razerzone.com/au-en/gaming-keyboards-keypads/razer-deathstalker-ultimate) |
| Model No.  | RZ03-0079, Gaming Keyboard |
| Product No. | RZ03-00790100-R3M1 |

Again, first screenshots since I am on Windows atm, will switch to Linux afterwards :)

[image]

You can click on individual keys on above layout. Clicking on one opens a new window.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Clicking the apps tab opens a new window:

[image]

It allows me to "Get More Apps":

[image]

minecrawler, 2017-09-06:
btw, PID is `0114` ;)

```
# ls -lah /dev/input/by-id/ |grep DeathStalker
lrwxrwxrwx 1 root root   9 Sep  6 19:59 usb-Razer_DeathStalker_Ultimate-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Sep  6  2017 usb-Razer_DeathStalker_Ultimate-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Sep  6  2017 usb-Razer_DeathStalker_Ultimate-if02-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Sep  6  2017 usb-Razer_DeathStalker_Ultimate-mouse -> ../mouse0
```

```
# lsusb -d 1532:0114 -v

Bus 001 Device 003: ID 1532:0114 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0114 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 DeathStalker Ultimate
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      75
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
[... 98 lines cut ...]
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

minecrawler, 2017-09-19:
@terrycain do you need any packet captures?

sedir, 2017-11-07:
Hey guys, I got a way to fix the missing keypresses by following this post: https://bbs.archlinux.org/viewtopic.php?pid=1317920#p1317920 
To sum up, you have to recompile the kernel after changing its source code _/drivers/hid/usbhid/hid-core.c_:

```
	switch (hid->collection->usage) {
	case HID_GD_MOUSE:
		if (hid_mousepoll_interval > 0)
			interval = hid_mousepoll_interval;
		break;
	case HID_GD_JOYSTICK:
		if (hid_jspoll_interval > 0)
			interval = hid_jspoll_interval;
		break;
	}
	/* Add this at line 1109~ */
	interval = 1;
```

At least now the keyboard is useable for common tasks.

terrycain, 2017-11-08:
@minecrawler Sorry completely missed this yes pcaps please
brightness slider
switchblade brightness slider
game mode on/off

static colour (change a few colours)
spectrum

We'll deal with switchblade after

Yuxael, 2018-02-12:
Any progress on those pcaps? If @minecrawler don't have the pcaps already, I'll try to get some maybe this week. I'm almost on the brink of selling my Deathstalker Utimate beacause of the issue with loosing key events and key "locking". Try working with that when using terminal on production server 😄

z3ntu, 2018-02-12:
No we still need them.

knelse, 2018-04-07:
Hey, I'll be able to provide pcaps tomorrow if you still need them. 
And, is there a way to get some basic support (without lightning, macros and other shiny stuff) just to check if missing keystrokes would be fixed? Polling interval doesn't seem to do anything for me :disappointed:

z3ntu, 2018-04-08:
I really don't think that binding the keyboard to the driver would change anything in that regard unfortunately...

knelse, 2018-04-08:
Well, kb is borderline unusable right now with my typing speed, worth a try. And I'm kinda interested in how anti-ghosting stuff is done on a driver level.

z3ntu, 2018-04-08:
So I'm guessing the keyboard works properly with Windows? Have you tried maybe FreeBSD or macOS to see if it works properly there? If it only doesn't work properly in Linux you should report it on the Linux kernel bugtracker.

terrycain, 2018-05-02:
Did we get pcaps?

Paladinleeds, 2018-05-05:
@terrycain I'm not sure. Unfortunately I no longer own this keyboard, so I cannot provide them (I have since moved to a Razer Cynosa Chroma, the UK layout option was too tempting to pass up and my DSU was falling apart anyway). Hopefully we can still get this sorted.

sedir, 2018-05-05:
I have this keyboard and I'd love to help. However I found a bit confusing on how to get pcaps at the time I read. I will give it a try again and post here.

Skote2, 2019-09-07:
Anybody still willing to work on this?
I'll happily get pcaps in here and even program a driver for the thing if I get a little help.
Your reverse engineering page looks pretty easy up to the tedious figuring out hex part.

HadesDX, 2019-09-07:
> Anybody still willing to work on this?
> I'll happily get pcaps in here and even program a driver for the thing if I get a little help.
> Your reverse engineering page looks pretty easy up to the tedious figuring out hex part.

I still use my deathstalker, i would be happy to help, im also a software developer, i might help a bit

minecrawler, 2019-09-07:
also still using DSU, it even has buggy Windows support, now. Would be great to get it working well on Linux...

Yuxael, 2019-09-10:
I tried to get pcaps but couldn't get USB passthrough to work and I gave up. Maybe I'll find some time this or the next week to try it again since I installed fresh new and updated OS.

Skote2, 2020-01-31:
I have Wireshark collecting packets. What exactly am I collecting? Is the idea legit just to flip on and off a feature a couple of times, change the settings, and save that set of packets with a label?

minecrawler, 2020-01-31:
@Skote2 exactly, that's the idea.

For Christmas, I finally rid myself of the DSU - too buggy, no numpad, problems in games, etc. I now have a Huntsman, so I will provide information for that soon.

juangea, 2020-05-17:
Do we have support for this already? 

I noticed the missing keystrokes, here is the output of the required commands:

`OUTPUT lsusb -d 1532: -v

Bus 005 Device 002: ID 1532:0114 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0114 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 DeathStalker Ultimate
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA																																																	
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      75
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         0 (Defined at Interface level)
[... 103 lines cut ...]
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

OUTPUT ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 180 may 17 14:32 .
drwxr-xr-x 4 root root 540 may 17 14:32 ..
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Razer_DeathStalker_Ultimate-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Razer_DeathStalker_Ultimate-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Razer_DeathStalker_Ultimate-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Razer_DeathStalker_Ultimate-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 may 17 14:32 usb-Razer_DeathStalker_Ultimate-mouse -> ../mouse0
`

64-Tesseract, 2020-12-30:
I'm also waiting on support for this keyboard... I'm not too fussed about the switcblade, it'd be pretty hard to code for such a niche thing, but it'd be nice to be able to bind the macro keys to do _something_ at least.

z3ntu, 2025-08-30:
@SilverDante I guess not, just nobody has added support for this yet. Basic support should be similar to most other keyboards, if you ignore this special screen portion of the keyboard.

z3ntu, 2025-08-31:
@minecrawler Support without the screen, as mentioned in the previous comment, shouldn't be too difficult and is well in-scope for OpenRazer. But I think for screen control, a project outside of OpenRazer would be good if someone really wants to support this, given this kind of screen is not present in any keyboard released within many years - as far as I know.