openrazer/openrazer #128: Razer Tartarus Chroma stops working after some time

duapruis, 2016-12-28:
I've noticed that when I try to use my Tartarus Chroma everything works great for some time but after a few minutes the device stops working and it either doesn't respond or starts repeating the last pressed key over and over, for example, if the last pressed key was spacebar it will continuously spam that key endlessly or it doesn't even work. I don't know if it is a driver issue or what but... just wondering.

terrycain, 2016-12-29:
Can you reproduce it reliably?
Anything in ~/.razer-service/logs?

duapruis, 2016-12-29:
This is the only error that I can see on the log:

2016-12-29 19:09:49 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/screensaver_thread.py", line 123, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NotSupported: This method is not implemented

terrycain, 2016-12-29:
That's a separate error which has been fixed but not released yet.

I take it without the driver the tartarus just works normally but without any real use?

terrycain, 2016-12-29:
Also what distro you running?, next time it stops working run `pkill razer-service` and then see if it spits out a stacktrace to either the log or syslog/journalctl

terrycain, 2017-01-01:
So it's working now?

duapruis, 2017-01-01:
Apparently yes, it has worked for 2 days.

terrycain, 2017-01-01:
Ok I deem this as magically fixed. Depending on what version you were on and other environmental factors, it could of been anything.

closed 2017-01-01 by terrycain