openrazer/openrazer #135: [Arch] Devices not discovered in log or peripheral utilities

Sclarki, 2016-12-30:
Everything was working smoothly until updating the packages (razer-daemon, python-razer, and I believe the two razercommander and polychromatic utilities?). Now things be borked. The daemon runs[0], but the logs do not show devices being discovered like usual. If this is related to the #67 problem then just close this issue as I've already tried that workaround to no avail.

Output of `uname -a`:

```Linux sharkbark 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64 GNU/Linux```

Razer packages installed:

```
polychromatic         0.3.5-1
python-razer          1.1.3-1
razer-daemon          1.1.3-1
razer-driver-dkms     1.1.3-1
razer-driver-meta     1.1.2-1
razercommander-git    1.0.3-8
```

The output from the log after trying the possible [Ubuntu fix](https://github.com/terrycain/razer-drivers/issues/67#issuecomment-260419314) (no different than starting it normally):

```
2016-12-29 17:56:01 | razer                          | INFO     | Initialising Daemon (v1.1.3). Pid: 2302
2016-12-29 17:56:01 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-12-29 17:56:01 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-12-29 17:56:01 | razer                          | INFO     | Serving DBus
2016-12-29 17:56:03 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-29 17:56:03 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-12-29 17:58:27 | razer                          | DEBUG    | DBus called get_serial_list
```

**[0]**: It probably isn't related, but one problem I ran into upon rebooting today was `python-pyudev` was not installed on my system. I have since installed it and that's when I ran across the issue of nothing working. Prior to the reboot -- but after updating these packages -- things were working.

JohnTheScout, 2016-12-30:
Same issue here. Also on Arch. Binding/unbinding works, manually echoing values works to control the keyboard but the daemon, razer commander and polychromatic all fail to see any devices connected. Also tried the workaround and no dice.

terrycain, 2016-12-30:
@z3ntu think they hit the same udev chmod +x bug

z3ntu, 2016-12-30:
Yeah I'll talk with gab about the Arch packages because they are in a really bad form right now... Hopefully AUR admins can help us a bit for making it better.

terrycain, 2016-12-31:
"Should" be fixed now

Sclarki, 2017-01-02:
It is working again! Many thanks.

closed 2017-01-02 by Sclarki