openrazer/openrazer #136: AsyncIO

terrycain, 2016-12-31:
Have opened an issue for the PR https://github.com/terrycain/razer-drivers/pull/115

Merged PR into - https://github.com/terrycain/razer-drivers/tree/cyanogen-async

Have made debian/ubunut packages for gbulb and python3-evdev 0.6.4

z3ntu, 2016-12-31:
* Fedora packages exist for python3-gbulb and python3-evdev.
* Packages don't build for openSUSE (all versions)
* The AUR has both.

terrycain, 2016-12-31:
Hmm might just be easier to bodge the opensuse ones if we need to

z3ntu, 2016-12-31:
Got [gbulb](https://build.opensuse.org/package/show/home:z3ntu/python3-gbulb) and [evdev](https://build.opensuse.org/package/show/home:z3ntu/python-evdev) building for Fedora & openSUSE :)

terrycain, 2017-01-01:
Nice

cyanogen, 2017-01-18:
Have the 3.4 stuff working, check my master branch. Queue is getting kinda deep :) I'll open a ticket in the next day or two to explain the new stuff!

terrycain, 2017-01-18:
Will have a look, all ive noticed from my own dabbling in asyncio is that it all works great apart from the ripple effect which just lags

cyanogen, 2017-01-20:
Rebased my branch onto master and squashed down a lot of the changes.

terrycain, 2017-01-20:
Am busy but will try and get a look at it this weekend

closed 2018-02-28 by stale[bot]