openrazer/openrazer #137: Diamonback Chroma && Deathstalker Chroma branch

terrycain, 2016-12-31:
Merged [PR](https://github.com/terrycain/razer-drivers/pull/121) into branch [CalcProgrammer1-diamondback_deathstalker_test](https://github.com/terrycain/razer-drivers/tree/CalcProgrammer1-diamondback_deathstalker_test)
@CalcProgrammer1 - have merged PR into branch, 

Copied this here for reference
>I have tested the Diamondback and the DeathStalker Chroma.
>
>The Diamondback works just like the Mamba TE except with a 21 LED matrix rather than a 16 LED matrix. The LED array column 0 is the top right LED and continues down and around and up the left side to LED array column 18. Column 19 is the logo LED and column 20 is the scroll wheel LED.
>
>The DeathStalker does not have per-key backlighting and functions more like the mice and Firefly than the other keyboards. I discovered that despite only having 6 zones exposed in the official Chroma SDK, it really has 12 controllable LEDs with every two LEDs blended together to light one zone. If you set them to two very different colors, you will get a mess. It can be used to make smoother gradients though. Treat it like a Firefly with 12 columns when using custom effects, otherwise it works like the other keyboards.
>
>IMPLEMENTED: Addition of these two devices to driver, changes to get custom effects working with these devices
>
>TESTED: All lighting modes except Reactive including custom effects, serial number, device type, firmware version requests, actual usage as keyboard/mouse, game mode LED, macro mode LED and macro mode LED effect
>
>NOT TESTED: Macros, key remaps, game mode, reactive effects, DPI changes, any other special non-lighting related stuff
>
>NOT IMPLEMENTED: Daemon/application layer stuff

terrycain, 2016-12-31:
I have merged the driver and resolved the conflicts. Would you mind checking out `CalcProgrammer1-diamondback_deathstalker_test` and testing that, once thats done we can go onto testing the parts you havent tested yet.

CalcProgrammer1, 2016-12-31:
I'm out of town for the week and didn't bring my DeathStalker with me.  I do have my Diamondback.  I can test next weekend for sure with both devices.

terrycain, 2017-01-01:
Awesome :smile:

usineur, 2017-01-08:
Hi. This branch seems deleted now. Is there any reason for that ?

terrycain, 2017-01-08:
ooh crap I think I might have deleted it by accident. lemme go and find it again

terrycain, 2017-01-08:
Its back again

terrycain, 2017-01-10:
Have pushed daemon support to said branch, can someone test so i can merge it.

usineur, 2017-01-10:
Yes, I will give it a try

terrycain, 2017-01-16:
PPA updated.

closed 2017-01-16 by terrycain