openrazer/openrazer #138: Razer mouse scripts needs update

MrBlaise, 2017-01-01:
Hey!

I have a DeathAdder Chroma and I wanted to customize the DPI, sadly neither of the frontends support my mouse, or if it does it doesn't work properly, forgets my settings etc.

So I wanted to change my dpi directly using the set_mouse_dpi script, but I needed to modify it because the write only file in the driver is now called simply "dpi" instead of "set_mouse_dpi". Maybe this is mouse specific, but it wasn't mentioned in the wiki. Also I've noticed that some other names changed as well.

terrycain, 2017-01-01:
Does the daemon work for you? As changing the dpi through that would be better than doing things directly as the idea is I will change the drivers interface at will as one is "supposed" to use the daemon as an abstraction layer.

Am away from pc but can give you some examples late today/tomorrow.

Basically I've only just added dpi support to the python library so it'll be a short time until it's supported in client apps

z3ntu, 2017-01-01:
Yeah, the wiki pages aren't up-to-date. The [feature_docs](https://github.com/terrycain/razer-drivers/tree/feature_docs) does contain documentation for the python library if you are interested.

terrycain, 2017-01-01:
@lah7 ^^

terrycain, 2017-01-01:
That docs isn't complete yet but I do think it has a dpi example somewhere

MrBlaise, 2017-01-01:
@z3ntu @terrycain Thanks the python library looks good I'll try it out. This direct solution felt very "hacky" indeed, and I couldn't read the dpi only set it which was an inconvenience, but I can see that the python library makes this very easy. Thanks :)

terrycain, 2017-01-01:
If you got newest driver, reading dpi "should" work though it only works if the device supports it, though i think all should support reading dpi

z3ntu, 2017-02-18:
@MrBlaise Can this be closed?

MrBlaise, 2017-02-18:
Yes @z3ntu thanks :)

closed 2017-02-18 by terrycain