openrazer/openrazer #139: Razer Naga Support

rquast, 2017-01-02:
Just support for the plain Naga mouse. :)
Bus 003 Device 005: ID 1532:0015 Razer USA, Ltd Naga Mouse

z3ntu, 2017-01-02:
Please post the output of
`lsusb -d 1532:0015 -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

rquast, 2017-01-03:
[naga_lsusb.txt](https://github.com/terrycain/razer-drivers/files/681654/naga_lsusb.txt)

ls -lah /dev/input/by-id/
```
lrwxrwxrwx 1 root root  10 Jan  3 09:02 usb-27C40-06N20-L01S4AE1TZ_HD_WebCam-event-if00 -> ../event20
lrwxrwxrwx 1 root root  10 Jan  3 09:02 usb-E-Signal_Keyboard-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Jan  3 09:02 usb-E-Signal_Keyboard-if01-event-mouse -> ../event13
lrwxrwxrwx 1 root root   6 Jan  3 09:02 usb-E-Signal_Keyboard-if01-mouse -> ../js0
lrwxrwxrwx 1 root root  10 Jan  3 09:02 usb-Razer_Razer_Naga-event-mouse -> ../event10
lrwxrwxrwx 1 root root  10 Jan  3 09:02 usb-Razer_Razer_Naga-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 Jan  3 09:02 usb-Razer_Razer_Naga-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan  3 09:02 usb-Samson_Technologies_Samson_GoMic-event-if03 -> ../event7
lrwxrwxrwx 1 root root   9 Jan  3 09:02 usb-Thermaltake_Tt_eSPORTS_KNUCKER_Gaming_Keyboard-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jan  3 09:02 usb-Thermaltake_Tt_eSPORTS_KNUCKER_Gaming_Keyboard-if01-event-kbd -> ../event9
```

rquast, 2017-01-03:
Will take a read of the packet capture and see if I can get it sorted out :)

z3ntu, 2017-01-06:
What's the model / product number which is on the bottom of the mouse and on the packaging? It should look something like `RZ01-XXXX`

rquast, 2017-01-06:
@z3ntu RZ01-0028(B) - Product Number: RZ01-00280500

z3ntu, 2017-02-27:
Could you please record some packet captures from Windows & Synapse with the different lighting effects?

jeferod83, 2018-10-08:
Hi @z3ntu !

I'll try to take this packets. See ya!

jeferod83, 2018-10-09:
Hi @z3ntu 
I've taken all the light effects available from this mice.
It only controls the wheel and logo lights, both on and off.
I've attached bellow the tar balled pcapng's.
If I made a mistake, please correct me.

Thanks a lot!

jeferod83, 2018-10-09:
[NagaMolten_LightsControl.tar.gz](https://github.com/openrazer/openrazer/files/2458597/NagaMolten_LightsControl.tar.gz)

jeferod83, 2018-11-15:
Hi @z3ntu 
Do you have any idea when this driver will included on the openrazer core?

Tks!

wbclark, 2018-11-24:
Hi @z3ntu 

Please let me know if there's any way to assist with getting the needed data, implementing, or testing this feature. I have a Naga Molten Edition and machines running RHEL, Fedora, and Ubuntu. I am eager to help!

Cheers

jeferod83, 2018-12-09:
Hi @z3ntu! I'm too. If there is a way that can I help you to make this driver works, call me, message me, send a fog signal to me please =D

z3ntu, 2018-12-16:
@jeferod83 Please message me on Telegram (@z3ntu), Matrix (@z3ntu:matrix.org), IRC (z3ntu on freenode)

RhoLouh, 2019-01-23:
Hi!
How is this topic going?

wbclark, 2019-08-06:
Hi @z3ntu , anything we can do to assist with this? What work needs done currently?

drizzit56, 2020-06-10:
Hey I see there has been an update in May for this I have this model is there anything I can do to help or any more information required? Would love to have this working also much like @AccaGihu the lighting does not concern me I am more interested in having profiles and macro support for work/gaming. Thanks for al your work on this project @z3ntu and others I'm sure :smile: !

jthomas281, 2020-07-04:
I have the same mouse and I've kept it in good repair for many years now. Would like to see it work with openrazer.

AccaGihu, 2020-07-20:
> I have the same mouse and I've kept it in good repair for many years now. Would like to see it work with openrazer.

Same as me, I've replaced both the LMB and RMB at different times when it would doubleclick and this mouse is still in top condition. In my eyes it was a great purchase. Looks like since it's no more on the shelves it doesn't get much love anymore, and no recognition either sadly.
I'm on manjaro, if I can help I'll gladly do so.

ilvipero, 2020-09-15:
I just bought a razer naga trinity and it works fine out of the box, all the side buttons do work, polychromatic allows me to change colors, dpi, and polling rate. No profiles,  but I have seen on main page that mouse and mats profiles will come at a later release, so I am very happy. https://github.com/polychromatic/polychromatic
If any help is needed to test experimental builds, I am happy to do my part.

GrbavaCigla, 2020-11-24:
I am using Gentoo linux and Windows. I have the same mouse. When will this mouse be added?

damianatorrpm, 2020-11-24:
@GrbavaCigla
I also have the naga (with the 3 side panels). It works perfectly fine (the buttons are mapped to ctrl + button_number) without openrazer, which is used to control the lightning effects. Since it relies on kernel stuff and I'm on Silverblue w/ secureboot I no longer use openrazer (the mouse works fine!).

GrbavaCigla, 2020-11-24:
For me it doesn't work. Ctrl+button_number doesn't do anything. I want to map keys on the side and I want to change the DPI. I am using custom kernel.

Martan404, 2023-06-18:
Any update on this?

z3ntu, 2024-10-02:
Hi all! Anybody up for testing #2312 which adds support for this device? See https://github.com/openrazer/openrazer/wiki/Building-a-package how to build from source.

closed 2024-10-06 by z3ntu

0xFADDAD, 2024-10-10:
Doesn't seem to like the original Naga, see below.
https://github.com/openrazer/openrazer/pull/2312#issuecomment-2403733411

PR #2312 (2024-10-06): Support for Razer Naga (1532:0015)
Initial implementation for Razer Naga (1532:0015).
Fixes #139
As far as I can tell, the Naga and the Naga Molten Edition has the same pid.
I have only tested this on the Molten edition (which has a pid of 0015), as I don't have an original Naga.
I used an image of the molten edition in the daemon because I couldn't find a transparent image of the original.
Setting the Scroll light also turns the side buttons to the same state, I have no idea how to fix this.

Features :
- Set Lighting Logo state
- Set Lighting Scroll state
- Set Lighting Backlight state
- Set [... cut]