openrazer/openrazer #141: Razer Orochi ( 2011 year )

theonedemon, 2017-01-03:
Not support old mouse:
`
Bus 002 Device 004: ID 1532:0013 Razer USA, Ltd Orochi mouse
`

terrycain, 2017-01-03:
https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

z3ntu, 2017-02-27:
@theonedemon Can you please provide screenshots and wireshark captures of the usb traffic sent by Synapse (see link from Terry)?

theonedemon, 2017-06-20:
Sorry for the delayed response
I made dump via wireshark: https://www.dropbox.com/sh/915jxsyvfb6wecz/AADoOjFw5ZS8qhDhycUgorV5a?dl=0

terrycain, 2017-06-25:
Also need `ls -lah /dev/input/by-id`

terrycain, 2017-06-25:
Also what OS are you on

theonedemon, 2017-06-25:
> Also need ls -lah /dev/input/by-id
lrwxrwxrwx 1 root root   9 Июн 23 20:34 usb-046d_081b_F5473740-event-if00 -> ../event7
lrwxrwxrwx 1 root root   9 Июн 23 20:34 usb-Logitech_USB_Receiver-if02-event-kbd -> ../event2
lrwxrwxrwx 1 root root  10 Июн 25 15:31 usb-Razer_Razer_Orochi-event-mouse -> ../event13
lrwxrwxrwx 1 root root  10 Июн 25 15:31 usb-Razer_Razer_Orochi-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Июн 25 15:31 usb-Razer_Razer_Orochi-mouse -> ../mouse1

> Also what distro are you on
`Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
`

terrycain, 2017-06-25:
install the debs from here
https://drive.google.com/open?id=0BydYgD2LzDUCZnplQU43Ti1Ibms
```
sudo dpkg -i debname*
sudo apt-get install -f
```

Reboot
then do
```
lsmod | grep razer
ls -lah /sys/bus/hid/drivers
```

theonedemon, 2017-06-25:
> lsmod | grep razer

`
razermouse             61440  0
hid                   122880  7 razermouse,hid_generic,usbhid,hidp,hid_logitech_dj,hid_logitech_hidpp
`
> ls -lah /sys/bus/hid/drivers

`
drwxr-xr-x 2 root root 0 Jun 25 18:58 hid-generic
drwxr-xr-x 2 root root 0 Jun 25 18:58 logitech-djreceiver
drwxr-xr-x 2 root root 0 Jun 25 18:58 logitech-hidpp-device
drwxr-xr-x 2 root root 0 Jun 25 18:58 razermouse

`

terrycain, 2017-06-25:
ls -lah /sys/bus/hid/drivers/razermouse

theonedemon, 2017-06-25:
```
total 0
drwxr-xr-x 2 root root    0 Jun 25 18:58 .
drwxr-xr-x 6 root root    0 Jun 25 18:58 ..
lrwxrwxrwx 1 root root    0 Jun 25 19:08 0003:1532:0013.0006 -> ../../../../devices/pci0000:00/0000:00:12.2/usb1/1-5/1-5.2/1-5.2:1.0/0003:1532:0013.0006
lrwxrwxrwx 1 root root    0 Jun 25 19:08 0003:1532:0013.0007 -> ../../../../devices/pci0000:00/0000:00:12.2/usb1/1-5/1-5.2/1-5.2:1.1/0003:1532:0013.0007
--w------- 1 root root 4.0K Jun 25 18:56 bind
lrwxrwxrwx 1 root root    0 Jun 25 19:08 module -> ../../../../module/razermouse
--w------- 1 root root 4.0K Jun 25 19:08 new_id
--w------- 1 root root 4.0K Jun 25 19:08 uevent
--w------- 1 root root 4.0K Jun 25 19:08 unbind
```

z3ntu, 2017-06-25:
`ls -la /sys/bus/hid/drivers/razermouse/0*/`

theonedemon, 2017-06-25:
> ls -la /sys/bus/hid/drivers/razermouse/0*/

```
/sys/bus/hid/drivers/razermouse/0003:1532:0013.0006/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 25 18:56 .
drwxr-xr-x 5 root root       0 Jun 25 19:11 ..
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 country
-rw-rw---- 1 root plugdev 4096 Jun 25 18:56 device_mode
-r--r----- 1 root plugdev 4096 Jun 25 18:56 device_serial
-r--r----- 1 root plugdev 4096 Jun 25 18:56 device_type
lrwxrwxrwx 1 root plugdev    0 Jun 25 18:56 driver -> ../../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 Jun 25 18:56 firmware_version
drwxr-xr-x 3 root plugdev    0 Jun 25 18:56 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 25 18:56 input
-rw-rw---- 1 root plugdev 4096 Jun 25 18:56 logo_led_state
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 modalias
drwxr-xr-x 2 root plugdev    0 Jun 25 18:56 power
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 report_descriptor
-rw-rw---- 1 root plugdev 4096 Jun 25 18:56 scroll_led_state
lrwxrwxrwx 1 root plugdev    0 Jun 25 18:56 subsystem -> ../../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 Jun 25 18:56 test
-rw-r--r-- 1 root plugdev 4096 Jun 25 18:56 uevent
-r--r----- 1 root plugdev 4096 Jun 25 18:56 version

/sys/bus/hid/drivers/razermouse/0003:1532:0013.0007/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 25 18:56 .
drwxr-xr-x 5 root root       0 Jun 25 19:11 ..
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 country
lrwxrwxrwx 1 root plugdev    0 Jun 25 18:56 driver -> ../../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Jun 25 18:56 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 25 18:56 input
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 modalias
drwxr-xr-x 2 root plugdev    0 Jun 25 18:56 power
-r--r--r-- 1 root plugdev 4096 Jun 25 18:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 25 18:56 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jun 25 18:56 uevent

```

z3ntu, 2017-06-25:
`cat /sys/bus/hid/drivers/razermouse/0*/{device_type,device_serial,firmware_version,device_mode,logo_led_state,scroll_led_state}`

theonedemon, 2017-06-25:
> cat /sys/bus/hid/drivers/razermouse/0*/{device_type,device_serial,firmware_version,device_mode,logo_led_state,scroll_led_state}

```
Razer Orochi 2011
PM000767010762
v9.99
0:0
0
0
```

z3ntu, 2017-06-25:
```
cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)
echo -n '1' > logo_led_state
echo -n '1' > scroll_led_state
cat logo_led_state
cat scroll_led_state
echo -n '0' > logo_led_state
echo -n '0' > scroll_led_state
cat logo_led_state
cat scroll_led_state
```

terrycain, 2017-06-25:
Run those commands in batches of 2 as they will change leds. If you run them all at once it'll just look like the leds are off.

If it works that is

theonedemon, 2017-06-25:
Output:
1
1
0
0

logo_led_state - control battery indicator
scroll_led_state - control scroll led

It's works!

theonedemon, 2017-06-25:
please add support to polychromatic-controller

z3ntu, 2017-06-25:
Yes, daemon support will come soon 😃

terrycain, 2017-06-25:
Install these debs, reboot https://drive.google.com/open?id=0BydYgD2LzDUCZnplQU43Ti1Ibms

Then see if it appears in polychromatic

theonedemon, 2017-06-25:
> Then see if it appears in polychromatic
thanks, mouse connected in polychromatic, but toogle indicator block works incorrect:

[image]

1. Settings:
[image]
1.1 Scrool whell led - off
1.2 Power indicator - off

2. Settings:
[image]
2.1 Scrool whell led - off
2.2 Power indicator - on

3. Settings:
[image]
2.1 Scrool whell led - on
2.2 Power indicator - on

4. Settings:
[image]
2.1 Scrool whell led - off
2.2 Power indicator - off

Please tell me what the program can be change bind key in mouse or change sensitivity ( DPI ) ?

terrycain, 2017-06-25:
I dont have pcaps or images for dpi in synapse

theonedemon, 2017-06-25:
1. I upload pcap change "sensitivity" and "Polling Rate"
https://www.dropbox.com/sh/915jxsyvfb6wecz/AADoOjFw5ZS8qhDhycUgorV5a?dl=0

2. Customize bind dont logging in wireshark

terrycain, 2017-06-25:
Customised binds are done in software

terrycain, 2017-06-25:
Install these debs and reboot
https://drive.google.com/open?id=0BydYgD2LzDUCZnplQU43Ti1Ibms
DPI "might" work

theonedemon, 2017-06-26:
i installed last debs:
[image]

but, change DPI not changed speed mouse cursor =( 

after restart daemon reset settings:
[image]

@terrycain you can write to me when you updated remotes/origin/feature_orochi_2011_0x0013
I can compile the drivers myself, this should speed up debugging

theonedemon, 2017-06-26:
@terrycain please push last commit with dpi

theonedemon, 2017-06-28:
i tested drivers 2 day, dpi and pooling_rate change incorrect:
`cat dpi` show:
```
76:76
```
change dpi: 
```
echo -n '500' > dpi 
#or
echo -n '500:500' > dpi
```
dont change real dpi in mouse and `echo dpi` always show 76:76

`cat poll_rate` show always =  0

I open my pcap ( 01_polling_rate_125.pcapng , 01_polling_rate_500.pcapng , 01_polling_rate_1000.pcapng ) and found second diffs except:
[image]

terrycain, 2017-06-28:
try 
`echo -ne '\x4c' > dpi`
or 
`echo -ne '\x9c' > dpi`

terrycain, 2017-06-28:
poll rate is fine, probs an issue with storing the value. first difference is poll rate, 2nd is a crc. As your mouse doesnt support querying the dpi/poll rate i have to store it (which gets lost on reboot) and there probably a line missing to save it after sending it to the device

theonedemon, 2017-06-28:
change dpi via console work 
`echo -ne '\x4c' > dpi` output ( cat dpi  ): 76:76
`echo -ne '\x9c' > dpi` output: 156:156
`echo -ne '\x1c' > dpi` output: 28:28

in polychromatic-controller settings not working:
[image]

terrycain, 2017-06-28:
Does the dpi changes from terminal actually change dpi though?

theonedemon, 2017-06-28:
yes, in terminal dpi change and no save dpi after reboot

terrycain, 2017-06-28:
Yeah the not saving dpi after reboot is a hardware thing (which synapse hides in software), newer mice do save said dpi.

Ok im not in the mood tonight, but remind me on the weekned and i'll have a look, i have a feeling i know what it is.

theonedemon, 2017-06-28:
ok, thank you.

terrycain, 2017-07-01:
get latest change from the branch.
Run python3 and then try this
```
from razer.client import DeviceManager
a = DeviceManager()
a.devices

b = a.devices[0]
b.dpi
```
Paste the output.

theonedemon, 2017-07-01:
```
>>> from razer.client import DeviceManager
>>> a = DeviceManager()
>>> a.devices
[<RazerMouse PM004209829121>]
>>> b = a.devices[0]
>>> b.dpi
(2011, 2011)
>>> 
```

terrycain, 2017-07-01:
ok continuing with that above try
```
b.dpi = 1000, 1000
b.dpi
```
Does that change the dpi?

theonedemon, 2017-07-01:
```
>>> b.dpi = 1000, 1000
>>> b.dpi
(979, 979)
>>> 
```

dpi has changed

terrycain, 2017-07-01:
ok, therefore, the python library is working, (reason the numbers arnt the same is due to fitting a value > 255 into 1 byte). If applications like polychromatic dont work with dpi, that an application issue. As far as I'm concerned the 2011 is supported :smile:. Will release this weekend and then it should be available in package form

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain