openrazer/openrazer #142: Missing dependency in Ubuntu package

christianhujer, 2017-01-04:
I have a Razer Blade Stealth (late 2016, Kaby Lake).
I am using Kubuntu 16.10.
I have just installed the razer-drivers by following the instructions given on https://terrycain.github.io/razer-drivers/#download

```
sudo add-apt-repository ppa:terrz/razerutils
sudo apt update
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo modprobe razerkbd
```

When starting `razer-service`, I get the following error message:

```
christian.hujer@christian-hujer-Blade-Stealth:~$ razer-service 
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 7, in <module>
    from razer_daemon.daemon import daemonize
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 20, in <module>
    gi.require_version('Gdk', '3.0')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 118, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Gdk not available
```

It seems that a dependency is missing.

terrycain, 2017-01-04:
You know what package satisfies that?

christianhujer, 2017-01-04:
I also tried install whatever was mentioned in https://terrycain.github.io/razer-drivers/#other that didn't help either. I don't know what package satisfies that, still trying to figure that out.

christianhujer, 2017-01-04:
I don't know, maybe in case you have a debian / ubuntu system, could you check where the Gdk library for Python is and check which package owns that file?

terrycain, 2017-01-04:
Will have a look when I get home

z3ntu, 2017-01-04:
Try installing `gtk3` (or whatever it is called)

terrycain, 2017-01-04:
Try installing `gir1.2-gtk-3.0`

christianhujer, 2017-01-06:
Installing gir1.2-gtk3.0 solved the problem. That's missing dependency. `razer-service` now starts fine.

closed 2017-01-06 by terrycain

z3ntu, 2017-01-17:
See #160

commit bd55c826 (2017-01-06): Added missing dependancy, closes #142