openrazer/openrazer #144: Unable to get Orochi (RZ01-0082) bind to the driver

m13253, 2017-01-06:
### First I found `0039` is missing from `/usr/lib/udev/rules.d/99-razer.rules`

I tried to add it manually, but resulted in the mouse completely not movable after plugging in.

### Second I tried to call `razer_mount` manually

```
$ sudo /usr/lib/udev/razer_mount razermouse 0003:1532:0039.0017
Driver razermouse
Device_ID 0003:1532:0039.0017
Unbinding 0003:1532:0039.0017 from hid-generic
Binding 0003:1532:0039.0017 to razermouse
/usr/lib/udev/razer_mount: line 50: printf: write error: No such device
```

After that the mouse is also not movable, until I reconnect it.

Refer to your webpage, I think Orochi should be supported. What's wrong with my configuration?

### My environment is ArchLinux, using libinput driver for Gnome on Xorg
```
$ uname -a
Linux brilliant-laptop 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64 GNU/Linux

$ dmesg
[ 1477.890264] usb 4-1.2.2: new full-speed USB device number 17 using ehci-pci
[ 1477.994297] input: Razer  Razer Orochi 2013 as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.0/0003:1532:0039.0017/input/input36
[ 1477.994691] hid-generic 0003:1532:0039.0017: input,hidraw3: USB HID v1.11 Mouse [Razer  Razer Orochi 2013] on usb-0000:00:1d.0-1.2.2/input0
[ 1477.996463] input: Razer  Razer Orochi 2013 as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.1/0003:1532:0039.0018/input/input37
[ 1478.050422] hid-generic 0003:1532:0039.0018: input,hidraw4: USB HID v1.11 Keyboard [Razer  Razer Orochi 2013] on usb-0000:00:1d.0-1.2.2/input1

$ lsusb | grep Razer
Bus 004 Device 018: ID 1532:0039 Razer USA, Ltd 

$ libinput-list-devices
Device:           Razer  Razer Orochi 2013
Kernel:           /dev/input/event10
Group:            8
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabledit
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a
```

Thank you!

z3ntu, 2017-01-06:
The device isn't yet supported (there are different models of the Orochi, yours seems to be the 2013 edition).
Please post the output of
`lsusb -d 1532:0039 -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

m13253, 2017-01-06:
```
$ lsusb -d 1532:0039 -v

Bus 004 Device 007: ID 1532:0039 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0039 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      87
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               8
```

```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 Jan  6 16:00 .
drwxr-xr-x 4 root root 520 Jan  6 16:00 ..
lrwxrwxrwx 1 root root  10 Jan  6 16:00 usb-AAB7L02WL_1.3M_HD_WebCam-event-if00 -> ../event18
lrwxrwxrwx 1 root root   9 Jan  6 16:00 usb-C-Media_Electronics_Inc._USB_Audio_Device-event-if03 -> ../event7
lrwxrwxrwx 1 root root   9 Jan  6 16:00 usb-Heng_Yu_Technology_Poker_II-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Jan  6 16:00 usb-Heng_Yu_Technology_Poker_II-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Jan  6 16:00 usb-Razer_Razer_Orochi_2013-event-mouse -> ../event10
lrwxrwxrwx 1 root root  10 Jan  6 16:00 usb-Razer_Razer_Orochi_2013-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 Jan  6 16:00 usb-Razer_Razer_Orochi_2013-mouse -> ../mouse0
```

I will post the screenshots tomorrow when I find a Windows machine.

m13253, 2017-01-07:
### Buttons
[image]
[image]
[image]

### DPI settings
[image]
[image]

### Lighting
[image]

terrycain, 2017-01-07:
@z3ntu Can you help @m13253 installing the driver from the branch `feature_orochi_2013_0x0039`

In theory all "should" work driver daemon etc.. but we just need to get to it being mounted for now.

z3ntu, 2017-01-07:
Will post the PKGBUILD shortly

m13253, 2017-01-08:
Now I got the mouse mounted. Thank you!

But I got 2 problems:

1. `scripts/razer_mouse/driver/set_mouse_dpi.py` is not working, because `set_mouse_dpi` is missing.

```
$ sudo ./set_mouse_dpi.py  -d 0003:1532:0039.0009 --dpi_x 500
Traceback (most recent call last):
  File "set_mouse_dpi.py", line 66, in <module>
    run()
  File "set_mouse_dpi.py", line 59, in run
    with open(set_mouse_dpi_filepath, 'wb') as set_mouse_dpi_file:
PermissionError: [Errno 13] Permission denied: '/sys/bus/hid/drivers/razermouse/0003:1532:0039.0009/set_mouse_dpi'
```

I found `set_mouse_dpi` is not present:
```
$ls -lah /sys/bus/hid/drivers/razermouse/{,0003:1532:0039.0009,0003:1532:0039.000A}
lrwxrwxrwx 1 root root 0 Jan  8 19:01 /sys/bus/hid/drivers/razermouse/0003:1532:0039.0009 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.0/0003:1532:0039.0009
lrwxrwxrwx 1 root root 0 Jan  8 19:01 /sys/bus/hid/drivers/razermouse/0003:1532:0039.000A -> ../../../../devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.1/0003:1532:0039.000A

/sys/bus/hid/drivers/razermouse/:
total 0
drwxr-xr-x 2 root root    0 Jan  8 19:01 .
drwxr-xr-x 4 root root    0 Jan  8 18:54 ..
lrwxrwxrwx 1 root root    0 Jan  8 19:01 0003:1532:0039.0009 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.0/0003:1532:0039.0009
lrwxrwxrwx 1 root root    0 Jan  8 19:01 0003:1532:0039.000A -> ../../../../devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.1/0003:1532:0039.000A
--w------- 1 root root 4.0K Jan  8 19:01 bind
lrwxrwxrwx 1 root root    0 Jan  8 18:54 module -> ../../../../module/razermouse
--w------- 1 root root 4.0K Jan  8 18:54 new_id
--w------- 1 root root 4.0K Jan  8 18:54 uevent
--w------- 1 root root 4.0K Jan  8 18:54 unbind
```

2. Reading from `dpi` sometimes returns `0:0`

```
$ cat /sys/bus/hid/drivers/razermouse/0003:1532:0039.0009/dpi
800:800
$ cat /sys/bus/hid/drivers/razermouse/0003:1532:0039.0009/dpi
800:800
$ cat /sys/bus/hid/drivers/razermouse/0003:1532:0039.0009/dpi
0:0
$ cat /sys/bus/hid/drivers/razermouse/0003:1532:0039.0009/dpi
800:800
```

Is it normal? Is it just because this device is not fully supported?

terrycain, 2017-01-08:
The sometimes returning 0 is noramally caused by the mouse being busy doing something else. Also cat `device_mode` `device_serial` `device_type` `firmware_version` `poll_rate` 

Also the following should toggle the led (run as root, not sudo)
```
# On
echo -n '1' > scroll_led_state
# Off
echo -n '0' > scroll_led_state
```

m13253, 2017-01-08:
> Also cat device_mode device_serial device_type firmware_version poll_rate

```
$ cat device_mode 
0:0
$ cat device_serial 
PM1448082401133
$ cat device_type 
Unknown Device
$ cat firmware_version 
v1.2
$ cat poll_rate 
125
```

Note 1: `echo -n '0' > scroll_led_state` does not turn off the light, but I don't think it's an urgent problem.

Note 2: Although I can set different `poll_rate` (125, 500, 1000), the `mouse-dpi-tool` (from package `libevdev`) always reports 1000 Hz. (Maybe it's a feature, not a bug?)

terrycain, 2017-01-08:
Not really any docs.

Try those scroll_led echo commands but with ´> test´ instead

terrycain, 2017-01-08:
I know what needs to be done to change dpi stages but I haven't implemented it yet.

m13253, 2017-01-09:
> Try those scroll_led echo commands but with ´> test´ instead

I tried `echo -n '0' > test` or `echo -n '1' > test`, nothing actually happens.

terrycain, 2017-01-09:
Ahh ok, packet captures it is then ;-)

m13253, 2017-01-17:
### DPI
Set DPI to 500:500:
```
0000   00 7f 0e 44 02 88 ff ff 53 02 00 08 04 00 00 00
0010   8d b8 7d 58 00 00 00 00 72 54 0c 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 07 04 05 00[01 f4 01 f4]00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00 06 00
```
Set DPI to 6400:6400:
```
0000   40 2b 5c 0d 01 88 ff ff 53 02 00 08 04 00 00 00
0010   5b b8 7d 58 00 00 00 00 c4 2c 08 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 07 04 05 00[19 00 19 00]00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00 06 00
```
### Poll rate
Set poll rate to 125:
```
0000   00 73 0e 44 02 88 ff ff 53 02 00 08 04 00 00 00
0010   f0 b8 7d 58 00 00 00 00 cd a3 0d 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 01 00 05[08]00 00 00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00[0c]00
```
Set poll rate to 500:
```
0000   c0 16 0c 82 01 88 ff ff 53 02 00 08 04 00 00 00
0010   e7 b8 7d 58 00 00 00 00 17 02 03 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 01 00 05[02]00 00 00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00[06]00
```
Set poll rate to 1000:
```
0000   c0 13 0c 82 01 88 ff ff 53 02 00 08 04 00 00 00
0010   17 b9 7d 58 00 00 00 00 40 b1 07 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 01 00 05[01]00 00 00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00[05]00
```
### LED
Turn LED off:
```
0000   80 c7 99 54 02 88 ff ff 53 02 00 08 04 00 00 00
0010   20 b6 7d 58 00 00 00 00 24 26 0c 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 03 03 00 01 01[00]00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00[00]00
```
Turn LED on:
```
0000   c0 46 11 44 02 88 ff ff 53 02 00 08 04 00 00 00
0010   21 b6 7d 58 00 00 00 00 69 f7 0e 00 8d ff ff ff
0020   5a 00 00 00 5a 00 00 00 21 09 00 03 00 00 5a 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   00 ff 00 00 00 03 03 00 01 01[01]00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0090   00 00 00 00 00 00 00 00[01]00
```

m13253, 2017-01-17:
The bytes starting from 0x0040 is the data fragment.

I am not sure what does the 0x0098 byte mean. Is it a checksum?

terrycain, 2017-01-17:
> The bytes starting from 0x0040 is the data fragment.

Yep you can usually tell by it starting 0x00FF000000

> I am not sure what does the 0x0098 byte mean. Is it a checksum?

Yep

terrycain, 2017-01-19:
BTW a pcap/pcapng would of been useful as i have a script that will pull out all of that information.

Ok remove driver from your system. Re-get the driver from the same branch. Scroll wheel should work, seems i had a copy and paste mistake in there where the scroll_wheel file was trying to toggle the logo_led ha

m13253, 2017-01-20:
> Scroll wheel should work

It worked.

But I hope we can solve the DPI problem. The LED won't affect gaming experience, but the DPI settings will do.

Thanks.

terrycain, 2017-01-20:
Yeah the dpi stages is an issue for all mice, I'll get round to that when I can.

I'll write some daemon code for this. We'll test that and then it should be good to go

m13253, 2017-01-20:
That's okay. Thanks.

I think it's okay to merge into the master branch. Would you please do it?

terrycain, 2017-01-20:
@m13253 We'll test daemon then yes, no point merging something that to me is currently half finished.

juliandroid, 2017-06-30:
I've tried the new branch with the support for my Orochi 2013, but it doesn't seem to work.
I can run the daemon with sudo systemctl start razerd.service
and then run the qrazercfg
but once the QUI is opened there are no detected devices.

z3ntu, 2017-06-30:
@juliandroid You are talking about https://github.com/mbuesch/razer which is another project.

idzikovsky, 2017-07-05:
I'm interested in helping with driver for this mouse.
It just need to be tested, or needs some code fixes?
Can I start with checking out this branch, and running `make ubuntu_install`?

z3ntu, 2017-07-05:
I'll send you some debs in a minute (apparently you're on Ubuntu)

z3ntu, 2017-07-05:
[feature_orochi_2013_0x0039.zip](https://github.com/terrycain/razer-drivers/files/1125759/feature_orochi_2013_0x0039.zip)
Please install these debs and reboot.

idzikovsky, 2017-07-05:
Done. How to check that something works?
Do I need to manually start some services?
I've installed RazerGeine from OBS, and got this: http://imgur.com/a/5KXbQ

z3ntu, 2017-07-05:
Please run `cat /sys/bus/hid/drivers/razermouse/0*/device_type` from your terminal. DPI change probably won't work right now as a variable in the daemon isn't set yet (we'll fix that soon).

z3ntu, 2017-07-05:
Also please try if the checkbox `Set Scroll Active` works and changes the lighting on the mouse.

idzikovsky, 2017-07-05:
```
ivan:~$ sudo cat /sys/bus/hid/drivers/razermouse/0003\:1532\:0039.0001/device_type
Unknown Device
```

"Set Scroll Active" works fine.

z3ntu, 2017-07-05:
[feature_orochi_2013_0x0039_v2.zip](https://github.com/terrycain/razer-drivers/files/1125841/feature_orochi_2013_0x0039_v2.zip)
Installt these debs and reboot.

Then please test dpi, poll rate (set to lowest and move your mouse fast, maybe you'll notice something^^). Also the "Unknown device" bug should be gone.

idzikovsky, 2017-07-05:
"Unknown device" bug has gone, but I'm not able to change DPI. Poll rate was working on DEBs from previous archive.

Here is some output:
```
ivan@ivanpc:~$ razergenie 
"PM1331082402788"
"Razer Orochi 2013"
"mouse"
QLayout: Attempting to add QLayout "" to QWidget "", which already has a layout
CURRENT LOCATION:  "2"
QLayout: Attempting to add QLayout "" to QWidget "", which already has a layout
QLayout: Attempting to add QLayout "" to QWidget "", which already has a layout
QLayout: Attempting to add QLayout "" to QWidget "", which already has a layout
librazer: There was an error in QList<int> librazer::QDBusMessageToIntArray(const QDBusMessage&) !
librazer: ""
librazer: ""
currDPI: (800, 800)
librazer: There was an error in int librazer::QDBusMessageToInt(const QDBusMessage&) !
librazer: "org.freedesktop.DBus.Error.UnknownMethod"
librazer: "Traceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/dbus/service.py\", line 654, in _message_cb\n    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)\n  File \"/usr/lib/python3/dist-packages/dbus/service.py\", line 246, in _method_lookup\n    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))\ndbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: maxDPI is not a valid method of interface razer.device.dpi\n"
maxDPI: 0
ADD POLLING RATE COMBOBOXES!
Download of image completed for  "PM1331082402788"  at  "/home/ivan/.local/share/razergenie/devicepictures/razer-orochi-2013-gallery-1.png"
```

FYI: I'm using Ubuntu 16.04 with Kernel 4.8.0 from `linux-generic-hwe-16.04` package.

z3ntu, 2017-07-05:
Thanks for the error message. @terrycain should know what's going on.

(btw RazerGenie v0.3 is released and should be out soon on OBS 😉)

idzikovsky, 2017-07-05:
I've tested with RazerGeine from GitHub v0.3 release.
Nice mouse image, but still the same python exception.

z3ntu, 2017-07-05:
Yeah it's a daemon issue and not a front-end issue (just wanted to let you know because quite a bunch of things have improved especially for the user experience).

idzikovsky, 2017-07-05:
I'm able to change DPI after I've made this changes:

```
ivan@ivanpc:~$ diff -u mouse.py.old /usr/lib/python3/dist-packages/razer_daemon/hardware/mouse.py
--- mouse.py.old        2017-07-05 23:19:53.512080457 +0300
+++ /usr/lib/python3/dist-packages/razer_daemon/hardware/mouse.py       2017-07-05 23:22:19.414297952 +0300
@@ -207,7 +207,7 @@
     HAS_MATRIX = False
     MATRIX_DIMS = [-1, -1]  # 1 Row, 15 Cols
     METHODS = ['get_firmware', 'get_matrix_dims', 'has_matrix', 'get_device_name', 'get_device_type_mouse', 'get_dpi_xy', 'set_dpi_xy',
-               'get_poll_rate', 'set_poll_rate', 'set_scroll_active', 'get_scroll_active']
+               'get_poll_rate', 'set_poll_rate', 'set_scroll_active', 'get_scroll_active', 'max_dpi']
 
     DPI_MAX = 6400
```

z3ntu, 2017-07-05:
Thanks for figuring that out!
So if you say, everything works, we can merge it :) Most likely it'll be included in the next release. Thanks for helping!

idzikovsky, 2017-07-05:
Cool.
Here is some small showcase of what finnaly I got: https://streamable.com/dtjye

z3ntu, 2017-07-05:
I'm glad you like my frontend 😊

ddnomad, 2017-07-21:
Hey, so does the mouse is supported by `razer-drivers` now?

terrycain, 2017-07-21:
Yes but it not out in packages release form yet

ddnomad, 2017-07-21:
Yep, got it. Thanks for the info!

commit f285ba08 (2017-07-30): Merge branch 'master' into stable

Possibly fixes https://github.com/lah7/polychromatic/issues/104
Close #235
Close #144
closed 2017-07-30 by terrycain