openrazer/openrazer #146: Suggestion for Windows VM

rquast, 2017-01-06:
Just a suggestion for updating the reverse engineering documentation. I think you can use the free Windows VMs for IE to do this? https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

This may help anyone who is looking for a Windows VM.

terrycain, 2017-01-07:
As long as you can run it in virtualbox or kvm AND install the Razer synapse software then sure I don't see why not. Try it if it works I'll add it to the wiki

rquast, 2017-01-07:
Will do. I've downloaded it and will attempt to install the synapse software and run wireshark. Will let you know how it goes.

CalcProgrammer1, 2017-01-10:
You don't really even need to run Windows in a VM on Linux.  If you install Wireshark on Windows it provides USBPcap as an option and you can log USB transfers from Windows itself.

z3ntu, 2017-01-10:
You can also download an ISO here: https://www.microsoft.com/software-download/windows10

terrycain, 2017-01-10:
@CalcProgrammer1 theres been cases where usb packets don't get captured. The VM method is more reliable though as long as i get packets im not bothered :wink:

rsandoz, 2017-01-11:
wireshark in windows has "sometimes works" behavior for me :(

terrycain, 2017-01-16:
Added link

closed 2017-01-16 by terrycain