openrazer/openrazer #148: Razer Nostromo Support (RZ07-0049)

sdyawg, 2017-01-10:
Heya, been stuck trying to get my poor Nostromo to operate in Archlinux but it's not been going well. This seemed like a good place to start. I'll be following [these](zzzzzzzzzzhttps://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) steps later but wanted to get some basic info up for the request. 

Model: RZ07-0049
Product: RZ07-00490100

Bus 007 Device 002: ID 1532:0111 Razer USA, Ltd

```
[yawg@yawgix ~]$ sudo lsusb -d 1532:0111 -v

Bus 007 Device 002: ID 1532:0111 Razer USA, Ltd
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0111
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Nostromo
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
[yawg@yawgix ~]$ sudo lsusb -d 1532:0111 -v

Bus 007 Device 002: ID 1532:0111 Razer USA, Ltd
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0111
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Nostromo
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

z3ntu, 2017-01-10:
Please post the output of
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

sdyawg, 2017-01-10:
Bah, realized I posted the same output twice and not the one you just requested, here you go:

```
[yawg@yawgix ~]$ sudo ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 240 Jan 10 05:18  .
drwxr-xr-x 4 root root 600 Jan 10 05:18  ..
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Logitech_Gaming_Mouse_G502_056933623533-event-mouse -> ../event1
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Logitech_Gaming_Mouse_G502_056933623533-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Logitech_Gaming_Mouse_G502_056933623533-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan 10 05:17 'usb-Microsoft_Natural'$'\302\256''_Ergonomic_Keyboard_4000-event-kbd' -> ../event5
lrwxrwxrwx 1 root root   9 Jan 10 05:17 'usb-Microsoft_Natural'$'\302\256''_Ergonomic_Keyboard_4000-if01-event-kbd' -> ../event6
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Razer_Razer_Nostromo-event-kbd -> ../event0
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Razer_Razer_Nostromo-if01-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Razer_Razer_Nostromo-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Valve_Software_Steam_Controller-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jan 10 05:17  usb-Valve_Software_Steam_Controller-mouse -> ../mouse2
```

sdyawg, 2017-01-11:
Alright, I come bearing screenshots and packet captures. 

[Screenshot Album](http://imgur.com/a/aPUlM)
[Pcap Files](http://wikisend.com/download/268842/nostromo_pcap.tar)

Wasn't sure what'd be a good filehost for the pcaps, let me know if there's a better resource.

terrycain, 2017-01-11:
Can we get a capture of you changing profiles so the profile leds change

terrycain, 2017-01-11:
And screenshots of the lighting tab would be useful as i could make an educated guess on what's needed

sdyawg, 2017-01-11:
Screenshots already in the album and the profile changing pcap already in the above file (named nostromo_profilecolortest).

For the pcaps I started in the no led profile (keymap 8), clicked the Blue profile (keymap 1) and went down the list to keymap 8 again.

sdyawg, 2017-01-11:
doh, lighting screenshot [here](http://imgur.com/tmooKwm)

terrycain, 2017-01-11:
Brightness the only lighting option?

Also could I get a packet capture from before the vm starts all the way until synapse is open?

sdyawg, 2017-01-11:
Brightness is the only option, single color Blue LED backlight and mousewheel

[system boot pcap](http://wikisend.com/download/275490/nostromo_systemboot.pcapng)

terrycain, 2017-01-11:
You know, if you drag the file into github window it will just upload (at least it works for me)

sdyawg, 2017-01-11:
[nostromopcap.zip](https://github.com/terrycain/razer-drivers/files/699755/nostromopcap.zip)

terrycain, 2017-01-11:
system boot pcap looks good. Will go over the rest soonish :wink:

sdyawg, 2017-01-11:
Excellent, please let me know if/how I can assist, first time asking for anything on github and I'd love to contribute where possible

terrycain, 2017-01-17:
@z3ntu Its arch :wink: Can you get branch `feature_nostromo_0x0111` working and grab me when nostromo is bound to `razerkbd` with `ls -lah` output of device directory in driver

z3ntu, 2017-01-17:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/ecad89f6aaa31baa26f86cc104973992725f5c0f/PKGBUILD.nostromo -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```
You can also reach me on Telegram @z3ntu.

sdyawg, 2017-01-18:
updating to that fork and neither polychromatic nor razercommander can connect to the daemon it appears.

```
[yawg@yawgix ~]$ razercommander
ERROR:dbus.proxies:Introspect error on :1.36:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include:
the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
DEBUG:dbus.proxies:Executing introspect queue due to error
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 30, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message
bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[yawg@yawgix ~]$ ps aux | grep razer
yawg      1119  0.4  0.1 379684 30660 ?        Sl   21:02   0:00 razer-service
yawg      1584  0.0  0.0   8992   808 pts/0    S+   21:03   0:00 grep razer
```

terrycain, 2017-01-18:
ok reboot and `ls -lah /sys/bus/hid/drivers`

sdyawg, 2017-01-18:
```
[yawg@yawgix ~]$ sudo ls -lah /sys/bus/hid/drivers
[sudo] password for yawg:
total 0
drwxr-xr-x 5 root root 0 Jan 18 00:46 .
drwxr-xr-x 4 root root 0 Jan 18 00:45 ..
drwxr-xr-x 2 root root 0 Jan 18 00:45 hid-generic
drwxr-xr-x 2 root root 0 Jan 18 00:45 microsoft
drwxr-xr-x 2 root root 0 Jan 18 00:45 razerkbd
```

terrycain, 2017-01-18:
`ls -lah /sys/bus/hid/drivers/hid-generic`
`ls -lah /sys/bus/hid/drivers/razerkbd`

sdyawg, 2017-01-18:
```
[root@yawgix ~]# ls -lah /sys/bus/hid/drivers/hid-generic
total 0
drwxr-xr-x 2 root root    0 Jan 18 00:46 .
drwxr-xr-x 5 root root    0 Jan 18 00:46 ..
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:046D:0A1F.0005 -> ../../../../devices/pci0000:00/0000:00:12.0/usb6/6-3/6-3:1.3/0003:046D:0A1F.0005
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:046D:C07D.0003 -> ../../../../devices/pci0000:00/0000:00:16.0/usb9/9-3/9-3:1.0/0003:046D:C07D.0003
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:046D:C07D.0004 -> ../../../../devices/pci0000:00/0000:00:16.0/usb9/9-3/9-3:1.1/0003:046D:C07D.0004
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:28DE:1142.0008 -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-5/7-5:1.0/0003:28DE:1142.0008
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:28DE:1142.0009 -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-5/7-5:1.1/0003:28DE:1142.0009
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:28DE:1142.000A -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-5/7-5:1.2/0003:28DE:1142.000A
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:28DE:1142.000B -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-5/7-5:1.3/0003:28DE:1142.000B
lrwxrwxrwx 1 root root    0 Jan 18 00:49 0003:28DE:1142.000C -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-5/7-5:1.4/0003:28DE:1142.000C
--w------- 1 root root 4.0K Jan 18 00:49 bind
lrwxrwxrwx 1 root root    0 Jan 18 00:49 module -> ../../../../module/hid_generic
--w------- 1 root root 4.0K Jan 18 00:49 new_id
--w------- 1 root root 4.0K Jan 18 00:45 uevent
--w------- 1 root root 4.0K Jan 18 00:45 unbind
[root@yawgix ~]# ls -lah /sys/bus/hid/drivers/razerkbd
total 0
drwxr-xr-x 2 root root    0 Jan 18 00:46 .
drwxr-xr-x 5 root root    0 Jan 18 00:46 ..
lrwxrwxrwx 1 root root    0 Jan 18 00:45 0003:1532:0111.0001 -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-1/7-1:1.0/0003:1532:0111.0001
lrwxrwxrwx 1 root root    0 Jan 18 00:45 0003:1532:0111.0002 -> ../../../../devices/pci0000:00/0000:00:13.0/usb7/7-1/7-1:1.1/0003:1532:0111.0002
--w------- 1 root root 4.0K Jan 18 00:45 bind
lrwxrwxrwx 1 root root    0 Jan 18 00:49 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Jan 18 00:49 new_id
--w------- 1 root root 4.0K Jan 18 00:45 uevent
--w------- 1 root root 4.0K Jan 18 00:49 unbind
```

terrycain, 2017-01-18:
`ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0001/`
`ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/`

sdyawg, 2017-01-18:
```
[root@yawgix ~]# ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0001/
total 0
drwxr-xr-x 5 root plugdev    0 Jan 18 00:45 .
drwxr-xr-x 5 root root       0 Jan 18 00:45 ..
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 country
lrwxrwxrwx 1 root plugdev    0 Jan 18 00:45 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jan 18 00:45 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 18 00:45 input
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 modalias
drwxr-xr-x 2 root plugdev    0 Jan 18 00:45 power
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 18 00:45 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4.0K Jan 18 00:45 uevent
[root@yawgix ~]# ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/
total 0
drwxr-xr-x 5 root plugdev    0 Jan 18 00:45 .
drwxr-xr-x 5 root root       0 Jan 18 00:45 ..
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 country
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 device_mode
-r--r----- 1 root plugdev 4.0K Jan 18 00:45 device_serial
-r--r----- 1 root plugdev 4.0K Jan 18 00:45 device_type
lrwxrwxrwx 1 root plugdev    0 Jan 18 00:45 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4.0K Jan 18 00:45 firmware_version
drwxr-xr-x 3 root plugdev    0 Jan 18 00:45 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 18 00:45 input
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 matrix_brightness
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 modalias
drwxr-xr-x 2 root plugdev    0 Jan 18 00:45 power
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 profile_led_blue
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 profile_led_green
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 profile_led_red
-r--r--r-- 1 root plugdev 4.0K Jan 18 00:45 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 18 00:45 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4.0K Jan 18 00:45 test
-rw-r--r-- 1 root plugdev 4.0K Jan 18 00:45 uevent
-r--r----- 1 root plugdev 4.0K Jan 18 00:45 version
[root@yawgix ~]# cat /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/device_type
Unknown Device

```

terrycain, 2017-01-18:
Cat device mode, device serial device type, firmware version, profile led red green blue and matrix brightness

sdyawg, 2017-01-18:
```
[root@yawgix ~]# cd /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/
[root@yawgix 0003:1532:0111.0002]# ls
country      device_serial  driver	      hidraw  matrix_brightness  power		   profile_led_green  report_descriptor  test	 version
device_mode  device_type    firmware_version  input   modalias		 profile_led_blue  profile_led_red    subsystem		 uevent
[root@yawgix 0003:1532:0111.0002]# cat device_mode
0:0
[root@yawgix 0003:1532:0111.0002]# cat device_serial

[root@yawgix 0003:1532:0111.0002]# cat device_type
Unknown Device
[root@yawgix 0003:1532:0111.0002]# cat firmware_version
v0.0
[root@yawgix 0003:1532:0111.0002]# cat profile_led_blue
0
[root@yawgix 0003:1532:0111.0002]# cat profile_led_green
0
[root@yawgix 0003:1532:0111.0002]# cat profile_led_red
0
[root@yawgix 0003:1532:0111.0002]# cat matrix_brightness
0

```

terrycain, 2017-01-18:
Ok as root

`echo -n '1' > profile_led_red`
Should turn on red led. '0' should turn it off. Should also work for blue green

sdyawg, 2017-01-18:
No change in LEDs. Default state is R0,G0,B1, the following had no effect on my device and maintained blue led on even though the profile for blue originally showed 0

[```
root@yawgix 0003:1532:0111.0002]# echo -n '1' > profile_led_red
[root@yawgix 0003:1532:0111.0002]# echo -n '1' > profile_led_green
```

terrycain, 2017-01-18:
Most odd. Will go over the driver again when i get home

terrycain, 2017-01-18:
Try 
`echo -ne '\x03\00' > device_mode`
Then retry colours

sdyawg, 2017-01-18:
```
[root@yawgix 0003:1532:0111.0002]# echo -ne '\x03\00' > device_mode
[root@yawgix 0003:1532:0111.0002]# echo -n '1' > profile_led_red
[root@yawgix 0003:1532:0111.0002]# cat profile_led_red
0
```

No change in LED colors :-/ Still B1,R0,G0 even though profile_led_blue is 0

terrycain, 2017-01-19:
Ok Ive updated driver, clear all traces of driver from your system. Then reboot, run `sudo rmmod razerkbd`
Then install new driver, reboot, 
```
cd /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/
cat device_type
cat device_serial
```

sdyawg, 2017-01-19:
Uh-oh :-/ 

```
yaourt -Rns razer-driver-dkms-git razer-daemon-git python-razer-git polychromatic razer-driver-meta razercommander-git
# reboot
sudo rmmod razerkbd
# no razerkbd found
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git/
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/ecad89f6aaa31baa26f86cc104973992725f5c0f/PKGBUILD.nostromo -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
# reboot

[yawg@yawgix ~]$ ps aux | grep razer
yawg      1103  2.2  0.1 453416 30788 ?        Sl   06:22   0:00 razer-service
yawg      1358  0.0  0.0   8992   856 pts/0    S+   06:22   0:00 grep razer
[yawg@yawgix ~]$ cd /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/
bash: cd: /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/: No such file or directory
[yawg@yawgix ~]$ sudo su
[sudo] password for yawg:
[root@yawgix yawg]# cd /sys/bus/hid/drivers/razerkbd/
bash: cd: /sys/bus/hid/drivers/razerkbd/: No such file or directory
[root@yawgix yawg]# lsmod | grep razerkbd
[root@yawgix yawg]# find /sys -name razerkbd
[yawg@yawgix yawg]# yaourt -Ss razer
aur/python-razer-git 1.1.5.r23.g4db31ba-1 [installed: 1.1.7.r2.g08acce9-1] (1) (0.35)
    A python library for controlling razer-daemon
aur/razer-daemon-git 1.1.5.r23.g4db31ba-1 [installed: 1.1.7.r2.g08acce9-1] (1) (0.35)
    A daemon for controlling razer-driver
aur/razer-driver-dkms-git 1.1.5.r23.g4db31ba-1 [installed: 1.1.7.r2.g08acce9-1] (1) (0.35)
    An entirely open source driver for managing Razer peripherals on Linux. (DKMS)
```

terrycain, 2017-01-19:
Does `sudo modprobe razerkbd` work?

sdyawg, 2017-01-19:
think this is the issue, from the install process:

```
(1/1) Install DKMS modules
==> dkms install razer_chroma_driver/1.0.0 -k 4.8.13-1-ARCH
Error! Bad return status for module build on kernel: 4.8.13-1-ARCH (x86_64)
Consult /var/lib/dkms/razer_chroma_driver/1.0.0/build/make.log for more information.

[yawg@yawgix Downloads]$ cat /var/lib/dkms/razer_chroma_driver/1.0.0/build/make.log
DKMS make.log for razer_chroma_driver-1.0.0 for kernel 4.8.13-1-ARCH (x86_64)
Thu Jan 19 14:53:42 PST 2017

:: Compiling Razer kernel modules
========================================
make -C /lib/modules/4.8.13-1-ARCH/build SUBDIRS=/var/lib/dkms/razer_chroma_driver/1.0.0/build/driver modules > /dev/null 2>&1
make: *** [Makefile:25: driver] Error 2
```

z3ntu, 2017-01-20:
Do you have `linux-headers` installed (if you have the default kernel `linux` installed)

sdyawg, 2017-01-20:
but of course:

```
[yawg@yawgix #]$ sudo pacman -Ss linux-headers
[sudo] password for yawg:
core/linux-headers 4.8.13-1 [installed]
    Header files and scripts for building modules for Linux kernel
```

z3ntu, 2017-01-20:
Run the following (use bash and not some other shell!)
```
git clone https://github.com/terrycain/razer-drivers.git
cd razer-drivers
_extramodules=extramodules-4.8-ARCH
_kernver="$(cat /usr/lib/modules/${_extramodules}/version)"
make KERNELDIR=/usr/lib/modules/$_kernver/build driver_verbose
```

sdyawg, 2017-01-20:
```
[yawg@yawgix razer-drivers]$ _extramodules=extramodules-4.8-ARCH
[yawg@yawgix razer-drivers]$ _kernver="$(cat /usr/lib/modules/${_extramodules}/version)"
[yawg@yawgix razer-drivers]$ make KERNELDIR=/usr/lib/modules/$_kernver/build driver_verbose

:: Compiling Razer kernel modules
========================================
make -C /usr/lib/modules/4.8.13-1-ARCH/build SUBDIRS=/home/yawg/Downloads/razer-drivers/driver modules
make[1]: Entering directory '/usr/lib/modules/4.8.13-1-ARCH/build'
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razerkbd_driver.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razercommon.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razerchromacommon.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerkbd.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razermouse_driver.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razermouse.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razerfirefly_driver.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerfirefly.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razerkraken_driver.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerkraken.o
  CC [M]  /home/yawg/Downloads/razer-drivers/driver/razermug_driver.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razermug.o
  Building modules, stage 2.
  MODPOST 5 modules
  CC      /home/yawg/Downloads/razer-drivers/driver/razerfirefly.mod.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerfirefly.ko
  CC      /home/yawg/Downloads/razer-drivers/driver/razerkbd.mod.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerkbd.ko
  CC      /home/yawg/Downloads/razer-drivers/driver/razerkraken.mod.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razerkraken.ko
  CC      /home/yawg/Downloads/razer-drivers/driver/razermouse.mod.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razermouse.ko
  CC      /home/yawg/Downloads/razer-drivers/driver/razermug.mod.o
  LD [M]  /home/yawg/Downloads/razer-drivers/driver/razermug.ko
make[1]: Leaving directory '/usr/lib/modules/4.8.13-1-ARCH/build'

```

z3ntu, 2017-01-20:
So that works...? Edit `/usr/src/razer_chroma_driver-1.0.0/dkms.conf` and replace the line `MAKE="KERNELDIR=/lib/modules/${kernelver}/build make driver"` with `MAKE="KERNELDIR=/lib/modules/${kernelver}/build make driver_verbose"`, then run `sudo dkms install razer_chroma_driver/1.0.0` (you might need to run `sudo dkms remove razer_chroma_driver/1.0.0` first)

sdyawg, 2017-01-20:
I had uninstalled the razer-driver-meta package prior to working on this new build so the only dkms.conf file was in razer-drivers/install_files/dkms/, but I did modify the line. However I'm now getting this:

```
[yawg@yawgix razer-drivers]$ makepkg -sr
==> ERROR: install file (razer-driver-dkms-git.install) does not exist or is not a regular file.
```

z3ntu, 2017-01-20:
Run `git reset --hard origin/master`

sdyawg, 2017-01-20:
oh the problem is I'm still using @z3ntu 's PKGBUILD for pacmaning it and that's where the dkms.install line is coming from. I've been using his instructions above:

```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/ecad89f6aaa31baa26f86cc104973992725f5c0f/PKGBUILD.nostromo -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```

z3ntu, 2017-01-20:
Okay just edit the dkms install file and do the stuff listed in https://github.com/terrycain/razer-drivers/issues/148#issuecomment-273992135 (do a fresh clone of the razer-drivers-git)

terrycain, 2017-01-20:
Try again with latest commit*

sdyawg, 2017-01-20:
alright, I found I just needed to comment a line in the PKGBUILD file from @z3ntu, grabbed latest commit and added the line to dkms.conf

after reboot:

```
[yawg@yawgix ~]$ cd /sys/bus/hid/drivers/razerkbd/0003:1532:0111.0002/
[yawg@yawgix 0003:1532:0111.0002]$ cat device_type
Razer Nostromo
[yawg@yawgix 0003:1532:0111.0002]$ cat device_serial

```

terrycain, 2017-01-20:
is weird the serial doesnt work. Will have a look

prispewnic, 2017-04-06:
Hello! I tried to install support for the Nostromo, on arch, according to this instruction, but failed.
Could you add support for this device to the next version of the program? Thanks.

lezed1, 2017-04-06:
@prispewnic are you having issues installing the drivers, or getting the drivers to work with your device?

z3ntu, 2017-04-06:
@prispewnic Without any logs or error messages we can't help you :/

prispewnic, 2017-04-06:
Okay, let's start in order : I have installed the latest razer-daemon, razer-driver-dkms, python-razer and razercommander, from AUR. Not git versions. After installation, I launched razercommander and didn't find Nostromo in the devices list. So, I found this topic and repeated all the described actions. Unsuccessfully. Can you give me some step-by-step instructions, to install Nostromo support?

P.S. Sorry for my weak english :)

z3ntu, 2017-04-06:
So Nostromo support only exists in the branch `feature_nostromo_0x0111` right now and it's not included in a stable release as there are still issues @terrycain is working on.

prispewnic, 2017-04-06:
If I understand you correctly, there are no working versions of drivers for this device yet?

z3ntu, 2017-04-06:
Not yet.

prispewnic, 2017-04-06:
Okay, thanks for the clarification.

terrycain, 2017-04-10:
Also have nostromo coming soon so will do it then.

prispewnic, 2017-04-10:
Also have nostromo coming soon so will do it then.

terrycain, 2017-04-15:
Packages are building in launchpad, will be available soon

closed 2017-04-15 by terrycain