openrazer/openrazer #149: Razer Naga Hex RZ01-0075 support

mjtorn, 2017-01-10:
https://github.com/mbuesch/razer/commit/280b239aec491356339bb398e17fbc70c396a4b2

If I'm reading that correct, and I'm probably wrong because I barely looked, this should be pretty much the same as the other Hex mice, right?

My `lsusb` tells me `ID 1532:0041`

mjtorn, 2017-01-10:
The `0041` is related to https://github.com/mbuesch/razer/issues/25 I think.

Here's some more dumping:

```
lrwxrwxrwx 1 root root 9 Jan 10 21:13 usb-Razer_Razer_Naga_Hex-event-mouse -> ../event6
lrwxrwxrwx 1 root root 9 Jan 10 21:13 usb-Razer_Razer_Naga_Hex-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root 9 Jan 10 21:13 usb-Razer_Razer_Naga_Hex-mouse -> ../mouse1
```
```
Bus 001 Device 006: ID 1532:0041 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0041 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Hex
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     103
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
```

Thanks!

z3ntu, 2017-01-10:
Screenshots from Synapse please :)

mjtorn, 2017-01-10:
I'll see what I can do.

If someone else wants to pitch in here, feel free to pwn this issue ;)

Just had to get the lowest-hanging fruit out here.

mjtorn, 2017-01-11:
[image]
[image]
[image]

I attempted some packet captures as well,did they succeed?

[scroll-wheel-off-and-on.pcapng.gz](https://github.com/terrycain/razer-drivers/files/698087/scroll-wheel-off-and-on.pcapng.gz)
[logo-off-and-on.pcapng.gz](https://github.com/terrycain/razer-drivers/files/698089/logo-off-and-on.pcapng.gz)
[DPI-slide.pcapng.gz](https://github.com/terrycain/razer-drivers/files/698088/DPI-slide.pcapng.gz)

terrycain, 2017-01-11:
Nope nope and nope :wink: What did you do trying to get the captures?

mjtorn, 2017-01-11:
As the names suggest, turning the LEDs on and off and adjusting the DPI setting.

I had Wireshark running with sudo (because my user wasn't allowed to access the USB), used the filter as given in the wiki, adjusted by my `lsusb` output.

Ran the capture, saved with gzip compression and uploaded.

I can get more details and a retry later on, but if there's a common obvious gotcha or something in the uploads that point to an error, let me know!

terrycain, 2017-01-11:
did you `sudo modprobe usbmod` ?
Otherwise I'd just say try again. Your looking for packets of length 154

mjtorn, 2017-01-11:
Yes, of course. At first I thought I had it loaded at boot time and got a file-not-found error, so I `modprobe`d it very explicitly.

Maybe I used the wrong usbmon option? There was a bunch and I think I took the first one. Gotta try that when I next get the chance.

z3ntu, 2017-01-11:
You have to choose the usbmon<busid> eg if lsusb says, your mouse is on Bus 4, you have to choose usbmon4 :)

mjtorn, 2017-01-11:
I checked, ` On Linux 2.6.22 and later, the special "usbmon0" interface receives events from all USB buses.` from the wiki, so I actually should have captured a lot more and filtered my bus out of the noise. But I'll try the specific one when I can.

mjtorn, 2017-01-12:
@terrycain 
Here are the two lighting-related dumps:
[logo-off-and-on.pcapng.gz](https://github.com/terrycain/razer-drivers/files/701874/logo-off-and-on.pcapng.gz)
[scroll-wheel-off-and-on.pcapng.gz](https://github.com/terrycain/razer-drivers/files/701875/scroll-wheel-off-and-on.pcapng.gz)

Here I start with the sensitivity of 1800 DPI, scroll it down to the minimum 100 DPI, reset to 1800 DPI, scroll up to the maximum 5600 DPI and reset to 1800 DPI.
[DPI-slide.pcapng.gz](https://github.com/terrycain/razer-drivers/files/701877/DPI-slide.pcapng.gz)

Hope I got them right this time around.

@z3ntu Yeah, this is what I get for not really having time for anything :D My first one in the list was not `usbmon0` but `usbmon1` which would not have worked.

mjtorn, 2017-01-12:
Here are the separate axes in case that's something that anyone wants or even a blocker, but I did not go through them one by one. I hope you can infer whatever you need from these.

[DPI-X-1800-900-100-1800-3000-4000-5000-5600-1800.pcapng.gz](https://github.com/terrycain/razer-drivers/files/701897/DPI-X-1800-900-100-1800-3000-4000-5000-5600-1800.pcapng.gz)
[DPI-Y-1800-900-100-1800-3000-4000-5000-5600-1800.pcapng.gz](https://github.com/terrycain/razer-drivers/files/701896/DPI-Y-1800-900-100-1800-3000-4000-5000-5600-1800.pcapng.gz)

terrycain, 2017-01-17:
What distro you on?

mjtorn, 2017-01-18:
Ah yeah, I'm running this stuff on a dedicated gaming device, a Steam Machine running SteamOS. Consider it Debian stable for all intents and purposes, it wasn't rocket surgery to get any missing dependencies going. I just took the Wireshark files on my desktop, everything else is on SteamOS.

mjtorn, 2017-01-18:
And I haven't had the opportunity yet to try the Orbweaver support, so thanks a million for the debs, I'll try them asap after you give them :)

terrycain, 2017-01-19:
Try debs [here](https://drive.google.com/drive/folders/0BydYgD2LzDUCenp0SEh0akcxMGc?usp=sharing)

Install, reboot,
```
ls -lah /sys/bus/hid/drivers/
ls -lah /sys/bus/hid/drivers/razermouse/
ls -lah /sys/bus/hid/drivers/razermouse/0*/
```

mjtorn, 2017-01-20:
```
desktop@steamos:~$ ls -lah /sys/bus/hid/drivers/
total 0
drwxr-xr-x 4 root root 0 Jan 20 16:47 .
drwxr-xr-x 4 root root 0 Jan 20 16:42 ..
drwxr-xr-x 2 root root 0 Jan 20 16:42 hid-generic
drwxr-xr-x 2 root root 0 Jan 20 16:42 razermouse
desktop@steamos:~$ ls -lah /sys/bus/hid/drivers/razermouse/
total 0
drwxr-xr-x 2 root root    0 Jan 20 16:47 .
drwxr-xr-x 4 root root    0 Jan 20 16:47 ..
lrwxrwxrwx 1 root root    0 Jan 20 16:42 0003:1532:0041.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:1532:0041.0006
lrwxrwxrwx 1 root root    0 Jan 20 16:42 0003:1532:0041.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:1532:0041.0007
--w------- 1 root root 4.0K Jan 20 16:42 bind
lrwxrwxrwx 1 root root    0 Jan 20 16:47 module -> ../../../../module/razermouse
--w------- 1 root root 4.0K Jan 20 16:47 new_id
--w------- 1 root root 4.0K Jan 20 16:42 uevent
--w------- 1 root root 4.0K Jan 20 16:47 unbind
desktop@steamos:~$ ls -lah /sys/bus/hid/drivers/razermouse/0*/
/sys/bus/hid/drivers/razermouse/0003:1532:0041.0006/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 20 16:42 .
drwxr-xr-x 5 root root       0 Jan 20 16:42 ..
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 country
-rw-rw---- 1 root plugdev 4.0K Jan 20 16:42 device_mode
-r--r----- 1 root plugdev 4.0K Jan 20 16:42 device_serial
-r--r----- 1 root plugdev 4.0K Jan 20 16:42 device_type
-rw-rw---- 1 root plugdev 4.0K Jan 20 16:42 dpi
lrwxrwxrwx 1 root plugdev    0 Jan 20 16:42 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4.0K Jan 20 16:42 firmware_version
drwxr-xr-x 3 root plugdev    0 Jan 20 16:42 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 20 16:42 input
-rw-rw---- 1 root plugdev 4.0K Jan 20 16:42 logo_led_state
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 modalias
-rw-rw---- 1 root plugdev 4.0K Jan 20 16:42 poll_rate
drwxr-xr-x 2 root plugdev    0 Jan 20 16:42 power
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 report_descriptor
-rw-rw---- 1 root plugdev 4.0K Jan 20 16:42 scroll_led_state
lrwxrwxrwx 1 root plugdev    0 Jan 20 16:42 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4.0K Jan 20 16:42 test
-rw-r--r-- 1 root plugdev 4.0K Jan 20 16:42 uevent
-r--r----- 1 root plugdev 4.0K Jan 20 16:42 version

/sys/bus/hid/drivers/razermouse/0003:1532:0041.0007/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 20 16:42 .
drwxr-xr-x 5 root root       0 Jan 20 16:42 ..
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 country
lrwxrwxrwx 1 root plugdev    0 Jan 20 16:42 driver -> ../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Jan 20 16:42 hidraw
drwxr-xr-x 3 root plugdev    0 Jan 20 16:42 input
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 modalias
drwxr-xr-x 2 root plugdev    0 Jan 20 16:42 power
-r--r--r-- 1 root plugdev 4.0K Jan 20 16:42 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 20 16:42 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4.0K Jan 20 16:42 uevent
```

And of course it's present in dmesg too.

mjtorn, 2017-01-20:
Would I be correct in thinking that Polychromatic and/or razerCommander would be the ones who configure the mouse sensitivity and such?

Thanks!

z3ntu, 2017-01-20:
I don't think polychromatic supports dpi right now and I don't know about razercommander.

mjtorn, 2017-01-20:
Sure, but it's on their turf and the driver part can be considered done? The back/forward buttons (for the browser) and the number keys on the mouse worked as well.

terrycain, 2017-01-20:
As root
```
cd /sys/bus/hid/drivers/razermouse/0003:1532:0041.0006  # (or one with lowest number)
cat device_mode
cat device_serial
cat device_type
cat firmware_version
cat poll_rate
cat version
cat dpi

# Should turn led's on
echo -n '1' > logo_led_state
echo -n '1' > scroll_led_state

# Should turn led's off
echo -n '0' > logo_led_state
echo -n '0' > scroll_led_state
```

mjtorn, 2017-01-21:
```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat device_mode 
0:0
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat device_serial 
PM1448075401473
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat device_type 
Unknown Device
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat firmware_version 
v2.1
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat poll_rate 
500
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat version 
1.1.7
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat dpi 
68:68
```

And I can confirm the LEDs work.

What's the unit for DPI here? X:Y I guess is the format. It cannot be set with `cat`.

Good work, thanks!

terrycain, 2017-01-21:
Need to update some daemon code for the dpi, sadly the naga hex dpi scaled 5600dpi into one byte with a max value of 255 (so obviously theres a fair amount of error in the conversion).

Once i've done that we'll test daemon and should be able to do dpi there.

mjtorn, 2017-01-21:
Nice, thanks!

terrycain, 2017-01-21:
Try [these](https://drive.google.com/open?id=0BydYgD2LzDUCenp0SEh0akcxMGc) debs. Install reboot, 

Then run `ps -A | grep razer`

mjtorn, 2017-01-22:
Installed the debs, the expected result followed

```
desktop@steamos:~$ ps -A | grep razer
 1309 ?        00:00:00 razer-service
```

terrycain, 2017-01-23:
Install d-feet, click session bus, search razer, expand options on the right like so and paste screenshot
[image]

mjtorn, 2017-01-23:
Had to mirror it elsewhere again, GH would not let me upload; http://imgur.com/a/hlwrK

terrycain, 2017-01-23:
In theory the get and set dpi/scroll/logo should work, the logo and scroll take 1/0 dpi should take 100-5600 but i think technically it has 6750 dpi but as the dpi is scaled to a byte theres a fair bit of inaccuraccy

mjtorn, 2017-01-26:
I'll take a look at setting DPI as soon as maybe tomorrow, cool :)

mjtorn, 2017-01-27:
Gave it a shot right now, with the updated packages in the repository, but it seems all the Naga is kinda lacking there. Setting the DPI did not obviously work in the .debs from above.

You wouldn't happen to have new packages for me to try?

Thanks!

terrycain, 2017-01-29:
I'll make some new ones when I get home

mjtorn, 2017-01-30:
Nice, thanks!

terrycain, 2017-01-30:
https://drive.google.com/drive/folders/0BydYgD2LzDUCbnRHRG9KcDY1eGM?usp=sharing

Try these

mjtorn, 2017-01-30:
I guess they should work? They do not :/

```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat dpi 
68:68
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# echo 72:72 > dpi 
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat dpi 68:68
```

```
$ sha1sum *
7259cceef48d4867a28aa58221abaf5f3bbf014f  python3-razer_1.1.7-0ubuntu1_all.deb
4183b67a68699f4e47fd6e787258d02b08cd9c71  razer-daemon_1.1.7-0ubuntu1_amd64.deb
a7e628ddc726a00a626b4d6c057c99ceffe5a477  razer-doc_1.1.7-0ubuntu1_all.deb
88c6afffc0bb5170cf2c131702641d51da010577  razer-kernel-modules-dkms_1.1.7-0ubuntu1_amd64.deb
```

terrycain, 2017-01-30:
try `echo -ne '\x20\x20' > dpi`

mjtorn, 2017-02-01:
That did it! I was just expecting the same format :)

Unplugging and replugging the mouse returns the DPI setting, but I'm sure it won't survive a reboot without daemon support, right?

```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0006# cat dpi 
68:68
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0006# echo -ne '\x20\x20' > dpi
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0006# cat dpi 0:0
```

mjtorn, 2017-02-01:
Looked at the daemon code and it's quite obvious how it's encoded. But I can't set anything;

```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# echo -ne '\xcc\xcc' > dpi
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# cat dpi 0:0
```

Even if I kill the daemon, thinking that maybe it's real fast to reset the zero value...

mjtorn, 2017-02-01:
Seems the state is stored on the device itself. I plugged it into Windows and back here;

```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.000C# cat dpi 
220:220
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.000C# echo -ne '\xcc\xcc' > dpi
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.000C# cat dpi 0:0
```

terrycain, 2017-02-01:
when you set the dpi, does it actually change?

mjtorn, 2017-02-02:
No, it drops to zero, becoming ridiculously slow to move around.

terrycain, 2017-02-02:
Well it dropping to 0 at least means the dpi is being set albeit to a shitty value. Will look into it

mjtorn, 2017-02-02:
Thanks, let me know! :)

terrycain, 2017-02-02:
Try [these](https://drive.google.com/open?id=0BydYgD2LzDUCbnRHRG9KcDY1eGM) think i've fixed dpi, it was always setting 0 as thats exactly what the code was supposed to do... as i used the wrong variable.

mjtorn, 2017-02-04:
```
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# echo -ne '\x60\x60' > dpi && cat dpi
96:96
root@steamos:/sys/bus/hid/drivers/razermouse/0003:1532:0041.0008# echo -ne '\xff\x60' > dpi && cat dpi
255:96
```

So it works now :) Even with derpy asymmetric precisions ;)

Thanks a lot for the work! Next I'm going to hope for some Polychromatic support :)

terrycain, 2017-02-04:
Awesome, will merge soon.

mjtorn, 2017-02-04:
Nice! Looking forward to a release with this :)

mjtorn, 2017-02-11:
I see this is still open. Is the code officially released yet or will you close this when the release is out?

terrycain, 2017-02-11:
Err, I'll see if i can get it out tomorrow, otherwise it'll be next weekend

commit d6aa71e7 (2017-03-04): Merge branch 'master' into stable

Closes #212
Closes #184
Closes #169
Closes #149
Closes #116
closed 2017-03-04 by terrycain