openrazer/openrazer #150: Razer BlackWidow Chroma, keyboard "h" key issue

jpfoote, 2017-01-13:
The "h" key is not recognized the first time it is pressed. When pressing the "h"
 the second time it is processed but only while the keyboard is lit. Once the keyboard goes dark the "h" key is no longer processed on the first key press. Once the keyboard is lit it is processed every time.

Any assistance in this matter would be greatly appreciated.

jpfoote, 2017-01-13:
Ubuntu 16.04, 
Driver version 0.3.5,
Configuration 4,
Installed from PPA

No other keys seem to be affected.
Moved the keyboard to 2 other systems for testing and no issue found.

terrycain, 2017-01-13:
You sure version 0.3.5 is right? Were around 1.1.7. I also have a bw chroma and have no issue

z3ntu, 2017-01-14:
Polychromatic is at 0.3.5, maybe he mixed it up.

terrycain, 2017-01-14:
`ls -lah /sys/bus/hid/drivers/razerkbd`

jpfoote, 2017-01-16:
Sorry I was going off of the display of the Controller, this is the output form the command listed above.

ls -lah /sys/bus/hid/drivers/razerkbd
total 0
drwxr-xr-x 2 root root    0 Jan 16 10:15 .
drwxr-xr-x 6 root root    0 Jan 16 10:15 ..
lrwxrwxrwx 1 root root    0 Jan 16 10:15 0003:1532:0203.0008 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1/1-5.1.1/1-5.1.1:1.0/0003:1532:0203.0008
lrwxrwxrwx 1 root root    0 Jan 16 10:15 0003:1532:0203.0009 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1/1-5.1.1/1-5.1.1:1.1/0003:1532:0203.0009
lrwxrwxrwx 1 root root    0 Jan 16 10:15 0003:1532:0203.000A -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1/1-5.1.1/1-5.1.1:1.2/0003:1532:0203.000A
--w------- 1 root root 4.0K Jan 16 00:13 bind
lrwxrwxrwx 1 root root    0 Jan 16 10:15 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Jan 16 10:15 new_id
--w------- 1 root root 4.0K Jan 16 10:15 uevent
--w------- 1 root root 4.0K Jan 16 10:15 unbind

terrycain, 2017-01-16:
`ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0203.000A`

jpfoote, 2017-01-17:
ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0203.000A
lrwxrwxrwx 1 root root 0 Jan 17 16:19 /sys/bus/hid/drivers/razerkbd/0003:1532:0203.000A -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1/1-5.1.1/1-5.1.1:1.2/0003:1532:0203.000A

jpfoote, 2017-01-17:
This problem seems to be resolved by recent updates.

closed 2017-01-17 by jpfoote