openrazer/openrazer #151: DeathAdder Elite custom mode?

rsandoz, 2017-01-15:
Any chance of adding custom mode for DeathAdder Elite?

From:
```
/**
 * Write device file "mode_custom"
 *
 * Sets the mouse to custom mode whenever the file is written to
 */
static ssize_t razer_attr_write_mode_custom(struct device *dev, struct device_attribute *attr, const char *buf, size_t count)
{
    struct usb_interface *intf = to_usb_interface(dev->parent);
    struct usb_device *usb_dev = interface_to_usbdev(intf);
    struct razer_report report = razer_chroma_standard_matrix_effect_custom_frame(NOSTORE);
    
    if(usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_MAMBA_WIRELESS) {
        report.transaction_id.id = 0x80;
    }
    
    razer_send_payload(usb_dev, &report);
    
    return count;
}
```

To:
```
/**
 * Write device file "mode_custom"
 *
 * Sets the mouse to custom mode whenever the file is written to
 */
static ssize_t razer_attr_write_mode_custom(struct device *dev, struct device_attribute *attr, const char *buf, size_t count)
{
    struct usb_interface *intf = to_usb_interface(dev->parent);
    struct usb_device *usb_dev = interface_to_usbdev(intf);
	struct razer_report report;
    
	switch (usb_dev->descriptor.idProduct) {
		case USB_DEVICE_ID_RAZER_NAGA_HEX_V2:
			report = razer_chroma_standard_matrix_effect_custom_frame(NOSTORE);
			report.transaction_id.id = 0x3f;
			break;

		case USB_DEVICE_ID_RAZER_DEATHADDER_ELITE:
			report = razer_chroma_extended_matrix_effect_custom_frame();
			report.transaction_id.id = 0x3f;
			break;

		case USB_DEVICE_ID_RAZER_MAMBA_WIRELESS:
			report = razer_chroma_standard_matrix_effect_custom_frame(NOSTORE);
			report.transaction_id.id = 0x80;
			break;
	}
    
    razer_send_payload(usb_dev, &report);
    
    return count;
}
```

Add:
```
/**
* Write device file "matrix_custom_frame"
*
* Format
* ROW_ID START_COL STOP_COL RGB...
*/
static ssize_t razer_attr_write_matrix_custom_frame(struct device *dev, struct device_attribute *attr, const char *buf, size_t count)
{
    struct usb_interface *intf = to_usb_interface(dev->parent);
    struct usb_device *usb_dev = interface_to_usbdev(intf);
	struct razer_report report;

	size_t offset = 0;
	unsigned char row_id;
	unsigned char start_col;
	unsigned char stop_col;
	unsigned char row_length;

	//printk(KERN_ALERT "razerkbd: Total count: %d\n", (unsigned char)count);

	while (offset < count)
	{
		if (offset + 3 > count)
		{
			printk(KERN_ALERT "razerkbd: Wrong Amount of data provided: Should be ROW_ID, START_COL, STOP_COL, N_RGB\n");
			break;
		}

		row_id = buf[offset++];
		start_col = buf[offset++];
		stop_col = buf[offset++];
		row_length = ((stop_col + 1) - start_col) * 3;

		// printk(KERN_ALERT "razerkbd: Row ID: %d, Start: %d, Stop: %d, row length: %d\n", row_id, start_col, stop_col, row_length);

		if (start_col > stop_col)
		{
			printk(KERN_ALERT "razerkbd: Start column is greater than end column\n");
			break;
		}

		if (offset + row_length > count)
		{
			printk(KERN_ALERT "razerkbd: Not enough RGB to fill row\n");
			break;
		}

		// Offset now at beginning of RGB data
		switch(usb_dev->descriptor.idProduct)
		{
			case USB_DEVICE_ID_RAZER_NAGA_HEX_V2:
				report = razer_chroma_standard_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
				report.transaction_id.id = 0x3f;
				break;

			case USB_DEVICE_ID_RAZER_DEATHADDER_ELITE:
				report = razer_chroma_extended_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
				report.transaction_id.id = 0x3f;
				break;

			case USB_DEVICE_ID_RAZER_MAMBA_WIRELESS:
				report = razer_chroma_standard_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
				report.transaction_id.id = 0x80;
				break;
		}

		razer_send_payload(dev, &report);

		// *3 as its 3 bytes per col (RGB)
		offset += row_length;
	}

	return count;
}
```

And update ddfile, probe, disconnect and your python callings.

terrycain, 2017-01-15:
as the elite actually got custom mode?

terrycain, 2017-01-15:
From what i can see it only has logo and scroll wheel leds ?

rsandoz, 2017-01-15:
I actually noticed it via Razer's SDK call.  Capped  it with wireshark win as well.

```
16	3.029098	1.17.0	host	URB_CONTROL out				USB		36
17	3.029341	host	1.17.0	URB_CONTROL out				USB		118
18	3.029366	host	1.17.0	SET CONFIGURATION Status	USB		28
19	3.031765	1.17.0	host	URB_CONTROL out				USB		36
20	3.032008	host	1.17.0	URB_CONTROL out				USB		118
21	3.032020	host	1.17.0	SET CONFIGURATION Status	USB		28

Frame 16: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface 0
USB URB
    [Source: host]
    [Destination: 1.17.0]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffa50c45f341e0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CLASS_INTERFACE (0x001b)
    IRP information: 0x00, Direction: FDO -> PDO
        0000 000. = Reserved: 0x00
        .... ...0 = Direction: FDO -> PDO (0x0)
    URB bus id: 1
    Device address: 17
    Endpoint: 0x00, Direction: OUT
        0... .... = Direction: OUT (0)
        .000 0000 = Endpoint value: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 8
    [Response in: 17]
    Control transfer stage: Setup (0)
    [bInterfaceClass: Unknown (0xffff)]
URB setup
    bmRequestType: 0x21
        0... .... = Direction: Host-to-device
        .01. .... = Type: Class (0x1)
        ...0 0001 = Recipient: Interface (0x01)
    bRequest: 9
    wValue: 0x0300
    wIndex: 0 (0x0000)
    wLength: 90
0000   1c 00 e0 41 f3 45 0c a5 ff ff 00 00 00 00 1b 00  ...A.E..........
0010   00 01 00 11 00 00 02 08 00 00 00 00 21 09 00 03  ............!...
0020   00 00 5a 00                                      ..Z.

Frame 17: 118 bytes on wire (944 bits), 118 bytes captured (944 bits) on interface 0
USB URB
    [Source: 1.17.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffa50c45f341e0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
        0000 000. = Reserved: 0x00
        .... ...1 = Direction: PDO -> FDO (0x1)
    URB bus id: 1
    Device address: 17
    Endpoint: 0x00, Direction: OUT
        0... .... = Direction: OUT (0)
        .000 0000 = Endpoint value: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 90
    [Request in: 16]
    [Time from request: 0.000243000 seconds]
    Control transfer stage: Data (1)
    [bInterfaceClass: Unknown (0xffff)]
CONTROL response data: 001f000000470f03000000000100000000ffff0000000000...
0000   1c 00 e0 41 f3 45 0c a5 ff ff 00 00 00 00 08 00  ...A.E..........
0010   01 01 00 11 00 00 02 5a 00 00 00 01 00 1f 00 00  .......Z........
0020   00 47 0f 03 00 00 00 00 01 00 00 00 00 ff ff 00  .G..............
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0040   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0070   00 00 00 00 4a 00                                ....J.

Frame 19: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface 0
USB URB
    [Source: host]
    [Destination: 1.17.0]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffa50c4a5c7900
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CLASS_INTERFACE (0x001b)
    IRP information: 0x00, Direction: FDO -> PDO
        0000 000. = Reserved: 0x00
        .... ...0 = Direction: FDO -> PDO (0x0)
    URB bus id: 1
    Device address: 17
    Endpoint: 0x00, Direction: OUT
        0... .... = Direction: OUT (0)
        .000 0000 = Endpoint value: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 8
    [Response in: 20]
    Control transfer stage: Setup (0)
    [bInterfaceClass: Unknown (0xffff)]
URB setup
    bmRequestType: 0x21
        0... .... = Direction: Host-to-device
        .01. .... = Type: Class (0x1)
        ...0 0001 = Recipient: Interface (0x01)
    bRequest: 9
    wValue: 0x0300
    wIndex: 0 (0x0000)
    wLength: 90
0000   1c 00 00 79 5c 4a 0c a5 ff ff 00 00 00 00 1b 00  ...y\J..........
0010   00 01 00 11 00 00 02 08 00 00 00 00 21 09 00 03  ............!...
0020   00 00 5a 00                                      ..Z.

Frame 20: 118 bytes on wire (944 bits), 118 bytes captured (944 bits) on interface 0
USB URB
    [Source: 1.17.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffffa50c4a5c7900
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
        0000 000. = Reserved: 0x00
        .... ...1 = Direction: PDO -> FDO (0x1)
    URB bus id: 1
    Device address: 17
    Endpoint: 0x00, Direction: OUT
        0... .... = Direction: OUT (0)
        .000 0000 = Endpoint value: 0
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 90
    [Request in: 19]
    [Time from request: 0.000243000 seconds]
    Control transfer stage: Data (1)
    [bInterfaceClass: Unknown (0xffff)]
CONTROL response data: 001f0000000c0f0200000800000000000000000000000000...
0000   1c 00 00 79 5c 4a 0c a5 ff ff 00 00 00 00 08 00  ...y\J..........
0010   01 01 00 11 00 00 02 5a 00 00 00 01 00 1f 00 00  .......Z........
0020   00 0c 0f 02 00 00 08 00 00 00 00 00 00 00 00 00  ................
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0040   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0070   00 00 00 00 09 00                                ......
```

rsandoz, 2017-01-15:
Much more responsive than the static call's for logo and scroll

rsandoz, 2017-01-15:
FWIW, I ported all of you driver code to windows and #ifdeffed out all the driver file stuff and called it using winusb using a filter driver and winusb's usb_control_msg.  Actually works better than the SDK with my BladePro.

terrycain, 2017-01-15:
That's pretty cool, will add support in the week sometime as i can test on my da elite v2 as well.

> FWIW, I ported all of you driver code to windows and #ifdeffed out all the driver file stuff and called it using winusb using a filter driver and winusb's usb_control_msg. Actually works better than the SDK with my BladePro.

Awesome, I'm surprised it actually works and even more surprised it works better than their SDK.

rsandoz, 2017-01-15:
Yep, SDK has some bugs.  For one, the BladePro2016 and FireFly are criss-crossed somewhere internally in their DLL.  I cannot light the media keys or touchpad leds with Keyboard CUSTOM commands and when I  command FireFly, it activates the touchpad (just the bottom, left and right).

terrycain, 2017-01-15:
HA, its always good to know that the driver beats the proper library, is there much differences from the windows port? I wouldn't be against including some ifdefs here and there.

rsandoz, 2017-01-15:
I have actually been using alot of #ifdef WIN32 and made a types.h with stuff kzalloc, kfree and usleep, etc.  I will try to get it cleaner a bit later.  Lots of slop in it now.

rsandoz, 2017-01-15:
At first I had a bunch of defines for stuff like uint8, etc, but I am finding it is not needed as there seems to be a delivered one with such defines, plus some defines are not need if I ifdef out the whole module file device at the bottom of the five drier files (kbd,mouse,firefly,kraken,mug)

rsandoz, 2017-01-15:
SO, I am cleaning all the defines, then I will try and come up with a diff way to a better types.h

rsandoz, 2017-01-15:
plus the whole usb handle aspect,  I added said init functions to your kbd,mouse,and firefly (like in your mug and kraken), then removed the usb handle parm off of all the razer_attr_... calls and just referenced it from an instance of the device struct. also did a bunch of your TODOs for moving things around.

rsandoz, 2017-01-15:
I plan to go back, grab your exact code again, the redo my effort (should be much faster the second time around), that way, I am not just trying to "make it compile and work".

terrycain, 2017-01-15:
Nice :), sounds like you've been putting alot of effort in

rsandoz, 2017-01-16:
Here is what I have so far if you want to try it:
http://www.richardsandoz.com/razer/hackedDriver.zip

I am sure I broke it for Ubuntu (changed parms to be winusb specific), but it is the same fore the mostpart.  Should be able to open with vstudio 15 and run main.c.  Multicolors my devices.

rsandoz, 2017-01-16:
back in Ubuntu world now.  Looks like my original change needs to be incorporated into razer_attr_write_set_key_row rather than my writing of the whole new method, razer_attr_write_matrix_custom_frame (replacing the report= with a switch statement)

```
        report = razer_chroma_misc_one_row_set_custom_frame(start_col, stop_col, (unsigned char*)&buf[offset]);
```

```
        switch(usb_dev->descriptor.idProduct)
        {
            case USB_DEVICE_ID_RAZER_NAGA_HEX_V2:
                report = razer_chroma_standard_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
                report.transaction_id.id = 0x3f;
                break;

            case USB_DEVICE_ID_RAZER_DEATHADDER_ELITE:
                report = razer_chroma_extended_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
                report.transaction_id.id = 0x3f;
                break;

            case USB_DEVICE_ID_RAZER_MAMBA_WIRELESS:
                report = razer_chroma_standard_matrix_set_custom_frame(row_id, start_col, stop_col, (unsigned char*)&buf[offset]);
                report.transaction_id.id = 0x80;
                break;
        }
```

rsandoz, 2017-01-16:
Can't test the USB_DEVICE_ID_RAZER_NAGA_HEX_V2, as I don't have one, but it seems to follow the same pattern as the Death Adder ELite.

rsandoz, 2017-01-16:
I originally did a dkms install on my Ubuntu 1604.  After I grab your git, is there a quick and easy way to update?  I tried `make ubuntu_install` and that updated the /usr... tree.  /var... tree still had original source.  I wound up just doing a manual rmmod and insmod for razermouse.ko and also copied the updated mouse source from the /usr... tree to the /var... tree manually.  Please forgive my ignorance on these details.

BTW,
echo -n -e "\x00\x00\x01\x00\xFF\x00\xFF\x00\x00" > matrix_custom_frame
echo -n "1" > matrix_effect_custom
 worked great on my Death Adder Elite locally.  I'll try to setup a pull request later on.

terrycain, 2017-01-16:
Am closing as added elite/naga mode (and want less issues open :wink) but will look into the windows port as that would be nice to have. Feel free to join us on telegram chat if you want (add @terrycain).

closed 2017-01-16 by terrycain

borgdylan, 2017-04-19:
Does the driver support the da elite's dpi buttons? Or does the driver and daemon combination support dpi setting from a front-end like polychromatic? (I've heard that polychromatic supports this for other mice like ouroboros).

Do any keys need to be remapped on the mouse or is it ok as it comes out the box?

terrycain, 2017-04-19:
dpi buttons should work out of the box, for changing dpi at least. Front ends should also be able to set dpi