openrazer/openrazer #153: Ornata (1532:021f) suppot

cdubz, 2017-01-16:
I see the Ornata listed as supported on the [Devices page](https://terrycain.github.io/razer-drivers/#devices), but not in the [Keyboards table](https://github.com/terrycain/razer-drivers#keyboards). Is the Ornata supported, but with a different device ID? Anything I can do to help get it supported?

```
~ $ lsusb
[...]
Bus 001 Device 011: ID 1532:021f Razer USA, Ltd 
[...]
```

terrycain, 2017-01-16:
Razer Ornata Chroma	1532:021E

The chroma is supported, is the 021f non chroma?

cdubz, 2017-01-16:
Nope. It's an Ornata that I purchased yesterday. I didn't check the device ID until I got everything installed. The Chroma is listed as 1532:021**E** in the devices table.

terrycain, 2017-01-16:
Should be a simple copy and paste job really.

cdubz, 2017-01-16:
Let me know if you need any additional info or testing.

terrycain, 2017-01-16:
It'll need testing, what distro you on?

cdubz, 2017-01-16:
Mint 17.3 (Trusty 14.04)

terrycain, 2017-01-16:
Output of `ls -lah /dev/input/by-id`

terrycain, 2017-01-16:
Install debs from [here](https://drive.google.com/drive/folders/0BydYgD2LzDUCQURlNlVseUdnSk0?usp=sharing), reboot (hopefully you dont have secureboot enabled as dkms doesnt like that yet)

the paste output of 
```
ls -lah /sys/bus/hid/drivers/
ls -lah /sys/bus/hid/drivers/razerkbd/
ls -lah /sys/bus/hid/drivers/razerkbd/0*/
ls ~/.razer-service/logs
pgrep razer-service
```

cdubz, 2017-01-17:
## Mint 17.3 (Trusty 14.04)

### Before installing packages

I wasn't sure if you wanted this before and/or after installing the debs --

```
~ $ ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 200 Jan 16 19:03 .
drwxr-xr-x 4 root root 560 Jan 16 19:03 ..
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-La-VIEW_Technology_SteelSeries-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-La-VIEW_Technology_SteelSeries-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-La-VIEW_Technology_SteelSeries-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Jan 16 19:03 usb-Microsoft_Microsoft®_LifeCam_Cinema_TM_-event-if00 -> ../event20
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-Razer_Razer_Ornata-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-Razer_Razer_Ornata-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-Razer_Razer_Ornata-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Jan 16 19:02 usb-Razer_Razer_Ornata-if02-mouse -> ../mouse1
```

### After installing packages
FYI - I had to modify the deb control file to look for `gir1.2-gtk-3.0` instead of `gir1.2-gtk3.0`.

```
~ $ ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 200 Jan 16 19:20 .
drwxr-xr-x 4 root root 560 Jan 16 19:20 ..
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-La-VIEW_Technology_SteelSeries-event-mouse -> ../event15
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-La-VIEW_Technology_SteelSeries-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root   9 Jan 16 19:20 usb-La-VIEW_Technology_SteelSeries-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-Microsoft_Microsoft®_LifeCam_Cinema_TM_-event-if00 -> ../event20
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-Razer_Razer_Ornata-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-Razer_Razer_Ornata-if01-event-kbd -> ../event17
lrwxrwxrwx 1 root root  10 Jan 16 19:20 usb-Razer_Razer_Ornata-if02-event-mouse -> ../event19
lrwxrwxrwx 1 root root   9 Jan 16 19:20 usb-Razer_Razer_Ornata-if02-mouse -> ../mouse1
```

```
~ $ ls -lah /sys/bus/hid/drivers/
total 0
drwxr-xr-x 4 root root 0 Jan 16 19:21 .
drwxr-xr-x 4 root root 0 Jan 16 19:20 ..
drwxr-xr-x 2 root root 0 Jan 16 19:20 hid-generic
drwxr-xr-x 2 root root 0 Jan 16 19:20 razerkbd
```

```
~ $ ls -lah /sys/bus/hid/drivers/razerkbd/
total 0
drwxr-xr-x 2 root root    0 Jan 16 19:21 .
drwxr-xr-x 4 root root    0 Jan 16 19:21 ..
lrwxrwxrwx 1 root root    0 Jan 16 19:22 0003:1532:021F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4:1.0/0003:1532:021F.0003
lrwxrwxrwx 1 root root    0 Jan 16 19:22 0003:1532:021F.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4:1.1/0003:1532:021F.0004
lrwxrwxrwx 1 root root    0 Jan 16 19:22 0003:1532:021F.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4:1.2/0003:1532:021F.0005
--w------- 1 root root 4.0K Jan 16 19:20 bind
lrwxrwxrwx 1 root root    0 Jan 16 19:22 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Jan 16 19:22 new_id
--w------- 1 root root 4.0K Jan 16 19:22 uevent
--w------- 1 root root 4.0K Jan 16 19:22 unbind
```

```
~ $ ls -lah /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:021F.0003/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 16 19:20 .
drwxr-xr-x 5 root root       0 Jan 16 19:23 ..
lrwxrwxrwx 1 root plugdev    0 Jan 16 19:20 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
[... 37 lines cut ...]
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_breath
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_custom
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_none
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_reactive
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_starlight
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_static
--w--w---- 1 root plugdev 4.0K Jan 16 19:20 matrix_effect_wave
-r--r--r-- 1 root plugdev 4.0K Jan 16 19:20 modalias
drwxr-xr-x 2 root plugdev    0 Jan 16 19:20 power
-r--r--r-- 1 root plugdev 4.0K Jan 16 19:20 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 16 19:20 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4.0K Jan 16 19:20 test
-rw-r--r-- 1 root plugdev 4.0K Jan 16 19:20 uevent
-r--r----- 1 root plugdev 4.0K Jan 16 19:20 version
```

```
~ $ ls ~/.razer-service/logs
razer.log
```

```
~ $ pgrep razer-service
3178
```

cdubz, 2017-01-17:
## Ubuntu 16.04 LTS

### Before installing packages

```
~$ ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 200 Jan 16 19:26 .
drwxr-xr-x 4 root root 520 Jan 16 19:26 ..
lrwxrwxrwx 1 root root  10 Jan 16 19:03 usb-DB04008LAA0AXL_2SF022_0x0001-event-if00 -> ../event10
lrwxrwxrwx 1 root root  10 Jan 16 19:26 usb-La-VIEW_Technology_SteelSeries-event-mouse -> ../event15
lrwxrwxrwx 1 root root  10 Jan 16 19:26 usb-La-VIEW_Technology_SteelSeries-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root   9 Jan 16 19:26 usb-La-VIEW_Technology_SteelSeries-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Jan 16 19:26 usb-Razer_Razer_Ornata-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Jan 16 19:26 usb-Razer_Razer_Ornata-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Jan 16 19:26 usb-Razer_Razer_Ornata-if02-event-mouse -> ../event14
lrwxrwxrwx 1 root root   9 Jan 16 19:26 usb-Razer_Razer_Ornata-if02-mouse -> ../mouse1
```

### After installing packages

Same note as above re: `gir1.2-gtk-3.0`.

```
~$ ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 200 Jan 16 19:40 .
drwxr-xr-x 4 root root 520 Jan 16 19:40 ..
lrwxrwxrwx 1 root root  10 Jan 16 19:40 usb-DB04008LAA0AXL_2SF022_0x0001-event-if00 -> ../event15
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-La-VIEW_Technology_SteelSeries-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-La-VIEW_Technology_SteelSeries-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-La-VIEW_Technology_SteelSeries-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-Razer_Razer_Ornata-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-Razer_Razer_Ornata-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-Razer_Razer_Ornata-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jan 16 19:40 usb-Razer_Razer_Ornata-if02-mouse -> ../mouse1
```

```
~$ ls -lah /sys/bus/hid/drivers/
total 0
drwxr-xr-x 4 root root 0 Jan 16 19:43 .
drwxr-xr-x 4 root root 0 Jan 16 19:40 ..
drwxr-xr-x 2 root root 0 Jan 16 19:40 hid-generic
drwxr-xr-x 2 root root 0 Jan 16 19:40 razerkbd
```

```
~$ ls -lah /sys/bus/hid/drivers/razerkbd/
total 0
drwxr-xr-x 2 root root    0 Jan 16 19:43 .
drwxr-xr-x 4 root root    0 Jan 16 19:43 ..
lrwxrwxrwx 1 root root    0 Jan 16 19:40 0003:1532:021F.0001 -> ../../../../devices/pci0000:00/0000:00:1a.0usb1/1-1/1-1.2/1-1.2:1.0/0003:1532:021F.0001
lrwxrwxrwx 1 root root    0 Jan 16 19:40 0003:1532:021F.0002 -> ../../../../devices/pci0000:00/0000:00:1a.0usb1/1-1/1-1.2/1-1.2:1.1/0003:1532:021F.0002
lrwxrwxrwx 1 root root    0 Jan 16 19:40 0003:1532:021F.0003 -> ../../../../devices/pci0000:00/0000:00:1a.0usb1/1-1/1-1.2/1-1.2:1.2/0003:1532:021F.0003
--w------- 1 root root 4.0K Jan 16 19:40 bind
lrwxrwxrwx 1 root root    0 Jan 16 19:43 module -> ../../../../module/razerkbd
--w------- 1 root root 4.0K Jan 16 19:43 new_id
--w------- 1 root root 4.0K Jan 16 19:40 uevent
--w------- 1 root root 4.0K Jan 16 19:43 unbind
```

```
~$ ls -lah /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:021F.0001/:
total 0
drwxr-xr-x 5 root plugdev    0 Jan 16 19:40 .
drwxr-xr-x 5 root root       0 Jan 16 19:40 ..
-r--r--r-- 1 root plugdev 4.0K Jan 16 19:40 country
lrwxrwxrwx 1 root plugdev    0 Jan 16 19:40 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
[... 39 lines cut ...]
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_breath
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_custom
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_none
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_reactive
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_starlight
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_static
--w--w---- 1 root plugdev 4.0K Jan 16 19:40 matrix_effect_wave
-r--r--r-- 1 root plugdev 4.0K Jan 16 19:40 modalias
drwxr-xr-x 2 root plugdev    0 Jan 16 19:40 power
-r--r--r-- 1 root plugdev 4.0K Jan 16 19:40 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jan 16 19:40 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4.0K Jan 16 19:40 test
-rw-r--r-- 1 root plugdev 4.0K Jan 16 19:40 uevent
-r--r----- 1 root plugdev 4.0K Jan 16 19:40 version
```

```
~$ ls ~/.razer-service/logs
razer.log
```

```
~$ pgrep razer-service
1891
```

terrycain, 2017-01-17:
Looks good, try one of the apps like polychromatic, does it work?

cdubz, 2017-01-17:
*Also included detailed notes about the Function keys. Should have tested them more thoroughly yesterday.*

## Mint 17.3 (Trusty 14.04)

### Function keys
- [x] **F1-3** Mute / Volume up / Volume down
- [x] **F4** (nothing)
- [ ] **F5-6** Previous song / Play/pause / Next song - These only seem to work if you do not use anything else to control the music (e.g. select a new song with the mouse). All testing was done in Banshee.
- [x] **F8** (nothing)
- [ ] **F9** OTF Macro - Macro indicator starts flashing, but there is no way to stop it, haha.
- [ ] **F10** Gaming mode - Gaming mode indicator lights up, but the keys that should be prevented all still work.
- [x] **F11-12** Keyboard dim / Keyboard brighten - Works well enough, but with some kinks: it seems to only go through six brightness levels and Mint flashes the mute/unmute icon for Keyboard dim.

### Polychromatic

I attempted to use the [legacy version](https://github.com/lah7/polychromatic#-ubuntu-1404--linux-mint-17) and, perhaps predictably, had issues:

Could not install `razer-chroma-driver_20160612_ubuntu_amd64.deb`
```
~/Downloads $ sudo dpkg -i razer-chroma-driver_20160612_ubuntu_amd64.deb 
(Reading database ... 653516 files and directories currently installed.)
Preparing to unpack razer-chroma-driver_20160612_ubuntu_amd64.deb ...
Unpacking razer-chroma-driver (1.0.0) ...
dpkg: error processing archive razer-chroma-driver_20160612_ubuntu_amd64.deb (--install):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/dkms.conf', which is also in package razer-kernel-modules-dkms 1.1.6-0ubuntu1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 razer-chroma-driver_20160612_ubuntu_amd64.deb
```

I removed that dependency from `polychromatic_0.1.0_all.deb` and it was able to install, but cannot launch.
```
~ $ polychromatic-controller 
Traceback (most recent call last):
  File "/usr/bin/polychromatic-controller", line 29, in <module>
    import razer.daemon_dbus
ImportError: No module named 'razer.daemon_dbus'
~ $ polychromatic-tray-applet 
Traceback (most recent call last):
  File "/usr/bin/polychromatic-tray-applet", line 30, in <module>
    import razer.daemon_dbus
ImportError: No module named 'razer.daemon_dbus'
```

Let me know if you want me to do any more testing, but also perfectly understandable if this is too convoluted to support at this point.

## Ubuntu 16.04 LTS

### Function keys
- [x] **F1-3** Mute / Volume up / Volume down
- [x] **F4** (nothing)
- [x] **F5-6** Previous song / Play/pause / Next song - All good with Rhythmbox!
- [x] **F8** (nothing)
- [ ] **F9** OTF Macro - See F9 notes for 10.04 (above). Same behavior.
- [ ] **F10** Gaming mode - See F10 notes for 10.04 (above). Same behavior.
- [x] **F11-12** Keyboard dim / Keyboard brighten - See F11-12 notes to 14.04 (above). Same behavior (including mute/unmute indicator for Keyboard dim).

### Polychromatic
- [x] Controller and Tray App launch
- [x] Keyboard recognized as Razer Ornata
- [x] **Brightness scale**
- [x] **None** effect
- [x] **Wave** effect (and directions)
- [x] **Reactive** effect
- [ ] **Ripple** effect - No change at all.
- [x] **Static** effect
- [ ] **Gaming mode toggle** - See above re: F10 key. Same behavior from Controller.
- [ ] **All color options** - None work. The Ornata can only do green so this is expected. See notes below re: Windows options for "Green scale".
- [ ] **Macro keys** - See above re: F9 key. Same behavior from Controller.
- [ ] **Application profiles** - the only option that appears is color setting and that does not work (per above).

### Other

Playing with the color options causes the Controller to crash. Let me know if you want me to test anything specific and send logs.

## Windows comparison

There are some missing features from Polychromatic that are available in Windows. I suspect there are reasons for this beyond just the Ornata/Chroma differentiation, but I will note them just in case.

### Missing features
- [ ] **Profile management** (not necessarily program triggered) and **switching** (with Fn + (num) keys).
- [ ] **Breathing** effect - Slow fade in and out from no lights to max (this effect is actually terrible, haha).
- [ ] **Ripple** effect - Ripples lights away from keys as they are pressed (**Reactive** style).
- [ ] **Starlight** effect - Random **Reactive**-style light ups without relying on key press.
- [ ] **Green scale** - Instead of multiple colors, there is a simple scale of eight shades of green in many places and a 0-255 slider in at least one place (Effects Configurator).
- [ ] **Effects Configurator** - Comprehensive setup to allow configuration of specific keys and layers/combinations of effects and settings.

### Macros

The flow of macro recording in Windows is a bit different. Not sure if this is a difference between Ornata and Chroma or has some other history in Polychromatic's development -

According to the [Razer Ornata master guide](http://drivers.razersupport.com//index.php?_m=downloads&_a=viewdownload&downloaditemid=2442&nav=0,77,608,611,612):

[image]

terrycain, 2017-01-17:
Ok am going to ignore mint for now. 

For game mode, did the led light up? And does the Windows key work when led is lit

On the fly macroing technically works as you described what happened if the driver is doing it's job. It doesn't work properly for you as have "limited" it to keyboards with dedicated macro keys for now until I can spend some time on it.

Brightness, Yeah we can look at that.

Ripple effect and any custom key lighting is done in software we can look at that at the end.

Breathing and starlight work on the chroma so we can investigate this.

Green scale and effects configurator, thats someone else's problem I don't deal with front end gui's, same with profile management. @lah7 is good person to ask about that. 

There's a group of us on telegram add me @terrycain if your interested.

lah7, 2017-01-17:
Polychromatic will be hit and miss with new devices the daemon supports, but if the daemon says it can, the Controller will follow along. :wink: The effects are handled by the daemon - it's quite literally a controller in that respect.

The legacy version (for 14.04 Trusty) will be no use here, since it is designed for the **old** driver/daemon... and the latest development of the Controller cannot satisfy the WebKit2 dependency under 14.04.

Thanks for the feedback -- the profile features I'd like to flesh out in v0.4.0, the green scale is a good suggestion and custom effects would be likely be a distant future feature. If there's anything else, [please raise an issue at that repository.](https://github.com/lah7/polychromatic)

It's not quite a Synapse alternate yet. :smile:

cdubz, 2017-01-17:
@lah7 & @terrycain - Thank you both for you efforts on these tools. It's very nice to see the level of functionality available based on your work. I will join the Telegram chat whenever we can work out getting me invited and am happy to test anything you end up working on.

> For game mode, did the led light up? And does the Windows key work when led is lit

The Gaming Mode led **does** light up, but the Alt+Tab and Super key (I assumed this is the "Windows" key) **are not** disabled. This is the case in both 16.04 and 14.04.

terrycain, 2017-01-17:
Thats odd, super key should be disabled, alt tab is done in synapse software

Mind reading [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) and getting some captures

cdubz, 2017-01-17:
Oops, meant to say Alt+F4. That also is **not** blocked in Gaming Mode.

I will work through the captures as soon as I can. I don't have a Windows VM set up at the moment and haven't used Wireshark in a long time.

terrycain, 2017-01-17:
Yeah game mode only blocks superkey, alt+f4 alt+tab is done in synapse.

And no rush. It ain't difficult, should be pretty straight forward :wink:

cdubz, 2017-01-18:
Captures attached -

[Ornata - Gaming Mode - Super Key.pcapng.zip](https://github.com/terrycain/razer-drivers/files/714077/Ornata.-.Gaming.Mode.-.Super.Key.pcapng.zip)

1. Enable Gaming Mode
1. Hit Super Key twice
1. Disable Gaming Mode

[Ornata - Gaming Mode - AltF4.pcapng.zip](https://github.com/terrycain/razer-drivers/files/714076/Ornata.-.Gaming.Mode.-.AltF4.pcapng.zip)

1. Enable Gaming Mode
1. Hit Alt+F4 twice
1. Disable Gaming Mode

terrycain, 2017-01-18:
Grab captures for startlight and breathing and ripple (be careful ripple will spam)

cdubz, 2017-01-18:
[Ornata - Starlight.pcapng.zip](https://github.com/terrycain/razer-drivers/files/714196/Ornata.-.Starlight.pcapng.zip)

1. Change effect from None to Starlight
2. Change Duration from Short to Medium
3. Change Duration from Medium to Long
4. Change effect from Starlight to None

_Note: Duration is a setting specific to the Starlight effect._

[Ornata - Breathing.pcapng.zip](https://github.com/terrycain/razer-drivers/files/714198/Ornata.-.Breathing.pcapng.zip)

1. Change effect from None to Breathing
2. Select green shade 7 (8 is default)
3. Select green shade 6
4. Select green shade 5
5. Select green shade 4
6. Select green shade 3
7. Select green shade 2
8. Select green shade 1
9. Change effect from Breathing to None

_Note: "Green shade" is an option for most effects (everything except Starlight and Wave). Just figured I'd throw it in there in case it could be useful._

[Ornata - Ripple.pcapng.zip](https://github.com/terrycain/razer-drivers/files/714197/Ornata.-.Ripple.pcapng.zip)

1. Change effect from None to Ripple
2. Tap "G" key
3. Change effect from Ripple to None

terrycain, 2017-01-19:
cd to `/sys/bus/hid/drivers/razerkbd/0003:1532:021F.0003/` (`0003` might have changed, pick one with highest number`) as root

echo -ne '\x03\x00\xFF\x00' > matrix_effect_starlight

cdubz, 2017-01-19:
Both work :+1:

terrycain, 2017-01-19:
The issue is either with daemon or polychromatic then. Can you install the program `d-feet`

Go to the `session` tab, search for `razer` and there should be an entry `org.razer`, click it.
List the options you get on the right hand side, expand the tree first. Screenshot is find, i dont expect you to type it all out lol, unless you want to.

cdubz, 2017-01-20:
[image]

terrycain, 2017-01-20:
Oh well it seems I didnt add breathing effect to the right place on daemon.

Double click on `setStarlightSingle` type `3, 0, 255, 0` and hit execute, that _should_ run startlight mode

cdubz, 2017-01-20:
Yep. It does.

terrycain, 2017-01-20:
Ok so polychromatic is broke for starlight, that or pylib is broke.

Get debs from [here](https://drive.google.com/open?id=0BydYgD2LzDUCQURlNlVseUdnSk0)
Then reboot etc...
open d-feet setBreathingEffectSingle or something similar and type in `0, 255, 0` and hit execute, should set breathing,

cdubz, 2017-01-21:
Installed debs, rebooted, `setBreathSingle` works with `0, 255, 0`.

terrycain, 2017-01-21:
Awesome. Will merge changes soonish, will also release a package once ive fixed some stuff im working on

commit 99a465eb (2017-01-21): Merge branch 'stable'

Launchpad has build packages, they are in the process of being published.
Closes #153
closed 2017-01-21 by terrycain