openrazer/openrazer #155: make ubuntu_install breaks my system

rsandoz, 2017-01-16:
When I do make ubuntu_install, I get:
```
:: Installing for Ubuntu
=====================================================
mv //usr/lib/python3.* //usr/lib/python3
mv: cannot move '//usr/lib/python3.5' to '//usr/lib/python3/python3.5': Directory not empty
Makefile:158: recipe for target 'ubuntu_install' failed
make: *** [ubuntu_install] Error 1
root@rsandoz-Blade-Pro:/home/rsandoz/razer-drivers# 
```

In Makefile, what is this supposed to do?
```
	mv $(DESTDIR)/usr/lib/python3.* $(DESTDIR)/usr/lib/python3
```
When I clear away (backup) the dest dir that the erroneous make command is complaining about, this breaks my Ubuntu installation to some degree resulting in core dumps and encoding error messages.  I put the dir (python3.5) back the dir to /usr/lib, all is well again.  Is the above command only supposed to be moving razer stuff (not the whole 3.5 lib)?

terrycain, 2017-01-16:
ahh it was really used for packaging when making .debs. You shouldnt really be using the makefile to install on ubuntu.

If you install whatever package to get dpkg-buildpackage, you can try running ./scripts/build_debs.sh

z3ntu, 2017-01-21:
@rsandoz Is the answer from Terry sufficient?

rsandoz, 2017-01-21:
yes.  Makes sense.  My main thing was being able to build Terry's latest and greatest (from git) onto my Ubuntu system and testing any source changes I may make locally.  Looks like I have to first, mod the source, second, run build_debs, last, update my system from the built debs.  

I was on Linux a few years ago and got intimately familiar with gcc, make, and even built some custom drivers, so it is just a matter of me learning these packaging things.

closed 2017-01-21 by rsandoz