openrazer/openrazer #156: Suggestion of supporting the Razer Ripsaw (Video Capture Device)

DerFetteVonGamer, 2017-01-16:
This is just a suggestion because I couldn't find any other software that was able to recognise this handy piece of hardware, so I think this also needs support from your daemon.

It would be perfectly adequate if It was possible to open a videostream of it using vlc for example.
Or if it was recognised as a webcam device from the kernel. (/dev/video0)

terrycain, 2017-01-16:
Get telegram add me `@terrycain` and we're discuss it, it'll be a massive undertaking to support it, though it would be epic if we could

hernol, 2017-01-22:
Any news on this? I'd like to help with what a can.

terrycain, 2017-01-23:
I have some captures from @DerFetteVonGamer, I have mailed my Razer contacts asking for some specs, not much we can do yet. Its rather different from current Razer stuff. I give them a couple of weeks, if we get nothing I'll have another look.

hernol, 2017-01-23:
Nice to hear that. Let's hope they reply with some useful information.
Cheers.

terrycain, 2017-01-23:
Will keep you guys updates. Yeah I'll give them a few weeks, if they dont reply I will harass customer support ;-)

hernol, 2017-02-22:
I think u can start harassing customer support ;)

terrycain, 2017-02-22:
Got a reply, sadly no spec as it's a separate line of devices

StephenFluin, 2017-04-22:
What do you need in terms of capture / debugging to get the ripsaw working with Linux?

jairBAR, 2017-09-10:
Hey guys :)
any news about this?

terrycain, 2017-09-11:
Nah sorry guys, until I get my hands on one theres little we can do.

jairBAR, 2017-09-11:
Bummer. well for anyone researching this topic and reading this comment: 
i found an alternative capture device to the ripsaw that features the same near-zero latency and works well for ubuntu (streams to vlc) with kernal versions 3.5.0 and up (without any need for drivers/modules). Its called AV.io HD by epiphan. costs way more though.

jobdiogenes, 2017-09-11:
Any UVC capture card will work on Linux. 
Razer need just make Ripsaw UVC compliant, that will solve Driver issues for all Platforms.!!
Like Magewell XI100DUSB-HDMI  (U$299) or 
Similars like: Imillet HDMI Video Capture (U$149)
HDMI Game Record Y&H (U$125)
Theres a plenty of options in Amazon. Just search for "hdml usb 3.0 capture UVC"
I have a Magewell, thats work without problems. But its price its twice the others.
Look at comments before buy. Some pieces of hardware get issues after a time.
Ripsaw is well recommended, but they not make UVC compliant until now.
I hope that they make the new ones UVC compliant since will solve a lot of problems for then.
Since they don't need to care more about drivers for Windows, Mac, Linux, Android, ..

jairBAR, 2017-09-12:
Roger that, UVC is a standard webcam type USB device class. 
It will work with linux as you said. but what about latency? If your'e looking for the lowest possible latency through the hdmi-usb conversion which UVC capture card would you use?

j3yps, 2017-09-12:
Hope you'll be able to get one soon @terrycain

christianhujer, 2017-12-08:
Folks, I have a Razer Ripsaw, and I would like to use it in Linux. I have solid development experience, and I even have a team that can work on this. Let's connect in some way and find out how we can get this running.

terrycain, 2017-12-08:
@christianhujer Add @terrycain on telegram, we've got a collection of razer poeple there

j3yps, 2017-12-08:
@terrycain is there a telegram group chat dedicated to this that we could join?

terrycain, 2017-12-08:
@j3yps Not dedicated to the ripsaw but we can make one

j3yps, 2017-12-08:
@terrycain that'll be awesome!

davidkuntner, 2017-12-24:
Hey folks.
I would also like to help getting the ripsaw supported.
Sadly, since I started this issue, I obviously still can't use it,
which is a bummer.

Babble-, 2018-10-14:
> Folks, I have a Razer Ripsaw, and I would like to use it in Linux. I have solid development experience, and I even have a team that can work on this. Let's connect in some way and find out how we can get this running.

Do you have any news?

christianhujer, 2018-10-16:
Hey. Not yet. The guy in my team who could work on this is actually
currently still busy getting the Razer StarGazer working on Linux, not the
normal stuff, but keying for depth sensing.

gytisrepecka, 2019-10-02:
Did **Razer Ripsaw HD** by any chance get UVC compliant?

adamculp, 2019-10-03:
I just received a Razer Ripsaw HD, and it also appears to not work on Ubuntu. Any chance this issue could/will be revived, and possibly solved?

Thanks in advance.

gytisrepecka, 2019-11-26:
Or perhaps anyone has specific model suggestion?
I'm interested in HDMI capture card with 4K passthrough (which Razer Ripsaw HD has) and UVC for Linux compatibility.

OverStruck, 2020-08-29:
Is unfortunate that this card has no linux support :(

dejanr, 2020-10-09:
It actually used to work, but with latest kernels most of apps that use video source from this capture device are stucking or frozing, from my research it probably has to do with partial support for UVC

gytisrepecka, 2020-10-12:
@dejanr Which Kernel version have you tried Ripsaw with?

Entropy512, 2021-11-08:
@dejanr - are you saying it appears as a UVC device but is not fully compatible/is glitchy?  (similar to how RealSense cameras and Sony PS4 cameras need kernel patches because they've got some nonstandard extensions)

Are you talking about the Ripsaw HD or an older (discontinued?) Ripsaw?

Since it's advertising uncompressed and advertises being "plug and play", I'm kind of surprised it's not quite working, unless there's an older Ripsaw that was not quite as standards-compliant?

dejanr, 2021-11-09:
> @dejanr - are you saying it appears as a UVC device but is not fully compatible/is glitchy? (similar to how RealSense cameras and Sony PS4 cameras need kernel patches because they've got some nonstandard extensions)

Exactly, most of the apps just freezes when they start using that camera input, for example zoom, wxcam etc...

> Since it's advertising uncompressed and advertises being "plug and play", I'm kind of surprised it's not quite working, unless there's an older Ripsaw that was not quite as standards-compliant?

I think they advertise it as a windows device, and they don't mention anywhere linux support.

PuregoEidriB, 2024-09-24:
Whos here 2024? :D Anyone working towards this anymore, its absolute disaster from Razer side tbh.

loonsthemoons, 2024-09-28:
> Whos here 2024? :D Anyone working towards this anymore, its absolute disaster from Razer side tbh.

basically a useless piece of tech at this point, would love to see this work but i dont think its ever gonna get support

terricain, 2024-09-28:
If someone fancies getting in touch with razer support and finds someone who's willing to work with us and get us a protocol spec, then I'll have a look, otherwise I don't really have any time to dedicate to this.

Hoerli1337, 2024-11-08:
Let me join the queue.
I also have a Ripsaw here that doesn't work with Linux.
I have to use Windows for the device to work at all :(

loonsthemoons, 2025-01-22:
> If someone fancies getting in touch with razer support and finds someone who's willing to work with us and get us a protocol spec, then I'll have a look, otherwise I don't really have any time to dedicate to this.

I just wrote the support a message, in hopes that the right person sees it, will keep you updated.