openrazer/openrazer #160: razer-daemon upgrade kept back

sranger, 2017-01-17:
I saw a new update for razer-daemon this morning but it was kept back. I attempted to "sudo apt-get install razer-daemon" but got the following message:

> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  razer-daemon : Depends: gir1.2-gtk3.0 but it is not installable
> E: Unable to correct problems, you have held broken packages.

However, there is a gir1.2-gtk-3.0; linked incorrectly in the package?

lah7, 2017-01-17:
[I was alerted to this typo earlier today](https://github.com/lah7/polychromatic/issues/49), it's been fixed.

@terrycain just needs to re-release new packages.

terrycain, 2017-01-19:
Fixed

closed 2017-01-19 by terrycain