openrazer/openrazer #162: Dependency issue on Ubuntu 14.04 with python3-setproctitle

ahiskali, 2017-01-21:
$sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:
    
    The following packages have unmet dependencies:
     razer-daemon : Depends: python3-setproctitle but it is not installable
    E: Unable to correct problems, you have held broken packages.
    
I get this error when trying to install.

lah7, 2017-01-22:
`python3-setproctitle` isn't available for 14.04 unfortunately. This package was introduced 16.04 onwards.

You could try installing this package manually using a deb for 16.04 (Xenial): http://packages.ubuntu.com/xenial/python3-setproctitle

terrycain, 2017-01-22:
ouch. @ahiskali See if the 16.04 deb works, if so i'll add that deb to the ppa

cdubz, 2017-01-23:
I kept forgetting to mention while we were doing all that Ornata testing that I had been unpacking and removing that dependency on 14.04, hah. It looks like the 16.04 deb won't fly because it depends on `python3 >= 3.5~`. [Trusty only gets 3.4](http://packages.ubuntu.com/trusty/python3) so it would require a bit of effort.

z3ntu, 2017-01-23:
You should be able to do `pip3 install setproctitle`

terrycain, 2017-01-23:
Repack the 16.04 deb lowering dependency

cdubz, 2017-01-23:
Yes, both of those options work. I installed via pip but that leaves the dependency unresolved for `dpkg` so the deb install still fails.

Repacking the 16.04 `python3-setproctitle` deb with the lower Python dependency would (I assume) also work. If it did would you be able to maintain a repacked deb on the ppa?

terrycain, 2017-01-23:
Well we woudlnt maintain it, it'll be a one off version as I also dont plan on supporting 14.04 forever. Plus that module, does one thing and one thing only, its unlikely that there will be any significant updates

cdubz, 2017-01-23:
Yeah, fair enough. I will test and confirm this tomorrow if the original poster doesn't come back and do so before then.

terrycain, 2017-01-23:
Package "should" appear here
https://launchpad.net/~terrz/+archive/ubuntu/razer-testing

ahiskali, 2017-01-23:
@cdubz please go ahead, I was trying to install 16.04 deb and fucked up dependencies, it will take time to fix.

terrycain, 2017-01-23:
PPA has build deb, here is direct link - https://launchpad.net/~terrz/+archive/ubuntu/razer-testing/+files/python3-setproctitle_1.1.9-1_amd64.deb

cdubz, 2017-01-24:
Confirmed working on 14.04 with the `razer-testing` ppa :+1:

terrycain, 2017-01-25:
Ok, am debating moving it as I believe it's a higher version than even the stock ubuntu one. So if I do everyone will update automatically. Will have a think

cdubz, 2017-01-25:
Yeah, [Xenial packages 1.1.8](http://packages.ubuntu.com/source/xenial/python-setproctitle). Is there any reason not to just repackage that version instead?

[Index of /ubuntu/pool/universe/p/python-setproctitle](http://archive.ubuntu.com/ubuntu/pool/universe/p/python-setproctitle/)

terrycain, 2017-01-25:
Requires too much effort with launchpad as a tar exists on launch pad of the normal version and then it complains.

terrycain, 2017-01-26:
After making version 1.1.7 (so it wouldnt get picked up by everyone) i realised that it would only be released on the trusty version of the PPA duh!. But setproctitle should be available from the main PPA soonish.

closed 2017-01-26 by terrycain