openrazer/openrazer #163: Errors while processing kernel-modules and python3-razer

kachmarma, 2017-01-22:
"Errors were encountered while processing:

/var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb
/var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb

E: Sub-process /usr/bin/dpkg returned an error code (1)

z3ntu, 2017-01-22:
Is there a more complete log? And what distro (+version) are you running?

kachmarma, 2017-01-22:
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.1 LTS"
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

Complete log when installing via Terminal:

kachmarma@Kachmar-RazerBlade:~$ sudo add-apt-repository ppa:terrz/razerutils
[sudo] password for kachmarma: 
 This PPA contains open source drivers that allow many Razer peripherals to utilise their full feature set and allow customisation via the daemon.

Supported devices include:

 - BlackWidow (Chroma, Classic, Ultimate)
 - BlackWidow Mamba
 - Blade Stealth
 - Firefly
 - Mamba
 - Abyssus
 - Tartarus

For more information, see here: https://github.com/terrycain/razer-drivers

Licensed under the GPL. These drivers are not affiliated with Razer, Inc.
 More info: https://launchpad.net/~terrz/+archive/ubuntu/razerutils
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmppz5hbj_d/secring.gpg' created
gpg: keyring `/tmp/tmppz5hbj_d/pubring.gpg' created
gpg: requesting key DC9CC596 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmppz5hbj_d/trustdb.gpg: trustdb created
gpg: key DC9CC596: public key "Launchpad PPA for Terry Cain" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
kachmarma@Kachmar-RazerBlade:~$ sudo apt update
Ign:1 cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial InRelease
Err:2 cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial Release
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Hit:3 http://us.archive.ubuntu.com/ubuntu xenial InRelease
Get:4 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Get:5 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Ign:6 http://dl.google.com/linux/chrome/deb stable InRelease                   
Get:7 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]  
Hit:8 http://dl.google.com/linux/chrome/deb stable Release                     
Hit:9 http://ppa.launchpad.net/linrunner/tlp/ubuntu xenial InRelease           
Hit:10 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial InRelease       
Get:11 http://us.archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [453 kB]
Hit:12 http://archive.canonical.com/ubuntu xenial InRelease                    
Get:14 http://us.archive.ubuntu.com/ubuntu xenial-updates/main i386 Packages [446 kB]
Get:15 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [381 kB]
Get:16 http://us.archive.ubuntu.com/ubuntu xenial-updates/universe i386 Packages [376 kB]
Reading package lists... Done                             
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:2 and /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:3
E: The repository 'cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:2 and /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:3
kachmarma@Kachmar-RazerBlade:~$ sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
[... 24 lines cut ...]
  libservlet3.1-java libswt-cairo-gtk-3-jni libswt-glx-gtk-3-jni
  libswt-gnome-gtk-3-jni libswt-gtk-3-java libswt-gtk-3-jni
  libswt-webkit-gtk-3-jni libtomcat7-java openjdk-8-jdk openjdk-8-jdk-headless
  sat4j
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-razer razer-kernel-modules-dkms
0 upgraded, 2 newly installed, 0 to remove and 7 not upgraded.
1 not fully installed or removed.
Need to get 0 B/54.5 kB of archives.
After this operation, 447 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 213672 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/dkms.conf', which is also in package razer-chroma-driver 1.0.0
Preparing to unpack .../python3-razer_1.1.8-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/razer/__init__.py', which is also in package razer-chroma-driver 1.0.0
Errors were encountered while processing:
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb
 /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

z3ntu, 2017-01-22:
Uninstall razer-chroma-driver first.

kachmarma, 2017-01-23:
kachmarma@Kachmar-RazerBlade:~$ sudo apt-get purge razer-chroma-driver
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 razer-daemon : Depends: razer-kernel-modules-dkms but it is not going to be installed
                Recommends: python3-razer but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
kachmarma@Kachmar-RazerBlade:~$

z3ntu, 2017-01-23:
Run `sudo apt-get purge razer-chroma-driver python3-razer razer-kernel-modules-dkms razer-daemon razer-doc`

z3ntu, 2017-01-25:
Did that work?

kachmarma, 2017-01-25:
kachmarma@Kachmar-RazerBlade:~$ sudo apt-get purge razer-chroma-driver python3-razer razer-kernel-modules-dkms razer daemon razer-doc
[sudo] password for kachmarma: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package razer
kachmarma@Kachmar-RazerBlade:~$ sudo add-apt-repository ppa:terrz/razerutils
 This PPA contains open source drivers that allow many Razer peripherals to utilise their full feature set and allow customisation via the daemon.

Supported devices include:

 - BlackWidow (Chroma, Classic, Ultimate)
 - BlackWidow Mamba
 - Blade Stealth
 - Firefly
 - Mamba
 - Abyssus
 - Tartarus

For more information, see here: https://github.com/terrycain/razer-drivers

Licensed under the GPL. These drivers are not affiliated with Razer, Inc.
 More info: https://launchpad.net/~terrz/+archive/ubuntu/razerutils
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmpffiej75g/secring.gpg' created
gpg: keyring `/tmp/tmpffiej75g/pubring.gpg' created
gpg: requesting key DC9CC596 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmpffiej75g/trustdb.gpg: trustdb created
gpg: key DC9CC596: public key "Launchpad PPA for Terry Cain" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
kachmarma@Kachmar-RazerBlade:~$ sudo apt update
Ign:1 cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial InRelease
Err:2 cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial Release
  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add new CD-ROMs
Hit:3 http://us.archive.ubuntu.com/ubuntu xenial InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease             
Hit:5 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease           
Ign:6 http://dl.google.com/linux/chrome/deb stable InRelease                   
Hit:7 http://archive.canonical.com/ubuntu xenial InRelease                     
Hit:8 http://ppa.launchpad.net/linrunner/tlp/ubuntu xenial InRelease           
Hit:9 http://security.ubuntu.com/ubuntu xenial-security InRelease              
Hit:10 http://dl.google.com/linux/chrome/deb stable Release             
Hit:11 http://ppa.launchpad.net/terrz/razerutils/ubuntu xenial InRelease
Reading package lists... Done                     
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:2 and /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:3
E: The repository 'cdrom://Ubuntu 16.04.1 LTS _Xenial Xerus_ - Release amd64 (20160719) xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:2 and /etc/apt/sources.list.d/linrunner-ubuntu-tlp-xenial.list:3
kachmarma@Kachmar-RazerBlade:~$ sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
razer-doc is already the newest version (1.1.8-0ubuntu1).
razer-daemon is already the newest version (1.1.8-0ubuntu1).
The following packages were automatically installed and are no longer required:
  ant ant-optional aspectj default-jdk default-jdk-headless
  eclipse-platform-data eclipse-rcp fastjar jarwrapper junit junit4
  libapache-pom-java libasm3-java libasm4-java libaspectj-java libbonoboui2-0
  libbonoboui2-common libcglib3-java libcommons-beanutils-java
  libcommons-cli-java libcommons-codec-java libcommons-collections3-java
  libcommons-compress-java libcommons-dbcp-java libcommons-digester-java
  libcommons-httpclient-java libcommons-logging-java libcommons-parent-java
  libcommons-pool-java libdb-java libdb-je-java libdb5.3-java
  libdb5.3-java-jni libeasymock-java libecj-java libequinox-osgi-java
  libfelix-bundlerepository-java libfelix-gogo-command-java
  libfelix-gogo-runtime-java libfelix-gogo-shell-java libfelix-osgi-obr-java
[... 8 lines cut ...]
  libswt-gnome-gtk-3-jni libswt-gtk-3-java libswt-gtk-3-jni
  libswt-webkit-gtk-3-jni libtomcat7-java openjdk-8-jdk openjdk-8-jdk-headless
  sat4j
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-razer razer-kernel-modules-dkms
0 upgraded, 2 newly installed, 0 to remove and 33 not upgraded.
1 not fully installed or removed.
Need to get 0 B/54.5 kB of archives.
After this operation, 447 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 213672 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/dkms.conf', which is also in package razer-chroma-driver 1.0.0
Preparing to unpack .../python3-razer_1.1.8-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.8-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/razer/__init__.py', which is also in package razer-chroma-driver 1.0.0
Errors were encountered while processing:
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.8-0ubuntu1_amd64.deb
 /var/cache/apt/archives/python3-razer_1.1.8-0ubuntu1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
kachmarma@Kachmar-RazerBlade:~$

z3ntu, 2017-01-25:
Are you serious...? Just read the output of the commands...

kachmarma, 2017-01-25:
Sorry, 1) new to Linux, 2) was in class. Got it to work. Question, what does this mean: 'sudo gpasswd -a <USERNAME> plugdev'

kachmarma, 2017-01-25:
Thank you very much btw

terrycain, 2017-01-25:
disable secureboot

kachmarma, 2017-01-25:
I've disabled it multiple times

z3ntu, 2017-01-25:
That error is a secure boot error afaik. Try reinstalling the driver

z3ntu, 2017-02-22:
@kachmarma Did anything change since your last message?

closed 2017-02-27 by z3ntu