openrazer/openrazer #166: issue with ripple effect on Late 2016 RBS

nickdesaulniers, 2017-01-22:
Hi there, I've tried both razerCommander and polychromatic and both seem to have issues with the Ripple effect.  When I enable it in either, I see the following in `dmesg`:

```
[  252.870405] razerkbd: Command not supported. Start Marker: 05 id: ff Num Params: 46 Reserved: 03 Command: 0b Params: ff050015000000000000000000000000 .
```
repeatedly.

```sh
nick@nick-Blade-Stealth:~$ lsusb | grep '1532:'
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd
```

I have installed the drivers via:

```sh
nick@nick-Blade-Stealth:~$ sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
```

terrycain, 2017-01-23:
ooh, Dont suppose your able of getting some packet captures of razer synapse in a vm doing ripple?
https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

z3ntu, 2017-02-22:
@nickdesaulniers Could you please get some packet captures of the ripple so we can fix this?

nickdesaulniers, 2017-02-26:
I'm having trouble install a working Windows 7 image in virtual box: 
[image]

nickdesaulniers, 2017-02-26:
looks like [this](https://askubuntu.com/questions/165665/installing-win7-x64-in-virtualbox-error-driver-device-missing)

nickdesaulniers, 2017-02-26:
actually I think my iso is bad, let me grab a new copy

lezed1, 2017-03-07:
Here is a capture of a few different ripples on the RBS late 2016. It starts with red (`0xFF0000`), green (`0x00FF00`) and blue (`0x0000FF`). It ends with a ripple of (`0xDDDDDD`).

From looking at the packets, this seems to just be using the individual key settings, so this may be solved by getting individual lighting working (#223).

[rbs-late-2016-ripple.pcapng.zip](https://github.com/terrycain/razer-drivers/files/824390/rbs-late-2016-ripple.pcapng.zip)

lezed1, 2017-03-07:
@z3ntu I'm also happy to try to write the patch myself. I think you mentioned there is a Telegram that would be helpful for that. If so, my username is lezed1 there too.

coreyog, 2017-07-31:
I originally opened [a bug](https://github.com/lah7/polychromatic/issues/110) under Polychromatic about this on my Ornata. I was given some python to be sure it wasn't Polychromatic's issue and the python never produced a ripple implying it's the daemon.

Simply adding this as a record that this affects the Ornata as well.

terrycain, 2017-08-05:
Ok test this

```
import razer.client
a = razer.client.DeviceManager()
b = a.devices[0]          // Assuming it's the only device
b.name
b.brightness = 100.0
b.fx.advanced.matrix[4,5] = (255, 0, 0)
b.fx.advanced.draw()
```

Does that light up a key near the spacebar?

SammDu, 2017-08-05:
yes the 'V' key lighted up red
btw im running py3.5.3

terrycain, 2017-08-05:
Ok well thats good for me and bad for you. It means that custom mode is not broken but some logic in my ripple code is, and as i dont wanna touch that mess of code yet its getting pushed down the list ;-)

lezed1, 2018-01-30:
to my knowledge ripple still doesn't work on my RBS. I just had to ship it back to Razer for a Core issue, so I can't test anything for ~1 week, but I can look into this when I have a device again! (I'm mostly making this comment to appease the StaleBot).

kenzierocks, 2018-01-30:
I have a 2017 Razer Blade with which the effect does not work either. I can do some testing if needed.

z3ntu, 2018-01-30:
Could you try #514 ?

PR #514 (2018-01-30): Fix ripple effect on non-standard keyboards
Until now, ripple only worked on keyboards with a (6, 22) matrix.
Blade keyboards have different sizes (e.g. (6, 16) or (6, 25)) and they
should be supported.

~~This is not tested yet but it could work.~~

Fixes #166 and fixes #353
closed 2018-01-30 by z3ntu