openrazer/openrazer #168: Can not run or configure anything, orbweaver is not detected

vitacell, 2017-01-24:
Arch gnulinux:
Linux Maquinon64 4.8.13-1-ARCH #1 SMP PREEMPT Fri Dec 9 07:24:34 CET 2016 x86_64 GNU/Linux

Running MATE with gtk3

+razer orbveawer and deathadder chroma

[vita@Maquinon64 ~]$ lsusb | grep '1532:'
Bus 002 Device 004: ID 1532:0043 Razer USA, Ltd 
Bus 001 Device 004: ID 1532:0113 Razer USA, Ltd RZ07-0074 Gaming Keypad [Orbweaver]

The razer-service is always running, already tryed everything. "razercfg" and "razerd" always worked fine for configure DA 3g and  DA Chroma. I uninstalled them, but now I am in trouble with all that razer-driver stuff. I can run razerCommander, only DA chroma is detected, and I can see only light value, nothing more, when I apply setting (even nothing is changed), in devices it adds another DA chroma after every apply.

[root@Maquinon64 vita]# pacman -Q | grep 'notify'
libnotify 0.7.7-1
notify-sharp 0.4.1-2
python-notify2 0.3-2
[root@Maquinon64 vita]# dbus-lauch /usr/bin/razercommander
bash: dbus-lauch: no se encontró la orden
[root@Maquinon64 vita]# dbus-launch /usr/bin/razercommander
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
[root@Maquinon64 vita]# razercommander
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 18, in __init__
    session_bus = _dbus.SessionBus()
  File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib/python3.6/site-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
[... 193 lines cut ...]
Exception: can't set attribute

[vita@Maquinon64 ~]$ /usr/bin/razercommander
WARNING:root:Brightness is not available
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 527, in on_universalApplyButton_clicked
    currentFX = fxListBox.get_selected_row().value
AttributeError: 'NoneType' object has no attribute 'value'
WARNING:root:Brightness is not available
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 527, in on_universalApplyButton_clicked
    currentFX = fxListBox.get_selected_row().value
AttributeError: 'NoneType' object has no attribute 'value'
WARNING:root:Brightness is not available
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 527, in on_universalApplyButton_clicked
    currentFX = fxListBox.get_selected_row().value
AttributeError: 'NoneType' object has no attribute 'value'
WARNING:root:Brightness is not available
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 527, in on_universalApplyButton_clicked
    currentFX = fxListBox.get_selected_row().value
AttributeError: 'NoneType' object has no attribute 'value'

---------------------------------------------------------------------------------------------

vitacell, 2017-01-24:
EDIT: in razerCommander, every time I push "update" button, it adds another DAchroma to devices

z3ntu, 2017-01-24:
The AUR package isn't up-to-date thanks to @GabMus .

vitacell, 2017-01-24:
Hi, thanks for answer.

What package exactly? razercommander-git 1.0.3-10? what do I need for fix all those problem, it seems that here is more than 1 package that fails or unstable.

z3ntu, 2017-01-24:
razercommander is just a front-end to the driver. The driver consist out of `razer-driver-dkms`, `razer-daemon` and `python-razer`.

z3ntu, 2017-01-25:
Did updating to 1.1.8 fix the issue?

vitacell, 2017-01-25:
[vita@Maquinon64 ~]$ pacman -Q | grep "razer"
python-razer 1.1.8-2
razer-daemon 1.1.8-3
razer-driver-dkms 1.1.8-1
razercommander-git 1.0.3-10

Still same problems, nothing works. I thinking to remove that stuff at the moment, and installing back "razercfg". Polychromatic doesn't work too. GTK is bugged. What should I do? can you help me?

z3ntu, 2017-01-25:
```
ls -lah /sys/bus/hid/drivers/
ls -lah /sys/bus/hid/drivers/razerkbd/
ls -lah /sys/bus/hid/drivers/razerkbd/0*/
```

vitacell, 2017-01-25:
[vita@Maquinon64 ~]$ ls -lah /sys/bus/hid/drivers/
total 0
drwxr-xr-x 5 root root 0 ene 25 16:50 .
drwxr-xr-x 4 root root 0 ene 25 00:16 ..
drwxr-xr-x 2 root root 0 ene 25 00:16 hid-generic
drwxr-xr-x 2 root root 0 ene 25 16:50 konepure
drwxr-xr-x 2 root root 0 ene 25 00:16 razermouse
--------------------------------------------------------------------
[vita@Maquinon64 ~]$ ls -lah /sys/bus/hid/drivers/razerkbd/
ls: no se puede acceder a '/sys/bus/hid/drivers/razerkbd/': No existe el fichero o el directorio
(Can not to access '/sys/bus/hid/drivers/razerkbd/': file or directory does not exist)

same for: ls -lah /sys/bus/hid/drivers/razerkbd/0*/
----------------------------------------------------------------------

http://imgur.com/a/zpY9Z

z3ntu, 2017-01-25:
@vitacell Go to the telegram search field and enter `@z3ntu`.

vitacell, 2017-01-25:
Seems to be some problem of drivers, because I have no output of 
"sudo modprobe razerkbd".

z3ntu, 2017-01-31:
Can be closed now (conversation was on Telegram).

closed 2017-01-31 by terrycain