openrazer/openrazer #169: Device Support: Razer Core

CalcProgrammer1, 2017-01-26:
If you have a Razer Core, it is now possible to use it on your Blade/Blade Stealth in Linux via a BIOS
mod described here to disable the unsupported Thunderbolt security:

https://www.reddit.com/r/linux_gaming/comments/5q2q80/razer_core_working_on_ubuntu_1610_razer_blade/

I modded my early 2016 RBS BIOS and got the Core working.  It presents a USB device for the lighting controller.  I will try to capture some packets this weekend and will try to work on a driver based on the Firefly/Chroma Mug code.  I expect it to be similar to the Firefly with 9 LEDs (inside LED strip is all one zone, then 8 individually controlled ones on the front).  It definitely supports custom effects through the official SDK.  I am not aware of any other funtionality provided by the Core's USB device, as the USB hub is just a hub, the Ethernet port is a separate USB 3.0 ASIX gigabit controller, and the GPU is presented on the PCIe bus.

terrycain, 2017-01-26:
Awesome dude. :smile: Be sure to drop some pcaps here ;-) would be the first core i've seen

CalcProgrammer1, 2017-01-26:
Added the Core's USB ID to the firefly driver and all the basic effects work, but custom effect fails.  Going to capture some packets.

Just a suggestion - It would be nice if you consolidated the various "light only" devices into one driver, maybe call it razerchroma or something, to contain Firefly, Core, Mug, and any new devices that don't provide anything other than lighting (the mug has a switch I guess, but that could just be one device-specific case).  It seems like it would make adding new devices easier than copy-pasting the entire Firefly driver and renaming razerfirefly, updating the Makefile, etc.

CalcProgrammer1, 2017-01-27:
[image]
[image]

First capture I sent all white (FF FF FF) to the Core, second capture I sent all red (FF 00 00) except for the side LED zone which was all white (FF FF FF).

CalcProgrammer1, 2017-01-27:
Ok, so it requires the razer_chroma_standard_matrix_set_custom_frame function rather than the razer_chroma_misc_one_row_set_custom_frame one, and does take a row ID in addition to a column start and stop ID.  I can drive it with the standard matrix function.  I'll go ahead and write a new module for it.

z3ntu, 2017-01-27:
[image]

> **Razer Core**
> * 1 zone for the internal lighting
> * 8 zones for the underglow

terrycain, 2017-01-28:
> Ok, so it requires the razer_chroma_standard_matrix_set_custom_frame function rather than the razer_chroma_misc_one_row_set_custom_frame one, and does take a row ID in addition to a column start and [...]

Nice.

> Just a suggestion - It would be nice if you consolidated the various "light only" devices into one driver, maybe call it razerchroma or something, to contain Firefly, Core, Mug, and any new devices [...]

And yeah it would be great but in the end it'll be far more hassle to move to that, and then comes the fact that I have maybe 5/30odd supported devices and would rather not break everything. I get enough issues raised as it is :wink:

commit d6aa71e7 (2017-03-04): Merge branch 'master' into stable

Closes #212
Closes #184
Closes #169
Closes #149
Closes #116
closed 2017-03-04 by terrycain