openrazer/openrazer #172: Kraken Chroma 7.1 v2 Custom Effect

CalcProgrammer1, 2017-01-28:
One of the big differences between v1 and v2 is that v2 supports custom effects in the official SDK.  If you send two static effects in a row, the firmware fades between the current color and the new color.  Applying a custom effect changes the color instantly, which is better when you're driving the effect from an application such as my visualizer.  I will try and capture some packets while sending custom effects using my visualizer in Windows and see what needs changed to support custom effects on the v2.  The ManOWar may also support custom effects, see https://github.com/CalcProgrammer1/KeyboardVisualizer/issues/62.  The v1 does not according to Razer.

CalcProgrammer1, 2017-01-28:
Ok, it seems to be alternating between these two commands rapidly when I run keyboard visualizer commanding white (and using custom effects).

[image]
[image]

CalcProgrammer1, 2017-01-28:
I got it working.  It looks like you already had the custom address discovered in your header file, but no implementation to write to it.  I went ahead and added the ability to send RGB+Intensity because I screwed up and set intensity to zero and needed a way to undo that.  I will submit a PR soon, going to test with the v1 and see if it is possible to use custom effects on it, but it probably isn't.

terrycain, 2017-01-28:
Yeah i have a list of the addresses, I just was too lazy at the time to implement it :wink:

z3ntu, 2017-02-28:
Uhm this is merged, right? https://github.com/terrycain/razer-drivers/pull/173

terrycain, 2017-02-28:
Yeah

closed 2017-02-28 by terrycain