openrazer/openrazer #178: ImportError: No module named 'setuptools'

materangai, 2017-01-30:
Sorry, this make be a basic one, tried copying 'razer' in 'pylibs' to python dir but that made no difference.

I get

 :: Installing Razer python library
=====================================================
Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from setuptools import setup, find_packages
ImportError: No module named 'setuptools'
Makefile:7: recipe for target 'install' failed
make[1]: *** [install] Error 1
Makefile:96: recipe for target 'python_library_install' failed
make: *** [python_library_install] Error 2

Thanks!!

z3ntu, 2017-01-31:
What distro are you running? You shouldn't install the driver manually if distribution packages are available. Anyways, you have to install python(3)-setuptools (package name depends on distro)

materangai, 2017-01-31:
I am running Kali which is based of Debian Jessie so can't add the PPA (unless there is a repo I can add?)
I managed to install "python3-setuptools" which seems to run okay but then went to install polychromatic to test and got  polychromatic : Depends: razer-chroma-driver but it is not installable

materangai, 2017-01-31:
Ah found away around that. Thanks for your help :)

closed 2017-01-31 by materangai

hoco21, 2017-07-01:
what was the way you found around it?

VolkerAH, 2018-02-01:
I would like to know away around it too, since I am running into it on gentoo. 

*  dev-python/setuptools
      Latest version available: 36.7.2
      Latest version installed: 36.7.2

is installed, but it just fails with the error, OP posted...

z3ntu, 2018-02-01:
@VolkerAH As I mentioned in my last comment, you are missing the python3 version of setuptools according to the error message.

VolkerAH, 2018-02-01:
no, I did not missed the python3 version

But the version was installed into python3.5 and the interpreter was set to python3.4

I just figured that out a minute ago, all is fine now ;)