openrazer/openrazer #179: razerkbd: module verification failed, kernel tainted.

KeiroD, 2017-02-01:
Hi,

Just wanted to let you know that it appears that `razerkbd` is not working properly.

**OS stats**

Ubuntu 16.10 x86_64
DE: Cinnamon
Razer device: Deathstalker Essential
USB ID: 
```
lsusb | grep '1532:'
Bus 003 Device 002: ID 1532:011f Razer USA, Ltd
```

I see this in `/var/log/syslog` with the following text:

```
kernel: [843124.164865] razerkbd: loading out-of-tree module taints kernel.
kernel: [843124.164917] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
```
So uh... you may want to look into that.

Also, some searching on this reveals that #98 may be at play as well. Any steps I should take to try helping with troubleshooting? Obviously, with 16.10 and UEFI/Secure Boot... far as I'm aware of, Secure Boot is not enabled on this computer but UEFI is...

Additional note: I never got the prompt to disable secure boot, so that seems to indicate that secure boot is not enabled here.

KeiroD, 2017-02-01:
Having a little trouble telling which one is the mouse driver for the keyboard, here's what's listed in `/sys/bus/hid/devices`

```
root@EmberAIO:/sys/bus/hid/devices# ls -lah
total 0
drwxr-xr-x 2 root root 0 Feb  1 06:57 .
drwxr-xr-x 4 root root 0 Jan 25 03:24 ..
lrwxrwxrwx 1 root root 0 Feb  1 06:57 0003:1532:011F.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1532:011F.0001
lrwxrwxrwx 1 root root 0 Feb  1 06:57 0003:1532:011F.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:1532:011F.0002
lrwxrwxrwx 1 root root 0 Feb  1 06:57 0003:1532:011F.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.2/0003:1532:011F.0003
lrwxrwxrwx 1 root root 0 Feb  1 06:57 0003:17EF:602E.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:17EF:602E.0004
lrwxrwxrwx 1 root root 0 Feb  1 06:57 0003:2149:5439.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:2149:5439.0005
```

z3ntu, 2017-02-01:
It's most probably a secure boot issue

terrycain, 2017-02-01:
what does `sudo modprobe razerkbd` output?

CalcProgrammer1, 2017-02-01:
Pretty sure that warning happens any time you use an out-of-tree kernel module.  It's a harmless warning if you know the source of said module.  I get it too, secure boot is definitely off on both my laptops where I've seen it.

KeiroD, 2017-02-01:
@terrycain `modprobe razerkbd` doesn't return an error. However, I do see that `razrkbd` has been loaded, but the software doesn't seem to recognize my keyboard, which is obvious and is as it should be, since there's no "official" support for this keyboard.

@CalcProgrammer1 I had suspected as much.

@z3ntu pretty sure I don't have Secure Boot on. The option is available, but it is most definitely off. I remember being frustrated with Secure Boot and the inability to install Ubuntu onto this computer with it on, along with UEFI being on so I turned it off until relatively recently where I had no issues in installing Ubuntu with Secure Boot off and UEFI on.

z3ntu, 2017-02-01:
Well yes, the keyboard isn't supported yet ;)

KeiroD, 2017-02-01:
@z3ntu Pretty sure I posted the output of `lsusb -d 1532:011f -v` earlier.

`/dev/input/by-id` output is as follows:

```
root@EmberAIO:/tmp# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 Jan 22 12:24 .
drwxr-xr-x 4 root root 440 Jan 31 07:13 ..
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-17ef_USB_Optical_Mouse-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-17ef_USB_Optical_Mouse-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Feb  1 07:09 usb-Generic_LENOVO_LBG_1080P_CAM_200901010001-event-if00 -> ../event13
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-Razer_Razer_DeathStalker_Essential_2014-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-Razer_Razer_DeathStalker_Essential_2014-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-Razer_Razer_DeathStalker_Essential_2014-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-Razer_Razer_DeathStalker_Essential_2014-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Feb  1 07:09 usb-Weida_Hi-Tech_CoolTouchR_System-event-if00 -> ../event9
```

terrycain, 2017-02-01:
Ahh yeah I wasn't paying much I've gotten that once or twice. Can be ignored. And yeah rename issue to something more like "can you support my device x plz", that way it's easily recognisable when I scan through them.

What options does kb have? Pics from Razer synapse would be useful.

KeiroD, 2017-02-01:
Ah... how would I get pics from Razer Synapse? Wait, you mean on Windows? Or...?

KeiroD, 2017-02-02:
So it looks like the lighting option no longer exists for the Deathstalker Essential.

https://cdn.discordapp.com/attachments/194864223958859776/276524129354645504/razer.png

Here's the pic as requested.

terrycain, 2017-02-02:
What lighting options did it use to have?

KeiroD, 2017-02-03:
It had the option to do a breathing mode.

F11/12 are not brightness keys. If I recall correctly, it was controlled by Razer Synapse. I won't be able to get a replacement until next month at the latest. However, I'll see what I can do to get the lighting options to show up again in Windows Synapse.

terrycain, 2017-02-03:
Ok your on ubuntu, gimme a couple of days, juding by the really old usb id, i can take a few educated guesses.

terrycain, 2017-02-03:
Actually isnt the 2014 essential not backlit? The expert has backlight but as far as im aware the essential doesnt

KeiroD, 2017-02-24:
@terrycain apparently I didn't get the notification that you'd responded... and I took a look at the Amazon reviews for the keyboard I ordered. It appears it does indeed have backlight support: https://www.amazon.com/Razer-DeathStalker-Essential-Gaming-Keyboard/dp/B00HKEI4HU/

So I'm not sure what happened to the backlight options...

z3ntu, 2017-02-24:
> The only difference between both models its that the essential version has no Green LED backlit keys and the Expert version its fully illuminated with the green backlit keys. 

comment from the amazon page you linked

KeiroD, 2017-02-24:
Alright, well, in that case, I suppose the only thing we'd need to have is driver support. I'd love to be able to pull and modify profiles on the keyboard.

terrycain, 2017-02-24:
The keyboards no longer store and retrieve macros, its all done in software now

jenden0, 2019-02-13:
Resurrecting this as I recently picked up this keyboard used and would love to see support.  Happy to supply any needed information/testing.

As there seemed to be some confusion, the device that I have (1532:011f) doesn't have backlight.

z3ntu, 2019-03-18:
@jenden0 Yes, we need packet captures, see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol for more details.

jenden0, 2019-03-27:
Ok, I've gone through what I think are some of the major UI elements of synapse and gotten packet captures as per the doc.  One thing that I wasn't clear on in the doc is whether I should also get some packet captures from the keybord itself (activating gaming mode with Fn+F10 for example).  If there are other captures in particular that would be useful (profile switching, key assignments?) please let me know, I'll try to get some more in the next week or so, but not sure when I'll have time.
[deathstalker_essential_2014_caps.zip](https://github.com/openrazer/openrazer/files/3013238/deathstalker_essential_2014_caps.zip)
File names are hopefully somewhat self explanatory, but a brief description of each capture.  Each action was captured twice (with the second capture ending in 2)
- gaming_on_off_on_off
 - In the gaming mode screen, turned gaming mode on, then off, then on again, then off again
- alt_tab_on_off_on_off, alt_f4_on_off_on_off
 - In the gaming mode screen with gaming mode already enabled, toggled disabling of alt-tab/alt-f4 on, off, on ,off
- link_fn_1_wmplayer
 - in the customize screen, with fn+1 selected in the shortcut menu, clicked the link executable checkbox and selected the wmplayer exe from the virtual machine
- unlink_fn_1_wmplayer
 - in the state resulting from link_fn_1_wmplayer, uncheck the link executable checkbox
- create_macro_1
 - press the + button on the macro screen to create a new macro (default name "Macro 1")
- delete_macro_1
 - press the trash icon to remove the macro created above
- rename_macro_test
 - Rename the "Macro 1" to "test".  Note that the "test" input characters are coming from the keyboard being captured
- record_a_b_c_with_delay
 - With the top most radio button (record delay) selected, start recording macro and type characters a, b, and c then stop recording
- record_a_b_c_with_default_delay_0.1
 - Same as above but instead of record delay default delay is selected and set to 0.1 seconds
- record_a_b_c_with_no_delay
 - Same as above but with the third option (no delay) selected
- record_ctrla_b_ctrlc_default_delay_0.1
 - same as record_a_b_c_with_default_delay_0.1 but the left ctrl key is held down while entering a and c (and released for b)
 - delete_...
  - Delete the macro contents by pressing the trash icon above the macro contents box, starting with the first entry (a)