openrazer/openrazer #180: Fedora 25 Razer Stealth(early 2016)

Afl0iv, 2017-02-01:
I have done the installation and gotten into the play dev, and can confirm that the Daemon is running. Though when it comes time to use a supported program I get issues for " device not being found" . 
If i try to bind the keyboard i get issues all around.  cannot move the razerkbd to where is should be
I get issues like this, 
[image]

z3ntu, 2017-02-01:
1. What distro is this?
2. How did you install the kernel driver & daemon & pylib?
3. Output of `ls -al /sys/bus/hid/drivers/{razerkbd,razerkbd/0*}/` (please use pastebin or a similar service)

Afl0iv, 2017-02-01:
1. This is the Fedora 25 workstation using GNOME. 
2. `$ sudo dnf config-manager --add-repo http://download.opensuse.org/repositories/home:z3ntu/Fedora_25/home:z3ntu.repo
 $sudo dnf install razer-drivers kernel-devel`
3. `$ ls -al /sys/bus/hid/drivers/{razerkbd,razerkbd/0*}/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/': No such file or directory
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory`

z3ntu, 2017-02-01:
Good...
Did you try replugging your keyboard? If not do that and do `sudo modprobe razerkbd`, if that fails `sudo dkms status razer_chroma_driver/1.0.0` and post the output of the commands.

Afl0iv, 2017-02-01:
Sorry, new to the comment style of git. 
The `sudo modprobe razerkbd`   failed

for `sudo dkms status razer_chroma_driver/1.0.0`
      sudo: dkms: command not found
I Should have mention(sorry about that), That while installing the dkms-2.3-3.1.noarch had failed to install,  I tried to install it though yum and i still get errors for installing the dkms. 
I get :
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
/var/tmp/rpm-tmp.rlf1gU: line 1: fg: no job control
error: %prein(dkms-2.3-3.1.noarch) scriptlet failed, exit status 1
Error in PREIN scriptlet in rpm package dkms
Error in PREIN scriptlet in rpm package dkms
dkms-2.3-3.1.noarch was supposed to be installed but is not!
  Verifying   : dkms-2.3-3.1.noarch                                         1/1
 
Failed:
  dkms.noarch 2.3-3.1

z3ntu, 2017-02-01:
That's obviously the problem... No idea why it fails though... I hope I remember tomorrow to find out why that error happens

ychen0113, 2017-02-02:
i have the same issue on razer blade stealth (early 2016) running fedora 25

Afl0iv, 2017-02-02:
Any way to help resolve the issue ?

z3ntu, 2017-02-03:
As I am not really into Fedora or anything with rpm I can't really help... I didn't experience that particular issue with Fedora so it's quite hard to fix the issue. But as the %prein thing for the dkms package fails it's not really our problem here but a problem with Fedora ;) If there is some kind of forum or mailing list, please ask there with all information you have.

Afl0iv, 2017-02-03:
So is there any other way to install the dkms? Or would we have to complain to ferdoa to get the issue %prein fixed

z3ntu, 2017-02-03:
Not that I know of. If it's possible please try a clean VM and just install dkms in there and see if it fails.

Afl0iv, 2017-02-03:
I will let you know how that goes some time today.

z3ntu, 2017-02-03:
Updated answer: Please try again. Apprently the opensuse spec file isn't compatible with Fedora and as Fedora provide the dkms package themselves I apparently overwrote the working official one with a non-working unofficial one. Sorry for the inconvenience :( Either you can now do `sudo dnf remove dkms` and reinstall the driver or some other command to reinstall the dkms package (if you know, please share it with others)

Original answer: Okay so I can reproduce the issue and I am pretty sure that it has to do with the dkms 2.3 update which happened a few days ago. I neither created the spec file for the package (even though it's in my repo) nor do I have any idea what's going on unfortunately... It's the spec file from the bumblebee project which I "forked"/branched

Afl0iv, 2017-02-03:
Hey Z3ntu, 
We hit another issue, So i did the `sudo dnf remove dkms` to make sure, I went a head and did `sudo dnf install dkms`  and what outputs is:
---------------------------------------------------------------------------------------------------------------------------
Total download size: 106 k
Installed size: 223 k
Is this ok [y/N]: y
Downloading Packages:
[MIRROR] dkms-2.3-3.1.noarch.rpm: Status code: 404 for http://download.opensuse.org/repositories/home:/z3ntu/Fedora_25/noarch/dkms-2.3-3.1.noarch.rpm
[FAILED] dkms-2.3-3.1.noarch.rpm: No more mirrors to try - All mirrors were already tried without success
 
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: Error downloading packages:
  Cannot download noarch/dkms-2.3-3.1.noarch.rpm: All mirrors were tried
-------------------------------------------------------------------------------------------------------------------------
If you run Root and use yum, and type 'yum install dkms`
it does a successful install of the dkms : 

 ` # yum install dkms`
Redirecting to '/usr/bin/dnf install dkms' (see 'man yum2dnf')

home:z3ntu (Fedora_25)                                                                                      7.2 kB/s | 6.8 kB     00:00    
Dependencies resolved.
============================================================================================================================================
 Package                  Arch                       Version                                              Repository                   Size
============================================================================================================================================
Installing:
 dkms                     noarch                     2.3-1.20161202gitde1dca9.fc25                        updates                      81 k

Transaction Summary
============================================================================================================================================
Install  1 Package

Total download size: 81 k
Installed size: 222 k
Is this ok [y/N]: y 
Is this ok [y/N]: y 
Is this ok [y/N]: y
Downloading Packages:
dkms-2.3-1.20161202gitde1dca9.fc25.noarch.rpm                                                               394 kB/s |  81 kB     00:00    
--------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                        78 kB/s |  81 kB     00:01     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Installing  : dkms-2.3-1.20161202gitde1dca9.fc25.noarch                                                                               1/1 
  Verifying   : dkms-2.3-1.20161202gitde1dca9.fc25.noarch                                                                               1/1 

Installed:
  dkms.noarch 2.3-1.20161202gitde1dca9.fc25                                                                                                 

Complete!

ychen0113, 2017-02-03:
ok that worked great. after removing dkms i just reinstalled the razer drivers and now poly recolonizes my stealth. thx!

z3ntu, 2017-02-03:
Great to hear 😄 👍

z3ntu, 2017-02-03:
@ychen0113 Question: how did you install polychromatic? I could theoretically create a package for easier install... (and put it into the same repo) :)

ychen0113, 2017-02-03:
sudo git clone https://github.com/lah7/polychromatic.git
cd polychromatic
sudo git checkout stable
sudo ./install/install.sh

z3ntu, 2017-02-03:
Okay thanks. I will try to create a package in the near future for easier install 👍

Afl0iv, 2017-02-03:
I still get errors, how did you remove  the  polychromatic program, i am still getting device not found after the reinstalling the drivers 
??

ychen0113, 2017-02-03:
i never removed poly. if you still have issues then you prob didnt install razer-drivers correctly, i would remove then reinstall them.

Afl0iv, 2017-02-04:
Alright, Could you  show me the way you did it? if you dont mind.

Afl0iv, 2017-02-04:
I seem to be doing something wrong here and i cant figure out what it is. 
So i, remove the drivers with, `sudo dnf remove dkms` . After  they are all removed i run the command `sudo dnf install razer-drivers -y`. I go to run poly and its still not found.. 

I have also gone and removed the driver kernels( including devel) and reinstalled the way  it says to for fedora,  which is 
`$ sudo dnf config-manager --add-repo http://download.opensuse.org/repositories/home:z3ntu/Fedora_25/home:z3ntu.repo`
`$sudo dnf install razer-drivers kernel-devel`

And still the same issue... I cannot figure it out.. I have even re installed poly at one point... :/

z3ntu, 2017-02-04:
I am pretty sure that you have secure boot enabled. Either you disable secure boot (which is easier) or you follow https://docs.fedoraproject.org/en-US/Fedora/22/html/System_Administrators_Guide/sect-signing-kernel-modules-for-secure-boot.html (or a similar guide to sign dkms modules)

Afl0iv, 2017-02-04:
I got it to work, taking it off of safe boot fixed it, Thank you very much

z3ntu, 2017-02-04:
Please close the issue if it's fixed for you :)

closed 2017-02-04 by terrycain