openrazer/openrazer #182: make ubuntu_install breaks python3 on Ubuntu 16.10

maciek134, 2017-02-02:
These two lines in Makefile break Ubuntu's python3 installation (and in turn make it impossible to even reinstall python3, as the prerm scripts require working python3):

```
	mv $(DESTDIR)/usr/lib/python3.* $(DESTDIR)/usr/lib/python3
	mv $(DESTDIR)/usr/lib/python3/site-packages $(DESTDIR)/usr/lib/python3/dist-packages
```

The second one also fails because it can't find `.../site-packages`. Output of `sudo apt-get install --reinstall python3` after running `sudo make ubuntu_install`:

```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x00007f2dae819700 (most recent call first):
Aborted (core dumped)
*** Error in `/usr/bin/dpkg': munmap_chunk(): invalid pointer: 0x000055cd89160ce8 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x790cb)[0x7f76b72450cb]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1f8)[0x7f76b7252338]
/usr/bin/dpkg(+0x20cc0)[0x55cd87c50cc0]
/usr/bin/dpkg(+0x21109)[0x55cd87c51109]
/usr/bin/dpkg(+0x283bd)[0x55cd87c583bd]
/usr/bin/dpkg(+0x17727)[0x55cd87c47727]
/usr/bin/dpkg(+0x17905)[0x55cd87c47905]
/usr/bin/dpkg(+0x17b4d)[0x55cd87c47b4d]
/usr/bin/dpkg(+0xaed7)[0x55cd87c3aed7]
/usr/bin/dpkg(+0x20bfb)[0x55cd87c50bfb]
/usr/bin/dpkg(+0x20df1)[0x55cd87c50df1]
/usr/bin/dpkg(+0xa962)[0x55cd87c3a962]
/usr/bin/dpkg(+0x72fd)[0x55cd87c372fd]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7f76b71ec3f1]
/usr/bin/dpkg(+0x743a)[0x55cd87c3743a]
======= Memory map: ========
55cd87c30000-55cd87c75000 r-xp 00000000 103:02 3145991                   /usr/bin/dpkg
55cd87e75000-55cd87e78000 r--p 00045000 103:02 3145991                   /usr/bin/dpkg
55cd87e78000-55cd87e79000 rw-p 00048000 103:02 3145991                   /usr/bin/dpkg
55cd87e79000-55cd8808d000 rw-p 00000000 00:00 0 
55cd88cbc000-55cd8b727000 rw-p 00000000 00:00 0                          [heap]
7f76b5cd6000-7f76b5cec000 r-xp 00000000 103:02 266725                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f76b5cec000-7f76b5eeb000 ---p 00016000 103:02 266725                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f76b5eeb000-7f76b5eec000 r--p 00015000 103:02 266725                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f76b5eec000-7f76b5eed000 rw-p 00016000 103:02 266725                    /lib/x86_64-linux-gnu/libgcc_s.so.1
7f76b5eed000-7f76b5ef8000 r-xp 00000000 103:02 262243                    /lib/x86_64-linux-gnu/libnss_files-2.24.so
7f76b5ef8000-7f76b60f7000 ---p 0000b000 103:02 262243                    /lib/x86_64-linux-gnu/libnss_files-2.24.so
7f76b60f7000-7f76b60f8000 r--p 0000a000 103:02 262243                    /lib/x86_64-linux-gnu/libnss_files-2.24.so
7f76b60f8000-7f76b60f9000 rw-p 0000b000 103:02 262243                    /lib/x86_64-linux-gnu/libnss_files-2.24.so
7f76b60f9000-7f76b60ff000 rw-p 00000000 00:00 0 
7f76b60ff000-7f76b610a000 r-xp 00000000 103:02 266638                    /lib/x86_64-linux-gnu/libnss_nis-2.24.so
7f76b610a000-7f76b6309000 ---p 0000b000 103:02 266638                    /lib/x86_64-linux-gnu/libnss_nis-2.24.so
7f76b6309000-7f76b630a000 r--p 0000a000 103:02 266638                    /lib/x86_64-linux-gnu/libnss_nis-2.24.so
7f76b630a000-7f76b630b000 rw-p 0000b000 103:02 266638                    /lib/x86_64-linux-gnu/libnss_nis-2.24.so
7f76b630b000-7f76b6321000 r-xp 00000000 103:02 262240                    /lib/x86_64-linux-gnu/libnsl-2.24.so
7f76b6321000-7f76b6520000 ---p 00016000 103:02 262240                    /lib/x86_64-linux-gnu/libnsl-2.24.so
7f76b6520000-7f76b6521000 r--p 00015000 103:02 262240                    /lib/x86_64-linux-gnu/libnsl-2.24.so
7f76b6521000-7f76b6522000 rw-p 00016000 103:02 262240                    /lib/x86_64-linux-gnu/libnsl-2.24.so
7f76b6522000-7f76b6524000 rw-p 00000000 00:00 0 
7f76b6524000-7f76b652c000 r-xp 00000000 103:02 262241                    /lib/x86_64-linux-gnu/libnss_compat-2.24.so
7f76b652c000-7f76b672b000 ---p 00008000 103:02 262241                    /lib/x86_64-linux-gnu/libnss_compat-2.24.so
7f76b672b000-7f76b672c000 r--p 00007000 103:02 262241                    /lib/x86_64-linux-gnu/libnss_compat-2.24.so
7f76b672c000-7f76b672d000 rw-p 00008000 103:02 262241                    /lib/x86_64-linux-gnu/libnss_compat-2.24.so
7f76b672d000-7f76b6b37000 r--p 00000000 103:02 3150668                   /usr/lib/locale/locale-archive
7f76b6b37000-7f76b6b4f000 r-xp 00000000 103:02 266706                    /lib/x86_64-linux-gnu/libpthread-2.24.so
7f76b6b4f000-7f76b6d4f000 ---p 00018000 103:02 266706                    /lib/x86_64-linux-gnu/libpthread-2.24.so
7f76b6d4f000-7f76b6d50000 r--p 00018000 103:02 266706                    /lib/x86_64-linux-gnu/libpthread-2.24.so
7f76b6d50000-7f76b6d51000 rw-p 00019000 103:02 266706                    /lib/x86_64-linux-gnu/libpthread-2.24.so
7f76b6d51000-7f76b6d55000 rw-p 00000000 00:00 0 
7f76b6d55000-7f76b6d58000 r-xp 00000000 103:02 262236                    /lib/x86_64-linux-gnu/libdl-2.24.so
7f76b6d58000-7f76b6f57000 ---p 00003000 103:02 262236                    /lib/x86_64-linux-gnu/libdl-2.24.so
7f76b6f57000-7f76b6f58000 r--p 00002000 103:02 262236                    /lib/x86_64-linux-gnu/libdl-2.24.so
7f76b6f58000-7f76b6f59000 rw-p 00003000 103:02 262236                    /lib/x86_64-linux-gnu/libdl-2.24.so
7f76b6f59000-7f76b6fcb000 r-xp 00000000 103:02 266813                    /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f76b6fcb000-7f76b71ca000 ---p 00072000 103:02 266813                    /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f76b71ca000-7f76b71cb000 r--p 00071000 103:02 266813                    /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f76b71cb000-7f76b71cc000 rw-p 00072000 103:02 266813                    /lib/x86_64-linux-gnu/libpcre.so.3.13.3
7f76b71cc000-7f76b7389000 r-xp 00000000 103:02 262233                    /lib/x86_64-linux-gnu/libc-2.24.so
7f76b7389000-7f76b7589000 ---p 001bd000 103:02 262233                    /lib/x86_64-linux-gnu/libc-2.24.so
7f76b7589000-7f76b758d000 r--p 001bd000 103:02 262233                    /lib/x86_64-linux-gnu/libc-2.24.so
7f76b758d000-7f76b758f000 rw-p 001c1000 103:02 262233                    /lib/x86_64-linux-gnu/libc-2.24.so
7f76b758f000-7f76b7593000 rw-p 00000000 00:00 0 
7f76b7593000-7f76b75b6000 r-xp 00000000 103:02 266840                    /lib/x86_64-linux-gnu/libselinux.so.1
7f76b75b6000-7f76b77b5000 ---p 00023000 103:02 266840                    /lib/x86_64-linux-gnu/libselinux.so.1
7f76b77b5000-7f76b77b6000 r--p 00022000 103:02 266840                    /lib/x86_64-linux-gnu/libselinux.so.1
7f76b77b6000-7f76b77b7000 rw-p 00023000 103:02 266840                    /lib/x86_64-linux-gnu/libselinux.so.1
7f76b77b7000-7f76b77b9000 rw-p 00000000 00:00 0 
7f76b77b9000-7f76b77de000 r-xp 00000000 103:02 262229                    /lib/x86_64-linux-gnu/ld-2.24.so
7f76b79bb000-7f76b79bd000 rw-p 00000000 00:00 0 
7f76b79d9000-7f76b79dd000 rw-p 00000000 00:00 0 
7f76b79dd000-7f76b79de000 r--p 00024000 103:02 262229                    /lib/x86_64-linux-gnu/ld-2.24.so
7f76b79de000-7f76b79df000 rw-p 00025000 103:02 262229                    /lib/x86_64-linux-gnu/ld-2.24.so
7f76b79df000-7f76b79e0000 rw-p 00000000 00:00 0 
7ffca65c4000-7ffca65e5000 rw-p 00000000 00:00 0                          [stack]
7ffca65f4000-7ffca65f6000 r--p 00000000 00:00 0                          [vvar]
7ffca65f6000-7ffca65f8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
E: Sub-process /usr/bin/dpkg exited unexpectedly
```

z3ntu, 2017-02-02:
You shouldn't install it on Ubuntu manually but use the PPAs provided by terrycain. See the website for installation instructions.

maciek134, 2017-02-02:
I guess that's correct for most users, but I'm trying to add a new device, so I have to install manually. To make it work I had to `sudo cp /usr/lib/python3.5/site-packages/razer* /usr/lib/python3/dist-packages` (maybe should've copied to site-packages, I'm no python expert).

artificial-immersion, 2017-02-02:
Had the same issue on debian stretch. Manually installing the python packages again fixed it for me.

z3ntu, 2017-02-02:
Again: Don't use the `ubuntu_install` target as it's supposed to only be used when creating packages.

terrycain, 2017-02-02:
> I guess that's correct for most users, but I'm trying to add a new device

Don't manually install, just set your PYTHONPATH appropriately and then just run the daemon from the checked out location. PyCharm does this for you, if you use it that is ;-)

z3ntu, 2017-02-18:
Can this be closed? @maciek134

maciek134, 2017-02-18:
Yeah, sorry, the issue was my fault.

closed 2017-02-18 by maciek134