openrazer/openrazer #183: Razerservice won't start under debian

artificial-immersion, 2017-02-02:
razer-service -v --forgeground:

`Traceback (most recent call last):
  File "/usr/bin/razer-service", line 102, in <module>
    run()
  File "/usr/bin/razer-service", line 96, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 114, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 217, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 431, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 207, in __init__
    super(RazerBlackWidowChroma, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 19, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 60, in __init__
    self.serial = self.get_serial()
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 194, in get_serial
    with open(serial_path, 'r') as serial_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:1a.0/usb5/5-1/5-1:1.2/0003:1532:0203.0005/device_serial'
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 325, in close`
    if not self._is_closed:
AttributeError: 'RazerBlackWidowChroma' object has no attribute '_is_closed'

z3ntu, 2017-02-02:
> PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:1a.0/usb5/5-1/5-1:1.2/0003:1532:0203.0005/device_serial'

Are you in the `plugdev` group?

Shildifreak, 2017-02-06:
I had the same issue. The owner of the files in /sys/bus/hid/drivers/razerkbd/... was root:root.
Changing it helped to some extend (still some files in dev that caused issues).
After reboot everything worked fine.

closed 2017-02-10 by terrycain