openrazer/openrazer #184: Razer Blackwidow Chroma V2 (released Jan 24, 2017) Support

V13Axel, 2017-02-02:
This device was released quite recently, and I wound up getting one. Decided it'd be helpful to myself and others to try and contribute to this project.

USB ID is `1532:0221`

Output of `sudo lsusb -d 1532:0221 -v`:

```Bus 003 Device 012: ID 1532:0221 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0221 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Chroma V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
          Report Descriptor: (length is 94)
            Item(Global): Usage Page, data= [ 0x01 ] 1
                            Generic Desktop Controls
            Item(Local ): Usage, data= [ 0x02 ] 2
                            Mouse
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Local ): Usage, data= [ 0x01 ] 1
                            Pointer
            Item(Main  ): Collection, data= [ 0x00 ] 0
                            Physical
            Item(Global): Usage Page, data= [ 0x09 ] 9
                            Buttons
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            Button 1 (Primary)
            Item(Local ): Usage Maximum, data= [ 0x05 ] 5
                            Button 5
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x05 ] 5
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x03 ] 3
            Item(Main  ): Input, data= [ 0x03 ] 3
                            Constant Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x40 ] 64
                            (null)
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x02 ] 2
            Item(Global): Logical Minimum, data= [ 0x81 ] 129
            Item(Global): Logical Maximum, data= [ 0x7f ] 127
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Usage Page, data= [ 0x01 ] 1
                            Generic Desktop Controls
            Item(Local ): Usage, data= [ 0x38 ] 56
                            Wheel
            Item(Global): Logical Minimum, data= [ 0x81 ] 129
            Item(Global): Logical Maximum, data= [ 0x7f ] 127
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x01 ] 1
            Item(Main  ): Input, data= [ 0x06 ] 6
                            Data Variable Relative No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x30 ] 48
                            Direction-X
            Item(Local ): Usage, data= [ 0x31 ] 49
                            Direction-Y
            Item(Global): Logical Minimum, data= [ 0x00 0x80 ] 32768
            Item(Global): Logical Maximum, data= [ 0xff 0x7f ] 32767
            Item(Global): Report Size, data= [ 0x10 ] 16
            Item(Global): Report Count, data= [ 0x02 ] 2
            Item(Main  ): Input, data= [ 0x06 ] 6
                            Data Variable Relative No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x5a ] 90
            Item(Main  ): Feature, data= [ 0x01 ] 1
                            Constant Array Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

`ls -lah /dev/input/by-id/`:

```
lrwxrwxrwx 1 root root  10 Feb  1 22:48 usb-Razer_Razer_BlackWidow_Chroma_V2-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10 Feb  1 22:48 usb-Razer_Razer_BlackWidow_Chroma_V2-if01-event-kbd -> ../event19
```

Not in a position currently where I can do the pcap(s) and take Synapse screenshots, just thought I'd go ahead and open an issue for this device to get the ball rolling.

z3ntu, 2017-02-02:
Could you tell me the product/model number of the mouse just for documentation reasons? :)

V13Axel, 2017-02-02:
@z3ntu not sure what you mean... It's a keyboard.

It's the BlackWidow Chroma V2, as mentioned in the issue title: http://www.razerzone.com/gaming-keyboards-keypads/razer-blackwidow-chroma-v2

Model RZ03-0203
Product number RZ03-02030200-R3U1

As for the synapse screenshots, I'm not currently in a position to provide them. I'll do so later this evening, around 6 or 7PM EST.

z3ntu, 2017-02-02:
Yeah sorry, in the first second I somehow read DeathAdder Chroma V2 and apparently I wrote that down. But yes, I meant exactly these numbers :)

terrycain, 2017-02-02:
@V13Axel Will also need a few packet captures, how-to [here](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol), I am assuming its functionally equivelent to the V1 but i have a strong suspicion that it uses the non-simple effects commands, hence I want captures just to make sure.

V13Axel, 2017-02-03:
@terrycain Yep, planning to do that as soon as I can. For now, here are the Chroma options:

http://imgur.com/a/zC3MO

z3ntu, 2017-02-03:
The dropdown menu ("Effect") in the second image would be nice :)

wtjerry, 2017-02-08:
I also got one of this keyboards.
lsusb gave the same result as V13Axel already posted. My product number however is a bit different: RZ03-02031200-R3Q1 (it is a swiss keyboard)

Here an print screen of the effect dropdown: https://imgur.com/a/snZl7

terrycain, 2017-02-08:
Can you get some packet captures? Pretty sure I already know what's going to be there but I like evidence

cptdaaaaave, 2017-02-14:
my Blackwidow Chroma v2 is on the way and I'm keen to help out if there is anything I can do :)

terrycain, 2017-02-15:
Packet captures :smile:

vbujanca, 2017-02-15:
@terrycain do you mean sniff packets with wireshark?

terrycain, 2017-02-15:
USB packets yes - https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

terrycain, 2017-02-15:
^^ Basically you need a windows vm with synapse installed. If anyone is capable then I'll list what I need.

vbujanca, 2017-02-15:
@terrycain I have most of the setup installed, I will be neding wireshark as well, on the guest machine I guess. I'll be doing this now

cptdaaaaave, 2017-02-15:
So you're running wireshark on your *nix host and relying on the virtual box USB pass through from the windows VM?

terrycain, 2017-02-15:
@cptdaaaaave yeah, @vbujanca you need wireshark on the linux, to _watch_ the traffic going to the windows.

cptdaaaaave, 2017-02-15:
@terrycain cool, have all that except my keyboard. So I'll get on it once it arrives next week 👍🏼

vbujanca, 2017-02-15:
I see, in this case I will be needing an extra keyboard as the vm needs to steal it in order to run synapse. I will probably just have to filter the razer keyboard. If you don't mind then I'll be doing this tomorrow morning. Cheers!

terrycain, 2017-02-15:
No probs, and as long as wireshark sees packets and synapse sees keyboard thats all that matters.

I need
* a capture from before the vm is started going all the way until synapse is open and on screen
* one capture per effect (excluding ripple, fire) and just click some options of each effcet, like for breathing, clikc single breathing, dual breathing colours and random.
* game mode on and off a few times

* fire effect, it will spam a shit load of packets, capture like a few seconds.

vbujanca, 2017-02-16:
I think I have captured all that you asked, please let me know if you need anything else. Good luck!

[capture.zip](https://github.com/terrycain/razer-drivers/files/780359/capture.zip)

terrycain, 2017-02-16:
Will have a look at the weekend

terrycain, 2017-02-17:
@vbujanca what os you on?

terrycain, 2017-02-17:
@vbujanca also what physical key layout does your keyboard have?

vbujanca, 2017-02-17:
@terrycain 

host os linux mint 17.3, guest vm (in virtualbox) is windows 10 the demo image from ms

the layout is plain english us

terrycain, 2017-02-17:
1. Disable secure boot if its on
1. Grab [these](https://drive.google.com/drive/folders/0BydYgD2LzDUCMUliMndjTzFZUmc?usp=sharing) debs, install them (`dpkg -i debname.deb`, will probably then need to also run `apt-get install -f` to fix dependencies)
1. Reboot, literally this step is crucial

Now 
```
ls -l /sys/bus/hid/drivers/
ls -l /sys/bus/hid/drivers/razerkbd
ls -l /sys/bus/hid/drivers/razerkbd/0*/
```

vbujanca, 2017-02-17:
```
ls -l /sys/bus/hid/drivers/
total 0
drwxr-xr-x 2 root root 0 Feb 17 14:56 hid-generic
drwxr-xr-x 2 root root 0 Feb 17 14:56 razerkbd

ls -l /sys/bus/hid/drivers/razerkbd
total 0
lrwxrwxrwx 1 root root    0 Feb 17 14:59 0003:1532:0221.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:1532:0221.0001
lrwxrwxrwx 1 root root    0 Feb 17 14:59 0003:1532:0221.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:1532:0221.0002
lrwxrwxrwx 1 root root    0 Feb 17 14:59 0003:1532:0221.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:1532:0221.0003
--w------- 1 root root 4096 Feb 17 14:56 bind
lrwxrwxrwx 1 root root    0 Feb 17 14:59 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Feb 17 14:59 new_id
--w------- 1 root root 4096 Feb 17 14:59 uevent

ls -l /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:0221.0001/:
total 0
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 hidraw
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 input
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_super
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 modalias
drwxr-xr-x 2 root plugdev    0 Feb 17 14:56 power
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb 17 14:56 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/:
total 0
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 hidraw
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 input
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 key_super
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 modalias
drwxr-xr-x 2 root plugdev    0 Feb 17 14:56 power
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb 17 14:56 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0221.0003/:
total 0
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 device_mode
-r--r----- 1 root plugdev 4096 Feb 17 14:56 device_serial
-r--r----- 1 root plugdev 4096 Feb 17 14:56 device_type
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Feb 17 14:56 firmware_version
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 game_led_state
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 hidraw
drwxr-xr-x 3 root plugdev    0 Feb 17 14:56 input
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 macro_led_state
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 matrix_brightness
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_none
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_static
--w--w---- 1 root plugdev 4096 Feb 17 14:56 matrix_effect_wave
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 modalias
drwxr-xr-x 2 root plugdev    0 Feb 17 14:56 power
-r--r--r-- 1 root plugdev 4096 Feb 17 14:56 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb 17 14:56 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Feb 17 14:56 test
-rw-r--r-- 1 root plugdev 4096 Feb 17 14:56 uevent
-r--r----- 1 root plugdev 4096 Feb 17 14:56 version

```

terrycain, 2017-02-17:
ok to start with, sudo to root, cd to `/sys/bus/hid/drivers/razerkbd/0003:1532:0221.0003/`
```
cat device_mode
cat device_serial
cat device_type
cat firmware_version
cat test
cat version
```

vbujanca, 2017-02-17:
```
cat device_mode
0:0

cat device_serial
IO1704F15600402

cat device_type
Razer BlackWidow Chroma V2

cat firmware_version
v1.0

cat test
018100

cat version
1.1.8
```

terrycain, 2017-02-17:
What physical key layout do you have on the keyboard? Pic would be nice, but thats looking good gimme a bit and ill post some more :)

vbujanca, 2017-02-17:
@terrycain sent you pic from phone, lemme know if you got it

terrycain, 2017-02-17:
Nope no pic

vbujanca, 2017-02-17:
[image]

terrycain, 2017-02-17:
Here is more, paste things line by line, dont paste multiple lines. I have doubts about these ones, they "should" work
```
# Dont try other things with device mode, trust me, though i think ive hardcoded safe options
echo -ne '\x03\x00' > device_mode
cat device_mode
# Should output something like 3:0, if it doesnt then :-(

# Should toggle game mode led (wont actually do game mode things)
cat game_led_state
echo '1' > game_led_state
cat game_led_state
echo '0' > game_led_state

# Macro led
echo '1' > macro_led_state
cat macro_led_state
echo '1' > macro_led_effect
cat macro_led_effect
# Wait as it should blink

# No blink
echo '0' > macro_led_effect
# macro led off
echo '0' > macro_led_state

# Try brightness
echo -n '255' > matrix_brightness
cat matrix_brightness
echo -n '128' > matrix_brightness
cat matrix_brightness
echo -n '64' > matrix_brightness
cat matrix_brightness
echo -n '255' > matrix_brightness
cat matrix_brightness
```

vbujanca, 2017-02-17:
@terrycain I can only continue testing this on Monday, sorry. Have nice weekend there!

terrycain, 2017-02-18:
@vbujanca np

vbujanca, 2017-02-20:
```
echo -ne '\x03\x00' > device_mode

cat device_mode
3:0

```

Al lthe other features seem to work, from root

terrycain, 2017-02-20:
you tested all the ones above? what about the ones not mentioned?

vbujanca, 2017-02-20:
the brightness, macro led state/effect and game led state react to echo commands

terrycain, 2017-02-20:
Awesome, take it the ones that output strings like serials work too. What f/w version is it on by the way. And i'll paste some more to do later. About 1/4 the way through all testing i think.

vbujanca, 2017-02-20:
isn't it the firmware_version I have sent you earlier (1.0) ? if not, can I find it in synapse?

terrycain, 2017-02-20:
My bad was being lazy and didnt look up, yep thats kl

terrycain, 2017-02-20:
```
# Static
echo -ne '\xFF\x00\x00' > matrix_effect_static
echo -ne '\x00\xFF\x00' > matrix_effect_static

# Specturm
echo -n '1' > matrix_effect_spectrum

# Wave
echo -n '1' > matrix_effect_wave
echo -n '2' > matrix_effect_wave

# No effect
echo -n '1' > matrix_effect_none

# Reactive
echo -ne '\x03\xFF\x00\x00' > matrix_effect_reactive

# Breathing
echo -ne '\xFF\x00\x00' > matrix_effect_breathing

```

vbujanca, 2017-02-20:
Spectrum works fine. So does wave effect (both directions) and none. Reactive mode works too.

Not sure what static effect does, but I assume it is also working as it reacts by illuminating all keys.

However, the line `echo -ne '\xFF\x00\x00' > matrix_effect_breathing` generates a 'Permission denied'

I am running all these commands as root

terrycain, 2017-02-20:
Reread the comment, added startlight to it after accidentally commenting, also corrected typo breathing should be breath.

Static should just change colour, thats it, a solid colour.

vbujanca, 2017-02-20:
ok, Breathing effect works now, after 2nd breathing command it also alternates between two or more colors?

the `echo -ne '\x02\xFF\x00\x00' > matrix_effect_starlight` end up in permission denied

terrycain, 2017-02-20:
> ok, Breathing effect works now, after 2nd breathing command it also alternates between two or more colors?
Yeah,

And crap yeah forgot a section, copy-paste mistake. Install [these](https://drive.google.com/open?id=0BydYgD2LzDUCMUliMndjTzFZUmc) debs, reboot (reboot crucial) then try starlight bit

vbujanca, 2017-02-20:
starlight effect works fine now, the second command makes it alternate colors, not sure about third starlight command as doesn't seem to change anything

terrycain, 2017-02-20:
np

terrycain, 2017-02-20:
Will add daemon support, and drum up some more tests tomorrow, but near enough there now :)

terrycain, 2017-02-20:
Try [these](https://drive.google.com/open?id=0BydYgD2LzDUCMUliMndjTzFZUmc) debs, reboot. Then go grab `d-feet` application (for troubleshooting) and grab either polychromatic or razer-commander (links in readme) and see if they work

vbujanca, 2017-02-21:
I'm afraid I can't help further, due to my linux distribution, both polychromatic and razer commander require gtk 3.14+, whereas I run linux mint 17.

I might be able to upgrade soon, will let you know when I can test again.

terrycain, 2017-02-21:
@vbujanca  can you run `d-feet`?

vbujanca, 2017-02-21:
I can. I am also upgrading mint so I can hopefully test the whole stack soon

terrycain, 2017-02-21:
Well if you open d-feet, click on the session tab, find an entry named razer.org (or something like that), on the right expand all the subtree options then send or link to a screenshot

z3ntu, 2017-02-21:
*`org.razer`

vbujanca, 2017-02-22:
[image]

z3ntu, 2017-02-22:
@vbujanca You sure that the device is connected? If `ls -al /sys/bus/hid/drivers/razerkbd/0*/` returns no error, please do a `pkill -e razer && razer-service --verbose -F` and look again into d-feet. If there's still no entry `/org/razer/device/<serialno>` please send the output from the razer-service command above.

vbujanca, 2017-02-22:
the razerkbd folder is missing! apparently the mint upgrade removed it? I will have to reinstall the earlier deb packages

z3ntu, 2017-02-22:
nah, it's missing because the keyboard isn't bound to the driver. Try replugging it...

vbujanca, 2017-02-22:
replugging it didn't work, are you sure I don't have to reinstall those packages?

[7 more replies]

commit d6aa71e7 (2017-03-04): Merge branch 'master' into stable

Closes #212
Closes #184
Closes #169
Closes #149
Closes #116
closed 2017-03-04 by terrycain