openrazer/openrazer #185: Razer DeathStalker Expert Support

TNTftw21, 2017-02-03:
I just got a DeathStalker Expert from a friend, and I already really like it. Problem is, with Linux being my OS of choice, I don't have any way to set it up. It works fine as-is, but I'd really like to get macros working with it. I tried following the "Setting up the keyboard driver" tutorial on the wiki, but I keep getting back `bash: echo: write error: No such device`. Near as I can tell, it thinks that the keyboard isn't HID (which is odd, considering I can (un)bind it from `hid-generic`). I know the keyboard mostly works, as I'm typing this very issue with it.

For me, it's not terrible if you don't get around to adding support for awhile. I just thought I'd bring it to your attention.

Side note: I read that apparently the daemon supports on-the-fly macros, but I'm not entirely sure how to use it, and I'm reasonably certain it will need the drivers working anyway, so yeah.

z3ntu, 2017-02-03:
The device isn't supported yet so this is normal behavior you are experiencing.

z3ntu, 2017-02-03:
Please post the output of
`lsusb | grep -i razer`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

Note to @terrycain: the long lsusb command is coming later because of unknown PID ;)

TNTftw21, 2017-02-04:
Wow, that was fast. Glad to see you guys mean business.
Output of `lsusb`:
`Bus 003 Device 006: ID 1532:0202 Razer USA, Ltd`

Output of `ls`:
`drwxr-xr-x. 2 root root 200 Feb  3 14:17 .
drwxr-xr-x. 4 root root 540 Feb  3 17:35 ..
lrwxrwxrwx. 1 root root   9 Feb  2 16:39 usb-413c_8157-event-kbd -> ../event9
lrwxrwxrwx. 1 root root   9 Feb  2 17:40 usb-Logitech_G100s_Optical_Gaming_Mouse-event-mouse -> ../event5
lrwxrwxrwx. 1 root root   9 Feb  2 17:40 usb-Logitech_G100s_Optical_Gaming_Mouse-mouse -> ../mouse0
lrwxrwxrwx. 1 root root   9 Feb  2 16:39 usb-Logitech_Logitech_G430_Gaming_Headset-event-if03 -> ../event4
lrwxrwxrwx. 1 root root   9 Feb  3 14:17 usb-Razer_Razer_DeathStalker-event-kbd -> ../event7
lrwxrwxrwx. 1 root root   9 Feb  3 14:17 usb-Razer_Razer_DeathStalker-if01-event-kbd -> ../event8
lrwxrwxrwx. 1 root root  10 Feb  3 14:17 usb-Razer_Razer_DeathStalker-if02-event-mouse -> ../event10
lrwxrwxrwx. 1 root root   9 Feb  3 14:17 usb-Razer_Razer_DeathStalker-if02-mouse -> ../mouse2`

All of the images have been uploaded via imgur and can be found [here](http://imgur.com/a/umGEK).

I also went through and used WireShark to capture packets from and to the keyboard for brightness and Gaming Mode, as these were the only features that required packets to be sent.

z3ntu, 2017-02-04:
Okay now `lsusb -d 1532:0202 -v` please. 

Did you upload the wireshark capture or did I miss them?

TNTftw21, 2017-02-04:
Output of `lsusb`:
```
Bus 003 Device 007: ID 1532:0202 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0202 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

I apparently forgot to upload the WireShark captures, so here they are. "Gaming Mode To" is activating the gaming mode from Synapse, "Gaming Mode From" is activating the gaming mode from the keyboard, and "Lighting Mode 1" is switching between Pulsating and Full Brightness. I can't seem to find the capture for adjusting the brightness level, and I'm not able to check it right now, so that'll be uploaded later.
[WireShark Captures.zip](https://github.com/terrycain/razer-drivers/files/752627/WireShark.Captures.zip)

terrycain, 2017-02-05:
what distro you on

terrycain, 2017-02-05:
@z3ntu Fedora instructions for branch `feature_deathstalker_expert_0x0202`

Once thats installed, reboot and run
```
ls -l /sys/bus/hid/drivers/
ls -l /sys/bus/hid/drivers/razerkbd
ls -l /sys/bus/hid/drivers/razerkbd/0*/
```

z3ntu, 2017-02-06:
Run the following commands as your normal user (copy-paste the first command alone, afterwards the rest can be copy-pasted in one block).
```
sudo dnf install rpm-build rpmdevtools
rpmdev-setuptree
cd ~/rpmbuild/SPECS
curl -o razer-drivers.spec https://gist.githubusercontent.com/z3ntu/724407a74e28a8b3c9079b01dc40c401/raw/eb2ef7cab5932ae21fec1b7f958ce82d1fb76d71/razer-drivers.spec.deathstalkerexpert
spectool -g -R razer-drivers.spec
sudo dnf -y builddep razer-drivers.spec
rpmbuild -bb razer-drivers.spec
sudo dnf install ../RPMS/noarch/*.rpm
```
Don't forget to add your user to the group `plugdev` afterwards! Theoretically re-plugging your device(s) should load the driver, if that doesn't happen, please reboot.

TNTftw21, 2017-02-07:
It took a little bit of work, but now the driver is working. The keyboard attaches to it immediately, and I can enable Macros now. I still can't seem to affect the lighting mode of it through razerCommand, but that doesn't bother me too much. Thanks for your help!

z3ntu, 2017-02-21:
@memoryleakno1 Feel free to test everything for the DeathStalker Expert.

memoryleakno1, 2017-02-21:
Seems like everythign but logo led works. I have no idea how to record macros though, I don't use them even in windows to be honest.

z3ntu, 2017-02-22:
How to record a macro:
Press Fn+F9 (or wherever the macro key is), then the M-key you want to record the macro to, then enter some stuff on the keyboard and press Fn+F9 (macro key) again.
Then you can play the macro by just pressing the M-key.

z3ntu, 2017-02-22:
> Seems like everything but led on logo works.

Synapse only has 'Pulsate' and 'Brightness' options for the DeathStalker Expert, so what do you mean?

austin6598, 2017-06-05:
I have the same issue on Kubuntu 17.04 when following the wiki. I am not sure if I followed it correctly but this is where I get stuck:
`root@Kubuntu-Desktop:/sys/bus/hid/drivers/hid-generic#  echo -n "0003:1532:0202.0003" > unbind`
`root@Kubuntu-Desktop:/sys/bus/hid/drivers/hid-generic# cd /sys/bus/hid/drivers/razerkbd`
`root@Kubuntu-Desktop:/sys/bus/hid/drivers/razerkbd# echo -n "0003:1532:0202.0003" > bind`
`bash: echo: write error: No such device`

terrycain, 2017-06-06:
you running the correct driver version

austin6598, 2017-06-06:
I installed the driver for Ubuntu as directed on https://terrycain.github.io/razer-drivers/
```  
sudo add-apt-repository ppa:terrz/razerutils
sudo apt update
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
```

z3ntu, 2017-06-22:
[feature_deathstalker_expert.zip](https://github.com/terrycain/razer-drivers/files/1095773/feature_deathstalker_expert.zip)
Please install these debs and reboot.

austin6598, 2017-06-22:
The daemon connects to the keyboard now but I can't adjust brightness or create device profiles in the polychromatic controller.

terrycain, 2017-06-22:
```
ls -l /sys/bus/hid/drivers/
ls -l /sys/bus/hid/drivers/razerkbd
ls -l /sys/bus/hid/drivers/razerkbd/0*/
```

austin6598, 2017-06-22:
https://pastebin.com/RKsDFbsu

terrycain, 2017-06-22:
```
cd /sys/bus/hid/drivers/razerkbd/0003:1532:0202.0003/
cat matrix_brightness
```

austin6598, 2017-06-22:
austin@Kubuntu-Desktop:~/Desktop/Linux$ cd /sys/bus/hid/drivers/razerkbd/0003:1532:0202.0003/
austin@Kubuntu-Desktop:/sys/bus/hid/drivers/razerkbd/0003:1532:0202.0003$ cat matrix_brightness
0
austin@Kubuntu-Desktop:/sys/bus/hid/drivers/razerkbd/0003:1532:0202.0003$

terrycain, 2017-06-22:
```
echo -n '255' > matrix_brightness
cat matrix_brightness
```
does that change brightness?

austin6598, 2017-06-22:
No, also it is at max brightness right now.

terrycain, 2017-06-22:
Anything in dmesg/journalctl relating to razer?

austin6598, 2017-06-22:
Sorry, I'm not sure what you're referring to

terrycain, 2017-06-22:
look through `sudo dmesg` or `sudo journalctl`
for any errors mentioning razer

austin6598, 2017-06-22:
Nothing in journalctl but dmesg has a lot of entries with razer.
I cut out everything before and after the razer entries as the file was quite large: https://pastebin.com/02zFLYK6

terrycain, 2017-06-22:
Ok remove installed debs. install these debs https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU then reboot, then retry the brightness

austin6598, 2017-06-22:
Awesome, brightness works now. Is there any way to assign different brightness levels to keyboard shortcuts?

terrycain, 2017-06-22:
Not yet, thats in progress

austin6598, 2017-06-22:
ok, thanks for the help :)

codeman6100, 2017-06-26:
I can confirm that it is indeed working.

z3ntu, 2017-06-26:
It'll be included in the next release :)

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain