openrazer/openrazer #186: Razer Blade Stealth (Original) - toggling logo_led_state seems to have no effect

thesurfaceofthevoid, 2017-02-03:
Steps to reproduce:
(starting with lid logo off)
- navigate to /sys/bus/hid/devices/[device id]
type command: `echo -n "1" > logo_led_state` into Terminal

Expected behavior
- Logo light comes on

Actual behavior
- Logo light remains off.
- Other commands worked and had immediate effect, such as `echo -n "1" > fn_toggle`

terrycain, 2017-02-03:
Can you get me some packet captures of razer synapse doing the same - https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

thesurfaceofthevoid, 2017-02-08:
Will do.  Was unable to connect to the device from within VirtualBox.  Will reinstall Windows again on the metal and attempt packet capture.

z3ntu, 2017-07-18:
@thesurfaceofthevoid Any update?

z3ntu, 2017-12-04:
Closing as there was no response.

closed 2017-12-04 by z3ntu