openrazer/openrazer #187: Razer Naga Chroma Support

ShaolinWarrior, 2017-02-03:
To whom it may concern,

I am desperate to use my Razer Naga Chroma with my 64bit Ubuntu 16.04 LTS distro. I am fairly new to linux, but I believe I have followed all the necessary steps to install 'Razer Drivers for Linux' and 'Polychromatic'. I know that the Razer Naga Chroma is not listed in the supported devices list, hence why I am raising the following issue. Here are the steps I have taken and the commands I have run:

```
sudo add-apt-repository ppa:terrz/razerutils
sudo apt update
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
reboot
```

No errors at this point, so I then installed 'Polychromatic...

```
sudo add-apt-repository ppa:lah7/polychromatic
sudo apt update
sudo apt install polychromatic
reboot
```

Still no errors, so I searched my apps and ran 'Polychromatic'. It loaded just fine, however it stated the following:

> No Devices Found
> 
> However, Polychromatic detected there is a Razer product connected to your computer. Try reloading the daemon, otherwise check whether your device is supported by the daemon.

I tried reloading the daemon, but same message upon restart of 'Polychromatic'.

I read a little further down your main page and found the following command:

`lsusb | grep '1532:'`

After running the above I got the following:

> Bus 002 Device 007: ID 1532:0053 Razer USA, Ltd 

Which doesn't match any of the devices on the list (of course) xD

I then ran the following commands which I was directed to do under the 'no device found' issue page:

```
pkill razer-service
# Wait 5 seconds
sudo udevadm trigger
# Wait 5 seconds
razer-service --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
```

Which again didn't seem to do anything.

I then opened up the 'Issues' page, and scrolled to this link: [https://github.com/terrycain/razer-drivers/issues/139](url)

I saw your comments and replies, and tried to follow as best as I could:

Here is the output of

`lsusb -d 1532:0053 -v`

[naga chroma lsusb.txt](https://github.com/terrycain/razer-drivers/files/749646/naga.chroma.lsusb.txt)

Here is the output of

`ls -lah /dev/input/by-id/`

[naga chroma ls -lah.txt](https://github.com/terrycain/razer-drivers/files/749650/naga.chroma.ls.-lah.txt)

Mouse Model Number: RZ01-0161
Mouse Product Number: RZ01-01610100

I will install wireshark and try to get a windows vm set up in the next few days. I have not used wireshark nor vm's all too extensively, but will try to sort it out to get you the info you need. I have synapse installed on a windows machine already, and will try to get screenshots of all the available options and features for you as well.

If you need any further information, please just let me know.

z3ntu, 2017-02-03:
Yes, the Razer Naga Chroma isn't supported yet :)

terrycain, 2017-02-03:
Should of looked in the issue list #59

closed 2017-02-03 by terrycain

ShaolinWarrior, 2017-02-03:
@terrycain my bad, did not see that there was a second page to the issue list xD I will be waiting patiently then lol