openrazer/openrazer #190: Razer BlackWidow Chroma 1532:0203	ripple effect lag

TrueNix, 2017-02-06:
Hi, is there any way to make ripple effect faster, it feels slow compare to windows.

Thanks.

terrycain, 2017-02-06:
Seems fine on my machine. What app are you using to set ripple?

TrueNix, 2017-02-06:
Thanks for replying, 
I'm using Polychromatic. 
It is not lag so much but it's much slower then on windows, maybe I should request speed tweaking for ripple effect?

terrycain, 2017-02-06:
Polychromatic could set the refresh rate to lower value than is default. Other than that, there's not much else can be done.

closed 2017-02-06 by terrycain

lah7, 2017-02-06:
I'm the author of Polychromatic - I'll set the refresh rate to `0.01` (the default is `0.05`) for the next release - that's assuming devices other than the BlackWidow Chroma will be OK at this value.

TrueNix, 2017-02-06:
Thank you so much!

terrycain, 2017-02-06:
Should be fine

tildearrow, 2017-02-06:
I tried to use `0.01` as refresh rate (via D-Bus) and it still is very slow.
Maybe a parameter for speed would be good?

TrueNix, 2017-02-06:
yeah a variable parameter would be ideal.

terrycain, 2017-02-06:
would require some tweaking, am too busy to do it now but next time im around that bit of code ill look into it