openrazer/openrazer #191: Why the command won't work

TramSevenTeen, 2017-02-07:
Hi!

I installed the razer-drivers succesfully. But when i type the command "echo -n "1" > mode_none" to close the lighting, nothing happened. Same for other commands. What is the problem? Thank you!

Best,

z3ntu, 2017-02-07:
What device, what distro, how installed?

TramSevenTeen, 2017-02-07:
@z3ntu Thank you for your quick reply! The device is razer blackwidow ultimate 2016. My system is ubuntu 14.04. I installed razer driver by running following command:
sudo add-apt-repository ppa:terrz/razerutils
 sudo apt update
 sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo modprobe razerkbd
From the output i believe those package are successfully installed, but seems the control command like "echo -n "1" > mode_none" won't work. Is there any further step that i need do? Thank you!

z3ntu, 2017-02-07:
Please send the output of `ls -al /sys/bus/hid/drivers/{razerkbd,razerkbd/0*}/`

TramSevenTeen, 2017-02-07:
@z3ntu The output is:
/sys/bus/hid/drivers/razerkbd/:
total 0
drwxr-xr-x 2 root root    0 Feb  6 16:58 .
drwxr-xr-x 4 root root    0 Feb  6 16:58 ..
lrwxrwxrwx 1 root root    0 Feb  6 23:01 0003:1532:0214.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0214.0005
lrwxrwxrwx 1 root root    0 Feb  6 23:01 0003:1532:0214.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/0003:1532:0214.0006
lrwxrwxrwx 1 root root    0 Feb  6 23:01 0003:1532:0214.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:1532:0214.0007
--w------- 1 root root 4096 Feb  6 23:01 bind
lrwxrwxrwx 1 root root    0 Feb  7 08:42 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Feb  7 08:42 new_id
--w------- 1 root root 4096 Feb  6 16:58 uevent
--w------- 1 root root 4096 Feb  7 08:42 unbind

/sys/bus/hid/drivers/razerkbd/0003:1532:0214.0005/:
total 0
drwxr-xr-x 4 root plugdev    0 Feb  6 23:01 .
drwxr-xr-x 6 root root       0 Feb  6 23:01 ..
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Feb  6 23:01 hidraw
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_super
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 modalias
drwxr-xr-x 2 root plugdev    0 Feb  6 23:01 power
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb  6 23:01 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0214.0006/:
total 0
drwxr-xr-x 4 root plugdev    0 Feb  6 23:01 .
drwxr-xr-x 6 root root       0 Feb  6 23:01 ..
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Feb  6 23:01 hidraw
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 key_super
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 modalias
drwxr-xr-x 2 root plugdev    0 Feb  6 23:01 power
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb  6 23:01 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:0214.0007/:
total 0
drwxr-xr-x 4 root plugdev    0 Feb  6 23:01 .
drwxr-xr-x 6 root root       0 Feb  6 23:01 ..
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 device_mode
-r--r----- 1 root plugdev 4096 Feb  6 23:01 device_serial
-r--r----- 1 root plugdev 4096 Feb  6 23:01 device_type
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Feb  6 23:01 firmware_version
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 game_led_state
drwxr-xr-x 3 root plugdev    0 Feb  6 23:01 hidraw
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 macro_led_state
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 matrix_brightness
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_none
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_starlight
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_static
--w--w---- 1 root plugdev 4096 Feb  6 23:01 matrix_effect_wave
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 modalias
drwxr-xr-x 2 root plugdev    0 Feb  6 23:01 power
-r--r--r-- 1 root plugdev 4096 Feb  6 23:01 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb  6 23:01 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Feb  6 23:01 test
-rw-r--r-- 1 root plugdev 4096 Feb  6 23:01 uevent
-r--r----- 1 root plugdev 4096 Feb  6 23:01 version

z3ntu, 2017-02-07:
As you can see there's no file called `mode_none` so it won't work. You should use `matrix_effect_none` instead.

terrycain, 2017-02-07:
The driver shoudnt be used directly anyway and is subject to change at any time without notice, hence the daemon exists.

closed 2017-02-07 by terrycain

TramSevenTeen, 2017-02-08:
@z3ntu Thank you for suggestion
Just went home to try the command. But it still won't change after i use "echo -n "1" > matrix_effect_starlight" or "echo -n "1" > matrix_effect_none"...

phspade, 2017-09-08:
me too.. i'm having trouble setting with `matrix_effect_reactive` and the others.