openrazer/openrazer #192: Can't start the razer service - NameError: name 'open' is not defined

shtuffit, 2017-02-07:
I installed the razer drivers as outlined here on a RAZER BLADE - QHD:

https://terrycain.github.io/razer-drivers/#ubuntu

When I start eh service in the foreground it exits with the following error:

> $ sudo razer-service -F
> 2017-02-07 12:34:18 | razer                          | INFO     | Initialising Daemon (v1.1.8). Pid: 4392
> 2017-02-07 12:34:18 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
> 2017-02-07 12:34:18 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
> 2017-02-07 12:34:18 | razer                          | INFO     | Found device.0: 0003:1532:020F.0003
> 2017-02-07 12:34:18 | razer.device0                  | INFO     | Initialising device.0 RazerBladeQHD
> Traceback (most recent call last):
>   File "/usr/bin/razer-service", line 71, in <module>
>     run()
>   File "/usr/bin/razer-service", line 65, in run
>     daemonize(**daemon_args)
>   File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 114, in daemonize
>     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
>   File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 217, in __init__
>     self._load_devices(first_run=True)
>   File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 430, in _load_devices
>     razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 475, in __init__
>     super(RazerBladeQHD, self).__init__(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 19, in __init__
>     super(_MacroKeyboard, self).__init__(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 80, in __init__
>     DBusService.__init__(self, self.BUS_PATH, object_path)
>   File "/usr/lib/python3/dist-packages/razer_daemon/dbus_services/service.py", line 85, in __init__
>     super(DBusService, self).__init__(bus_object, object_path)
>   File "/usr/lib/python3/dist-packages/dbus/service.py", line 455, in __init__
>     validate_object_path(object_path)
> ValueError: Invalid object path '/org/razer/device/Default string': contains invalid character ' '
> Exception ignored in: <bound method RazerDevice.__del__ of RazerBladeQHD:Default string>
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
[... 2 lines cut ...]
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 32, in _close
>   File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 244, in set_device_mode
> NameError: name 'open' is not defined

the following razer packages are installed:

> $ dpkg -l | grep razer
> ii  python3-razer                               1.1.8-0ubuntu1                                    all          Razer Python Library
> ii  razer-daemon                                1.1.8-0ubuntu1                                    amd64        Razer Device Daemon
> ii  razer-doc                                   1.1.8-0ubuntu1                                    all          Documentation and examples for the Razer Tools
> ii  razer-kernel-modules-dkms                   1.1.8-0ubuntu1                                    amd64        Razer Peripheral Drivers
> 

module is loaded:

> $ lsmod|grep razer
> razerkbd               49152  0
> hid                   118784  6 i2c_hid,hid_multitouch,hid_generic,razerkbd,hid_rmi,usbhid
> 

and the device is recognized:

> $ lsusb | grep '1532:'
> Bus 001 Device 004: ID 1532:020f Razer USA, Ltd 
>

z3ntu, 2017-02-07:
I believe

> ValueError: Invalid object path '/org/razer/device/Default string': contains invalid character ' '

is the issue... @terrycain will probably know why ;)

terrycain, 2017-02-07:
Restart, does it do it again?

shtuffit, 2017-02-07:
This post was made after restarting, results are the same

z3ntu, 2017-02-07:
Please post the output of `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial`

shtuffit, 2017-02-07:
> $ cat /sys/bus/hid/drivers/razerkbd/0*/device_serial
> Default string
>

terrycain, 2017-02-08:
HA, it seems the getting of the serial is returning something crap. Dont suppose you can mess around dmidecode and see if you can find your laptops serial number - http://www.tecmint.com/how-to-get-hardware-information-with-dmidecode-command-on-linux/

shtuffit, 2017-02-08:
dmidecode gives "Default string" as well

> $ sudo dmidecode -t 1
> # dmidecode 3.0
> Getting SMBIOS data from sysfs.
> SMBIOS 3.0.0 present.
> 
> Handle 0x0001, DMI type 1, 27 bytes
> System Information
> 	Manufacturer: Razer
> 	Product Name: Blade
> 	Version: 4.04
> 	Serial Number: Default string
> 	UUID: 20AE3A44-23F4-DC11-B3BC-001EEC32A2AB
> 	Wake-up Type: Power Switch
> 	SKU Number: Default string
> 	Family: Default string
>

z3ntu, 2017-02-08:
lol
I now believe Razer forgot to change the serial string...

shtuffit, 2017-02-08:
Whelp, thanks for the help, guess it may be RMA time

terrycain, 2017-02-08:
Not really a valid rmi

terrycain, 2017-02-08:
Don't suppose you have a Windows partition on there?

Worst case there's a hacky fix I can do ;-)

z3ntu, 2017-02-23:
@shtuffit So did you return the laptop or is there a need for a workaround?

SammDu, 2017-04-05:
Exact same problem here w/ Razer Blade 14" August 2016 :-(
Yes I do have a Windows partition (actually multiple, counting in recovery etc.)
Please fix this problem bro, thanks a lot!

z3ntu, 2017-04-05:
So the demidecode command is also returning 'Default String' for you?

terrycain, 2017-04-05:
Hmm. I guess it's another exception to the rule, will work on this soonish

Alx31, 2017-04-06:
@SammDu Gtfo fcking dirty bot

z3ntu, 2017-04-06:
@Alx31 Please be nice here...

SammDu, 2017-04-06:
@Alx31 dirty bot eh? :-)
Dirty bots have 4 periodically maintained projects on their github page eh? :-)
Dirty bots contribute to an opensource project by reporting a problem eh? :-)
I don't wanna start a beef or anything, but try and be more responsible for your words.

SammDu, 2017-04-06:
@z3ntu Yup same "Default string" value for serial number.
@terrycain Also I've sent my blade for repair once...and it ended up needing to get the motherboard replaced.
I guess the back-up boards doesn't have their firmware configured correctly...cuz I remember the daemon working before repair...though sadly blade was not supported then :-(

Alx31, 2017-04-07:
hey im sorry, i never mentionned anyone, and i was only speaking about github bot in telegram chat, i don't even know who you guys are, so no reason for me to speak that way, i can even screenshot the chat if you want; im really really sorry about that, we will throw that github bot away asap from chat because it apparently happened few times; that very first reason why i said that in our chat.

terrycain, 2017-04-08:
Np np lool

terrycain, 2017-04-11:
Will get this patched in the driver for next release, Will just create a random serial and if default string is seen then use that instead

terrycain, 2017-06-15:
Possibly fixed in 8b8999a, have a look after the next release (v1.1.14)

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain

BilyakA, 2018-02-02:
I have Razer Blade Stealth (early 2016 I guess) and very funny thing up here:

```
~ $ cat /sys/bus/hid/drivers/razerkbd/0*/device_serial
As printed in the D cover
```

also

```
~ $ sudo dmidecode -t 1
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Razer
        Product Name: Blade Stealth
        Version: 1.04
        Serial Number: As printed in the D cover
        UUID: XXXXXXXX-23F4-DC11-B3BC-XXXXXXXXXXXX
        Wake-up Type: Power Switch
        SKU Number:  
        Family:
```

note SKU and Family are empty...

Any solution to this one?

z3ntu, 2018-02-02:
@BilyakA So...? What's the issue?

BilyakA, 2018-02-02:
@z3ntu the issue is the same as OP's one:
```
....
2018-02-02 17:35:13 | razer                          | INFO     | Found device.1: 0003:1532:0205.0006
2018-02-02 17:35:13 | razer.device1                  | INFO     | Initialising device.1 RazerBladeStealth
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 126, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/daemon.py", line 107, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/daemon.py", line 398, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/keyboards.py", line 50, in __init__
    super(_RippleKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/keyboards.py", line 19, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 441, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 88, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/dbus_services/service.py", line 87, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib64/python3.5/site-packages/dbus/service.py", line 455, in __init__
    validate_object_path(object_path)
ValueError: Invalid object path '/org/razer/device/As printed in the D cover': contains invalid character ' '

Stopping daemon.
Exception ignored in: <bound method RazerDevice.__del__ of RazerBladeStealth:As printed in the D cover>
Traceback (most recent call last):
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 426, in __del__
    self.close()
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/device_base.py", line 340, in close
    self._close()
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/keyboards.py", line 55, in _close
    super(_RippleKeyboard, self)._close()
  File "/usr/lib64/python3.5/site-packages/openrazer_daemon/hardware/keyboards.py", line 39, in _close
    self.key_manager.close()
AttributeError: 'RazerBladeStealth' object has no attribute 'key_manager'

```

In the [8b8999a](https://github.com/openrazer/openrazer/commit/8b8999adbb65f27fdb32ce757b8eece05d0dc70d) commit you are handling only "Default string" as serial number. Obviously mine serial number sting does not handeld that cause driver crash

z3ntu, 2018-02-02:
Oh you mean it's literally `As printed in the D cover` ? Sorry for misunderstanding you.

BilyakA, 2018-02-02:
@z3ntu Yeah, exactly. That made me laugh a lot when I saw it

z3ntu, 2018-02-02:
Can you apply the following patch on the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py` or `/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py` (depending on your distro)

```diff
diff --git a/daemon/openrazer_daemon/hardware/device_base.py b/daemon/openrazer_daemon/hardware/device_base.py
index ab63edb..711daac 100644
--- a/daemon/openrazer_daemon/hardware/device_base.py
+++ b/daemon/openrazer_daemon/hardware/device_base.py
@@ -215,10 +215,10 @@ class RazerDevice(DBusService):
                 if len(serial) == 0:
                     self.logger.debug('getting serial: {0} count:{1}'.format(serial, count))
 
-            if serial == '' or serial == 'Default string':
+            if serial == '' or serial == 'Default string' or serial == 'As printed in the D cover':
                 serial = 'UNKWN{0:012}'.format(random.randint(0, 4096))
 
-            self._serial = serial
+            self._serial = serial.replace(' ', '_')
 
         return self._serial
```

BilyakA, 2018-02-02:
@z3ntu 
Yeah, thought about it... But just wanted to ask if there is some more elegant solution. Even to reflash this value in bios if it is possible.

z3ntu, 2018-02-14:
I pushed the patch to master btw.

BilyakA, 2018-02-14:
@z3ntu wow, that is very unexpected. I am using Gentoo so generally for me it is not a problem to write my own patch and apply it every time I build package.
But thanks! Who knows, maybe I am not alone with this :)

BilyakA, 2018-05-28:
@shtuffit Hey, I've found solution for our problem
You will need windows to do all below.

Download AMI Firmware Update from [here](https://ami.com/en/products/bios-uefi-tools-and-utilities/bios-uefi-utilities/)
Download AMIBCP from [here](http://www.mediafire.com/download/irxk9p1m43kfk68/AMIBCP_v5.0.1.7Z)
Start AFUWINGUI from "AMI AFU for Aptio V\afu\afuwin\64\AfuWin64"
Click "Save" right away, I will create dump of your current BIOS
Open your dump with AMIBCP. Go to "DMI TAbles" tab and find "1 system Information (Type 1)" from the list on the left. When you select it - you will see Text Strings on bottom right. String number 4 should be serial number. Just double click on it and change it as you like.
Save your new dump (better in a new file). Open AFUWINGUI, Click Open and select your new dump. Flash.
Now you should be fine with valid serial number inside your BIOS.

P.S. I take no responsibiliy for anything could happen :)

Parts of credits goes to kamueone from notebookreview.com forum ([archived topic](https://web.archive.org/web/20180125184527/http://forum.notebookreview.com/threads/razer-blade-2016-bios-mod.792601/)