openrazer/openrazer #193: can't start razer service on Razer a Blade 1060 QHD

federicoweber, 2017-02-08:
Hi there,
thank you for doing this amazing work 😄.

I'm getting the following error starting the service on my RazerBladeQHD 1060, running Arch.
I've installed it via `razer-driver-meta`.

Thanks!

```
razer-service -F
2017-02-09 00:40:41 | razer                          | INFO     | Initialising Daemon (v1.1.8). Pid: 4347
2017-02-09 00:40:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-02-09 00:40:41 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-02-09 00:40:41 | razer                          | INFO     | Found device.0: 0003:1532:020F.0004
2017-02-09 00:40:41 | razer.device0                  | INFO     | Initialising device.0 RazerBladeQHD
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 71, in <module>
    run()
  File "/usr/bin/razer-service", line 65, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 114, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 217, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 430, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/keyboards.py", line 475, in __init__
    super(RazerBladeQHD, self).__init__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/keyboards.py", line 19, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 60, in __init__
    self.serial = self.get_serial()
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 194, in get_serial
    with open(serial_path, 'r') as serial_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0004/device_serial'
Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 412, in __del__
    self.close()
  File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 325, in close
    if not self._is_closed:
AttributeError: 'RazerBladeQHD' object has no attribute '_is_closed'
```

terrycain, 2017-02-09:
Are you member of the plugdev group?

federicoweber, 2017-02-09:
Oh you are right, I missed that part. Added it in and it's working smoothly now. 
Thank you!

closed 2017-02-09 by federicoweber

terrycain, 2017-02-09:
Np