openrazer/openrazer #195: Razer DeathAdder 2013 support

rodjun, 2017-02-10:
Hey, i noticed that my ~~good~~ ol' DeathAdder 2013 is not supported, let's change that!

Some preliminary info:

MODEL NO.      : RZ01-0084, Gaming Mouse
PRODUCT NO. : RZ01-00840100-R3U1

Output of "lsusb -d 1532:0015 -v": https://gist.github.com/rodjun/0a512d4682c61dfb026e0f0e485f6de1

Output of "ls -lah /dev/input/by-id/":
`total 0
drwxr-xr-x 2 root root 180 Fev 10 12:09 .
drwxr-xr-x 4 root root 480 Fev 10 12:09 ..
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-CM_Storm_Quickfire_TKL_6keys-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-CM_Storm_Quickfire_TKL_6keys-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-Logitech_Logitech_G930_Headset-event-if03 -> ../event12
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-Razer_Razer_DeathAdder_2013-event-mouse -> ../event15
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-Razer_Razer_DeathAdder_2013-if01-event-kbd -> ../event16
lrwxrwxrwx 1 root root  10 Fev 10 12:09 usb-Razer_Razer_DeathAdder_2013-if02-event-kbd -> ../event17
lrwxrwxrwx 1 root root   9 Fev 10 12:09 usb-Razer_Razer_DeathAdder_2013-mouse -> ../mouse0
`
Currently installing VirtualBox, i'll take some pictures of Razer Synapse when i'm done setting up the VM!

z3ntu, 2017-02-10:
You copy-pasted the `0015` from the Naga Classic ;) `0037` is correct (the output you actually posted is correct) 👍

rodjun, 2017-02-10:
Yeah i was using other issues as a start point for mine, totally missed that hahaha, it's fixed now.

After some google searches i found out that razercfg supports the DA 2013, can any info be infered from their [code](https://github.com/mbuesch/razer/blob/master/librazer/hw_deathadder2013.c)?

z3ntu, 2017-02-10:
No idea, ask @terrycain ;) But apparently Terry has all (at least older) devices in his heart so he can add support very easily 😆

RedactedProfile, 2017-06-02:
I was shocked actually to find that the 2013 model isn't supported. Its a pretty basic mouse, can the elite driver not be copied and basically stripped of the special features the Elite offers uniquely?

terrycain, 2017-06-25:
Just to confirm is this a 0037 or a 0015

QuakeDasKueken, 2017-07-20:
0037

z3ntu, 2017-07-21:
@QuakeDasKueken @RedactedProfile Could one of you please make screenshots from the available options on the mouse in Synapse (on Windows)?

rodjun, 2017-07-22:
@z3ntu 
http://i.imgur.com/QX6ZIwB.png
http://i.imgur.com/Dwtyow1.png
http://i.imgur.com/9bgd8j3.png
http://i.imgur.com/NjeYunz.png
http://i.imgur.com/LoPisMx.png

binarykitchen, 2017-07-27:
Yeah me too, I wish Razer DeathAdder 2013 support for Ubuntu 17.04 too - if there is anything I can help with, lemme know.

terrycain, 2017-08-03:
Ok ideally packet captures of 
* sliding dpi from min to max
* changing poll rate
* toggling scroll and logo leds

I could probably grab the needed info from razercfg but thats more hassle from my point of view as the packet captures near enough directly translate to what is in the driver

najashark, 2017-08-21:
hi, here is my capture for button click, polling rate and lighting
this is list how i do the capture for easy guide

button
left click
right click
forward button
back button
scroll up
scroll down

polling rate
125
500
1000

lighting
scroll ON/OFF
logo ON/OFF

[deathadder2013.zip](https://github.com/terrycain/razer-drivers/files/1239355/deathadder2013.zip)

najashark, 2017-08-21:
just to add, how to capture DPI and Acceleration?
because if I just slide the setting in synapse, nothing is captured in wireshark

terrycain, 2017-08-21:
accleration is a windows thing, dpi should work, if it doesnt then your filters probably wrong

najashark, 2017-08-21:
this is new capture for button(I missed the scroll click) and profile button below the mouse.
below is the image i use to do the click in ascending order. Later I will provide DPI capture

[image]

[deathadder.zip](https://github.com/terrycain/razer-drivers/files/1239497/deathadder.zip)

najashark, 2017-08-21:
still never get the DPI captured from synapse but i captured DPI change from another driver from [here](https://github.com/mbuesch/razer)
the step are 450, 900, 1800, 3500

[deathadder-DPIsteps.pcapng.zip](https://github.com/terrycain/razer-drivers/files/1239902/deathadder-DPIsteps.pcapng.zip)

ZakharH, 2017-09-19:
Any idea on when support will be available for the DeathAdder 2013?

Currently using `da2013ctl` but id love to use this software instead

terrycain, 2017-09-19:
Am without internet for a while as just moved house, remind me again in like 2 weeks :wink:

najashark, 2017-09-28:
its not 2 weeks, but just a reminder 😉

terrycain, 2017-09-28:
Snooze for 1 week. I have internet now, but am now catching up with all the backlog :wink:

ZakharH, 2017-10-02:
Any progress yet?

ZakharH, 2017-10-10:
its a week and a half, but just wondering if there's something in the works ;)

ZakharH, 2017-10-23:
Checking in, just wondering if there is anything in the works

binarykitchen, 2017-11-08:
yo, any news on this one?

furai, 2017-12-11:
Isn't it exactly like the other DA? Just different usb id and that's all? It goes up to 64000DPI.

terrycain, 2017-12-11:
@furai You mean 6400DPI right?

furai, 2017-12-11:
Uh, yeah, 6400 DPI. Pressed 0 one time too much. Edited.

danshou, 2018-01-03:
Is there any way to help with this ? At least with the testing maybe.

manu212, 2018-05-11:
is it possible to get this mouse supported?

ZakharH, 2018-05-11:
If you're looking for something to use right now, look into `da2013ctl`. I'm using it currently, but I really am looking forward to when this is done

hamjo, 2018-11-11:
Any update on this? It's been in progress for 7 months now.

arno01, 2019-02-06:
I think this can be closed now @z3ntu

z3ntu, 2019-02-06:
Fixed by 791a4ad3050be477d741b921d1b522494327bced

closed 2019-02-06 by z3ntu