openrazer/openrazer #197: Razer Naga Epic support

TooPointOhh, 2017-02-10:
Linux device identifier: 1532:001F
Product ID: RZ01-005101
12 Macros

z3ntu, 2017-02-10:
It has a wireless dock, right? That one should be `1532:0021`.

TooPointOhh, 2017-02-10:
hmm okay. I am pretty sure that it is 001F but I will try the other

z3ntu, 2017-02-10:
Please post the output of
`lsusb -d 1532:00 -v`
and
`ls -lah /dev/input/by-id/`
and read [this on how to get packet captures](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol
). Also screenshots of all available options on Razer Synapse are needed.

z3ntu, 2017-02-16:
@etxzay this here is another project that supports changing the color of Chroma devices. Unfortunately the Razer Naga Epic isn't support yet but you can help with getting there :)

(note to others: redirected him here from another repo)

etxzay, 2017-02-18:
@z3ntu it has a dock which can be a bluetooth access point for wireless mouse connection, and docks has id `1532:0021`

```
etxzay@blink ~> lsusb -d 1532:001F -v

Bus 003 Device 009: ID 1532:001f Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x001f 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
```

```
etxzay@blink ~> ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 100 фев 18 08:13 .
drwxr-xr-x 4 root root 460 фев 18 08:13 ..
lrwxrwxrwx 1 root root   9 фев 18 08:13 usb-Razer_Razer_Naga_Epic-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 фев 18 08:13 usb-Razer_Razer_Naga_Epic-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 фев 18 08:13 usb-Razer_Razer_Naga_Epic-mouse -> ../mouse0`
```

Both commands where executed while mice was connected wired (direct USB cable to PC)

spannerman79, 2017-03-28:
> Also screenshots of all available options on Razer Synapse are needed.

I'm about to shift from Windows 7 to Linux, thought that I could help out with the screenshots @z3ntu 

[image]
[image]
[image]
[image]
[image]
[image]

terrycain, 2017-06-25:
Need packet captures for the following (each item in its own file please)

* Static Effect (diffferent colours, red blue etc...)
* Spectrum cycling toggle on off
* Wired Brightness from off -> bright -> off -> bright a few times
* Wireless Brightness from off -> bright -> off -> bright a few times
* DPI 100 -> 5600 a few times
* Polling rate 125 -> 1000 a few times
* Sleep mode from 1 -> 15 a few times
* Flash battery indicator 5->25 a few times

Destroyer, 2017-07-16:
Tested in wired mode on Kali 2016.2 w/ W10 x64
```
~ lsusb -d 1532:001F -v

Bus 001 Device 009: ID 1532:001f Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x001f 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
```

Static effect - I've selected all colors from the top row (before capture white)
Spectrum cycling on->off 3x
Wired Brightness (before capture off) -> bright -> off 3x
Wireless Brightness (before capture off) -> bright -> off 3x
DPI (before capture 100) -> 5600->100 3x
Polling rate (before capture 125) -> 1000->125 3x
Enter sleep mode (before capture 1) -> 15->1 3x
Blink scroll wheel light when power level is below (before capture 5%) -> 25->5 3x

Let me know if this is ok

[Captures.zip](https://github.com/terrycain/razer-drivers/files/1151334/Captures.zip)

terrycain, 2017-07-17:
Will have a look soon, am super busy but good job on all teh pcaps

terrycain, 2017-08-05:
Install debs from here - https://drive.google.com/open?id=0BydYgD2LzDUCenp0SEh0akcxMGc - reboot, then run 
```
ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razermouse
ls -l /sys/bus/hid/drivers/razermouse/0*/
```

jcstryker, 2018-01-20:
@terrycain I don't have a debian/ubuntu workstation at the moment. Is there anything else can I do to help test this model?

z3ntu, 2018-01-20:
You can build it yourself for any other distro. See the wiki for a few pointers (I think the site is called 'Building packages')

spannerman79, 2018-01-20:
@jcstryker https://openrazer.github.io/#download

Destroyer, 2018-01-21:
Sorry, I completly forgot to reply because I stopped using that mouse on my laptop. Anyway I installed it and this is my status on kali with latest updates and 4.14.0 kernel

```
# ls -l /sys/bus/hid/drivers
total 0
drwxr-xr-x 2 root root 0 Jan 21 02:04 hid-generic
drwxr-xr-x 2 root root 0 Jan 21 02:04 razermouse
```
```
# ls -l /sys/bus/hid/drivers/razermouse
total 0
--w------- 1 root root 4096 Jan 21 02:04 bind
lrwxrwxrwx 1 root root    0 Jan 21 02:04 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Jan 21 02:04 new_id
--w------- 1 root root 4096 Jan 21 02:04 uevent
--w------- 1 root root 4096 Jan 21 02:04 unbind
```

```
# ls -l /sys/bus/hid/drivers/razermouse/0*/
ls: cannot access '/sys/bus/hid/drivers/razermouse/0*/': No such file or directory
```

n-scope, 2018-01-22:
same exact output on a Linux Mint 18

z3ntu, 2018-01-28:
@Destroyer @n-scope Can you two join our community chat, either via Telegram (https://t.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q) or via Matrix (https://matrix.to/#/#openrazer:matrix.org) ? Debugging via issue comments takes forever.

Raineer, 2018-03-16:
If this is still open for testing, I'd like to help. I use a Naga Epic in wired mode; Mint 18.3.

The debs seem to have been removed from the Google Drive, but if they are still available I can give it a shot.

z3ntu, 2018-05-30:
@Raineer You can follow https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint with `feature_naga_epic_001F` as `TARGET_BRANCH`

jcstryker, 2018-06-14:
I have had success getting the device recognised with this branch.

Configuring the lighting works through polychromatic.

z3ntu, 2018-06-14:
@jcstryker Does *everything* work?

jcstryker, 2018-06-14:
Ah sorry, so after some more testing with a openrazer-git PKGBUILD for Arch Linux, modified to build the 'feature_naga_epic_001F' branch, this is what I have found.

Naga Epic Dock:
- Not Recognized at all

Naga Epic Wired Mode:
- Device is not owned by plugdev group
- After manual chown:
    - Device is recognized on daemon restart
    - Lighting (both Spectrum/Static), Brightness, DPI, and Polling Rate are all working and configurable
    - DPI setting is not saved on reboot (polychromatic)

[razer.log](https://github.com/openrazer/openrazer/files/2101894/razer.log)

jcstryker, 2018-06-14:
Rebuilt the package with the new rebased branch and everything looks good.

- Udev rule issue is fixed, I had rebooted for my last test, but since the rebuild everything works fine

- My DPI/Lighting/Poll Rate settings are now being preserved, not sure if this is the daemon or polychromatic

Concerning the dock, it is still not being recognized. I understand that you are looking into it.

Here is the output of `lsusb -d 1532:0021 -v`:
```
Bus 002 Device 004: ID 1532:0021 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0021 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Epic Dock
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
and `ls -lah /dev/input/by-id/`:
```
total 0
drwxr-xr-x 2 root root 180 Jun 14 09:17 ./
drwxr-xr-x 4 root root 580 Jun 14 09:17 ../
lrwxrwxrwx 1 root root   9 Jun 14 09:17 usb-Razer_Razer_Naga_Epic_Dock-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 Jun 14 09:17 usb-Razer_Razer_Naga_Epic_Dock-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9 Jun 14 09:17 usb-Razer_Razer_Naga_Epic_Dock-mouse -> ../mouse0
```
If you need anything else, e.g. packet captures, I can setup a Windows VM. Just let me know.

Thanks a bunch for your work on this.

ArmaanT, 2018-07-31:
I was wondering what the status of the 12 macro keys was for the Naga Epic. I'd be happy to upload packet captures or anything else needed. Thank you for all the work you've done on this already!

bansalani, 2018-11-16:
I am unable to get the debs folder on google drive.
I am a current user of Razar naga epic 
Bus 002 Device 007: ID      1532:001f    Razer USA, Ltd 
Can anyone please help me out to get it working on Ubuntu?

z3ntu, 2018-11-16:
@bansalani Just follow https://github.com/openrazer/openrazer/issues/197#issuecomment-393343688

guppy42, 2018-11-26:
Following the guidlines in https://github.com/openrazer/openrazer/issues/197#issuecomment-393343688 it works for me ( static color & spectrum ) - but only in wired mode, wireless just nets me a "unkown razer device" in polychromatic

bansalani, 2018-11-27:
> @bansalani Just follow [#197 (comment)](https://github.com/openrazer/openrazer/issues/197#issuecomment-393343688)

Thanks

franciscojmarin, 2019-02-21:
Sorry reposting in an "old" topic but im having problems with installing it following #197 instructions. I have a screenshot, part of it is in Spanish since my system have that language. Hope you can help me, i'm having problems with my mouse and i'm trying to fix it with this. It is when i use the command "_./scripts/build_debs.sh_" after the long one "_sudo apt install debhelper linux-headers-generic python3-setuptools lsb-release dkms # build dependencies_"

[image]

z3ntu, 2019-03-19:
@franciscojmarin Install `dh-python` with apt

guppy42, 2019-03-29:
What is the reason this wasn't added into release 2.5.0 ?
I more testing needed or?

heathkraft, 2019-05-14:
Poking this issue again, I've followed all of the instructions, and it's still not picking up my mouse.

heathkraft, 2019-08-01:
what testing needs to be done? I'll be more than happy to assist. I just want my mouse to work...

ChaoticAce, 2020-10-12:
Any update on this?

z3ntu, 2020-10-13:
Please test #1289, it's the code from a couple of years rebased on master. Installation instructions are linked in the PR (or ask if you need help).

ChaoticAce, 2020-10-20:
I asked about the wrong thread. I really wanted the status to the Naga Epic Chroma mentioned here: #899. I don't own the non-chroma naga epic _-_

Nickelbawker, 2020-12-11:
Does this open razer thing allow for DPI hot swap function? That's all I would need for linux the rest I can alter while I'm on windows. I have the wireless 2012 Naga Epic and it's my favorite mouse because of it's sleekness compared to the newer numpad style mice. Will be trying to keep this mouse until they release a similar product again, nothing else currently on the market close enough to it for me.

z3ntu, 2020-12-20:
@Nickelbawker What's "DPI hot swap function"?

Nickelbawker, 2020-12-26:
> @Nickelbawker What's "DPI hot swap function"?

[image]

kwamaking, 2021-10-30:
I have a razer naga epic and i've installed the daily release for openrazer and it shows this in polychromatic: 
[image]

i'm not sure if mine is an older or newer gen than the pid you have added in the above pull request: https://github.com/openrazer/openrazer/commit/40ec2b2822b57eba5c4a0d9614f85d67a326e105

This pid (1532:0021) shows up for both wired and wireless modes for my naga epic. 

If I can provide any additional details let me know, I just recently switched back to Linux on my gaming rig and would love to have full support for this.

ChaoticAce, 2022-04-03:
Can also confirm that this branch works for both wired and wireless mode. In order to get it running I had to remove the following packages:
```
razergenie-git
polychromatic-git
openrazer-meta-git
openrazer-driver-dkms-git 
openrazer-daemon-git 
python-openrazer-git 
```

Once removed I reinstalled the following packages with a modified PKGBUILD file:
```
openrazer-meta-git
openrazer-driver-dkms-git 
openrazer-daemon-git 
python-openrazer-git 
```

`PKGBUILD` File modifications: 
```
# source=("git+https://github.com/openrazer/openrazer.git")
source=("git+https://github.com/openrazer/openrazer.git#branch=feature_nagaepicchroma_0x003e")
```

Then I reinstalled the packages below and rebooted my machine:
```
razergenie-git
polychromatic-git
```
Both `polychromatic` and `razergenie` recognized my mouse after a reboot. I was able to modify Mouse: Brightness, Pollrate, sleep mode, lower power mode and DPI. Scroll Wheel: Brightness, Spectrum/static. Backlight, and all color zones. Glad to finally be able to modify my mouse. I haven't tried keymapping yet

Zellogie, 2022-07-03:
I get the same problem as @kwamaking
Same issue with both polycromatic and razercommander (which to me means it's openrazer that doesn't support this mouse)
I have tried with the #1289 fix.

Both wireless (the dock, pid 1532:0021) and wired(pid 1532:001f).
It's the Razer Naga Epic 2012 version.

And I have tried restarting the system and daemon using the wire or dock.

If there is something you want me to try please let me know. I'm still kind of new to Linux, since a few months, so I'm still in the learning phase.

CerealKiller-John, 2022-11-28:
I too have the same issue, and I don't understand what is meant by `feature_naga_epic_001F` as `TARGET_BRANCH`

jackk845, 2023-09-09:
i have the the same issue

z3ntu, 2023-09-21:
@jackk845 (and everybody else reading) Please check out #1289 and https://github.com/openrazer/openrazer/wiki/Building-a-package , otherwise this pull request will be stuck forever.

w-j-r, 2024-06-16:
What else needs to be done for this mouse? From what I have gathered so far is that the mouse in wired mode is recognized but the dock is not? I have just been using this mouse without any drivers since everything works OOTB, dock included, but some options like turning off the lights would be nice, so I am interested in seeing if I could be of any help.

Zellogie, 2024-06-16:
> What else needs to be done for this mouse? From what I have gathered so far is that the mouse in wired mode is recognized but the dock is not? I have just been using this mouse without any drivers [...]

My mouse broke (wired/battery button broke and battery swelled + something more I can't remember). But I do believe I have it somewhere. Should work in wired mode.

Can try to find it in case you or anyone need something tested.

w-j-r, 2024-06-16:
> @jackk845 (and everybody else reading) Please check out #1289 and https://github.com/openrazer/openrazer/wiki/Building-a-package , otherwise this pull request will be stuck forever.

Just did some testing on this and wanted to say the `feature_naga_epic_001F` branch works with the `1532:001f` razer naga epic in wired mode at least partially. The dock is not recognized, at least in razer-cli. I was not able to test with Polychromatic because it seems to crash on the "device" tab, perhaps because these drivers are old? Polychromatic had worked previously with the up to date openrazer drivers.

 So far in razer-cli I was able to successfully control the 
* DPI 
* poll rate 
* brightness settings (from off, and from 1-100%). I tested that the brightness settings saved on reboot. 

Things I am not sure about 
* Battery thresholds (not sure if actually works, but razer-cli did not complain, requires more testing)
* RGB effects (If I recall correctly RGB modes were only available for the Chroma models and not the Razer Naga Epic, because none of the effects worked, nor did changing RGB color.)

Let me know what else is needed.

My lsmod
`1532:001f Razer USA, Ltd Naga Epic (Wired)`

My device info via razer-cli 
```
Found 1 Razer device(s)
Razer Naga Epic:
   type: mouse
   DPI: (1800, 1800)
   max DPI: 5600
   polling rate: 500
   battery:
      charge: 64
      charging: True
   generic zone:
   scroll_wheel zone:
      brightness: 0.0
      colors: ['FF0000', '00FFFF', '0000FF']
      effect: static
      speed: 1
      wave_dir: 1
      effects available:
         brightness
         none
         spectrum
         static
   serial: PM1217051T01586
   firmware version: v1.4
   driver version: 3.6.1
   supported capabilities:
      name
      type
      firmware_version
      serial
      battery
      set_idle_time
      set_low_battery_threshold
      poll_rate
      dpi
      lighting
      lighting_scroll
      lighting_scroll_brightness
      lighting_scroll_spectrum
      lighting_scroll_static
      lighting_scroll_none

```

PR #1289 (2024-07-04): Add support for Razer Naga Epic
Fixes #197

For installation instructions see https://github.com/openrazer/openrazer/wiki/Building-a-package
closed 2024-07-04 by z3ntu