openrazer/openrazer #198: Deleted User account originally installed on linux mint

magnum93, 2017-02-11:
Hi Deleted the original user account to create a new linux user account, now when I log in the daemon cannot connect, how do I change the password for the daemon or allow the new user accesss?

magnum93, 2017-02-11:
I fixed it myself the new user has to be added to the plug-dev group

z3ntu, 2017-02-11:
Yes, the user has to be in the group `pludev`. If the issue is solved, please close the issue.

closed 2017-02-11 by terrycain

magnum93, 2017-02-11:
Many thanks for your assistance can we donate towards devlopment

terrycain, 2017-02-11:
> Many thanks for your assistance can we donate towards development

No probs, and sure if you want

magnum93, 2017-02-11:
Where can we send paypal donation? thanks

On 02/11/2017 06:48 PM, Terry Cain wrote:

Many thanks for your assistance can we donate towards development

No probs, and sure if you want

terrycain, 2017-02-11:
You can send it to `terry@terrys-home.co.uk` :smile: