openrazer/openrazer #199: RBS Late 2016 Keyboard does not work until a key is pressed (Ubuntu 16.10)

ioiobzit, 2017-02-13:
Hello everyone,

the keyboard on the RBS Late 2016 does not start to write until a key is pressed and the lights turn on.

The following link shows the error, whilst writing this ticket:
https://www.youtube.com/watch?v=YGxIlOaztbw

What information would you need to verify this?

Thank you,
ioio

z3ntu, 2017-02-13:
How long does it take to happen (that you need to double-press a key)? But it looks like its some sort of timeout...

terrycain, 2017-02-13:
@z3ntu Does it happen with and without the razer-driver installed?

terrycain, 2017-02-13:
Also someone on the chat reminded me of auto suspend. See if you can disable it for the keyboard and see if that stops it.

ioiobzit, 2017-02-14:
Hello,
@z3ntu: this new video shows the timeout
https://www.youtube.com/watch?v=hkBbBzmHQM4

@terrycain:
- with and without razer-drivers installed

- I've set the keyboard in the blacklist:
`/etc/laptop-mode/conf.d/runtime-pm.conf`
as follows:
`AUTOSUSPEND_USBID_BLACKLIST="1532:0220"`
folowed by `sudo service laptop-mode restart`

The issue is still there. What should i check?

vifino, 2017-02-14:
Well, that fixed it for me on my Blade 14. ( I'm the "someone on the chat". :P )

Maybe try using `powertop` to manually set autosuspend to enabled/disabled?

I can also provide a script that'll do the same thing, if you can't figure it out.

If it does help disabeling autosuspend, we can at least be sure of the issue and get a fix for it.
I think you use ubuntu, so laptop-mode-tools should be the thing to change settings of, like you did.

Did you... try rebooting? :smile:

ioiobzit, 2017-02-14:
Hi @vifino,
I'm not able to find my keyboard with powertop, a little help?
Correct I use Ubuntu 16.10 and yes :) not even rebooting helps.

vifino, 2017-02-14:
In `powertop` in the tab `Tunables` (which you navigate to using tab), you should see a device named `Razer Blade` somewhere, line looks somewhat like this `>> Good          Autosuspend for USB device Razer Blade [Razer]`, select it using the arrow keys, hit space to toggle it to bad.

ioiobzit, 2017-02-14:
Well, now the keyboard stays on and it always works.
Hmm so it's not really about razer-drivers or is it?

vifino, 2017-02-14:
No, it is not. It is about usb autosuspend and a not-so-great implementation on Razer's side.

Adding it to the blacklist should've stopped the device from being set as autosuspend capable.
However, that does not seem to be the case.

Here is a little script, which I wrote for myself, and with some luck, should allow the same thing to be done more easily.
```
#!/bin/sh

# Razer Blade 14" 2016 and Razer Blade Stealth Keyboard USB Autosuspend fix.
# Due to Razers Quality Engineering(tm), suspending the keyboard results in the LEDs turning off and it being non-responsive for the first key press (and the fans turning off alongside the keyboard on the RB 14).
# Disabling that results in the keyboard never turning off - Even when suspended!
# So, call this script with on to turn it on, off to turn it off. Doable with a systemd unit for hibernation handeling.

pm() {
        echo $2 > /sys/bus/usb/devices/$1/power/control
}

find_bladekb() {
        #lsusb -d 1532:020f | awk '{tmp=$2":"$4; print substr(tmp, 1, length(tmp)-1)}'
        for usbdev in /sys/bus/usb/devices/[0-9]*; do
                if [ -e "$usbdev/idVendor" ] && [ "`cat $usbdev/idVendor`" == 1532 ]; then
                        [ "`cat $usbdev/idProduct`" == 020f ] && echo $(basename $usbdev)
                        [ "`cat $usbdev/idProduct`" == 0220 ] && echo $(basename $usbdev)
                fi
        done
}

case "$1" in
        off|auto|pre|suspend|hibernate) pm $(find_bladekb) auto; break;;
        on|post|resume|thaw) pm $(find_bladekb) on; ;;
esac
```

Place this script somewhere in your $PATH, mark it as executable, run it with `on` or `off`.

It's not a perfect fix, but maybe it'll work.

closed 2017-02-14 by ioiobzit